arduino/arduino-cli #164: Upload not working

mo-pyy, 2019-03-07:
I am trying to upload the test sketch from the readme to an arduino leonardo. I have compiled it sucessfully, but when I try to upload I get the following: ```Connecting to programmer: Found programmer: Id = "CATERIN"; type = S Software Version = 1.0; No Hardware Version given. Programmer supports buffered memory access with buffersize=128 bytes. Programmer supports the following devices: Device code: 0x44```

per1234, 2019-03-08:
Is that the full output from the upload command?

mo-pyy, 2019-03-08:
Yes

karelfonteyn, 2019-09-23:
Any update/workaround about how to upload a sketch to an Arduino Leonardo using Arduino CLI? I still get the same error using the latest nightly build.

Lisiadito, 2019-12-17:
I get this message with the Arduino Micro

wtgee, 2020-01-10:
We are getting this same issue although that's not the full message in our case:

```bash
Connecting to programmer: .                                                                                                                                                          │····················································································
Found programmer: Id = "CATERIN"; type = S                                                                                                                                           │····················································································
    Software Version = 1.0; No Hardware Version given.                                                                                                                               │····················································································
Programmer supports auto addr increment.                                                                                                                                             │····················································································
Programmer supports buffered memory access with buffersize=128 bytes.                                                                                                                │····················································································
                                                                                                                                                                                     │····················································································
Programmer supports the following devices:                                                                                                                                           │····················································································
    Device code: 0x44                                                                                                                                                                │····················································································
                                                                                                                                                                                     │····················································································
avrdude: Expected signature for ATmega32U4 is 1E 95 87                                                                                                                               │····················································································
         Double check chip, or use -F to override this check.                                                                                                                        │····················································································
Error during Upload: uploading error: exit status 1  
```

We have successfully uploaded via the Arduino IDE on this same device with same cable/etc setup.  

We've also been having this issue for a long time but are only able now to address it.

Any updates or anything I can try?  We are trying to get this working for a network of remote devices and can't rely on the IDE upload.

cmaglie, 2020-01-10:
May you run the same command adding: `--log-level debug -v` and copy the output here?

In case you are on MacOSX Catalina you must install all the system updates available.

wtgee, 2020-01-10:
```

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "~/.avrduderc"
avrdude: warning at ~/.avrduderc:2: programmer arduino overwrites previous definition ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf:361.

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.16s

avrdude: Device signature = 0x3f0d0d
avrdude: Expected signature for ATmega32U4 is 1E 95 87
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

Error during Upload: uploading error: exit status 1

```

I was trying to change the baudrate with the .avrduderc (which I think I am doing wrong), so it just contains:

```bash
$ cat ~/.avrduderc 
programmer
  id    = "arduino";
  desc  = "Arduino";
  type  = "arduino";
  connection_type = serial;
  baudrate = 2400;
;
```

System
```bash
$ uname -a
Linux xxx 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

thekunalsaini, 2020-03-03:
Try to upload full output @mo-pyy

yan007931, 2022-10-22:
I have the same issue .
Please find below my detailed logs
Version: 2.0.0
Date: 2022-09-14T07:06:37.759Z
CLI Version: 0.27.1 [a900cfb2]

Copyright © 2022 Arduino SA
*********************************************************************************************
I installed this version sofware version : 
Version: 2.0.0
Date: 2022-09-14T07:06:37.759Z
CLI Version: 0.27.1 [a900cfb2]

Copyright © 2022 Arduino SA
**********************************************************************************************

per1234, 2022-10-23:
Hi @mo-pyy and @wtgee. You have posted the output from a successful upload so all is well. Congratulations!

If you have any other questions, you are welcome to post on the Arduino forum. We'll be happy to help you out over there:

https://forum.arduino.cc/

closed 2022-10-23 by per1234

feivur, 2022-11-20:
get this message with the Arduino Micro