arduino/arduino-cli #166: Adafruit SAMD Boards are not found by arduino-cli core search

BobCochran, 2019-03-10:
I just installed arduino-cli on a Ubuntu 18.04 LTS box.

I have an Adafruit ItsyBitsy M0 board. When I plug this in, the operating system detects it as

> Mar 09 19:09:20 deaf58 kernel: usb 3-1: new full-speed USB device number 4 using xhci_hcd
> Mar 09 19:09:20 deaf58 kernel: usb 3-1: New USB device found, idVendor=239a, idProduct=800f
> Mar 09 19:09:20 deaf58 kernel: usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> Mar 09 19:09:20 deaf58 kernel: usb 3-1: Product: ItsyBitsy M0 Express
> Mar 09 19:09:20 deaf58 kernel: usb 3-1: Manufacturer: Adafruit
> Mar 09 19:09:20 deaf58 kernel: usb 3-1: SerialNumber: F384ECC6504B5952312E314AFF021834
> Mar 09 19:09:20 deaf58 kernel: cdc_acm 3-1:1.0: ttyACM0: USB ACM device
> Mar 09 19:09:20 deaf58 kernel: usbcore: registered new interface driver cdc_acm
> Mar 09 19:09:20 deaf58 kernel: cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

...and arduino-cli sees it as:

`usbob2@deaf58:~/Arduino/MyFirstSketch$ arduino-cli board list
FQBN    Port        	ID       	Board Name
    	/dev/ttyACM0	239a:800f	unknown   
`

Searching for a core under the 'adafruit' name yields this:

```
usbob2@deaf58:~/Arduino/MyFirstSketch$ arduino-cli core search 'adafruit'
Searching for platforms matching 'adafruit'

ID          	Version	Name                                        
arduino:avr 	1.6.23 	Arduino AVR Boards                          
arduino:samd	1.6.20 	Arduino SAMD Boards (32-bits ARM Cortex-M0+)
```
However it cannot find contributed boards under the name "Adafruit SAMD Boards" which the Arduino 1.8.8 GUI on my Mac does locate, if I open "Boards Manager..." and search for contributed Adafruit boards.

I installed this:

` usbob2@deaf58:~/Arduino/MyFirstSketch$ arduino-cli core install arduino:samd`

...but my board is still not detected:

> usbob2@deaf58:~/Arduino/MyFirstSketch$ arduino-cli board list
> FQBN    Port        	ID       	Board Name
>     	/dev/ttyACM0	239a:800f	unknown   

I'm not familiar with the Go programming language, but if there is some way I can contribute to arduino-cli that will help resolve this issue, I'm happy to do so.

Thanks!

BobCochran, 2019-03-10:
I re-read the README.md and noticed the text for adding third-party cores and then update the package index. I attempted to add this package index:

`https://adafruit.github.io/arduino-board-index/package_adafruit_index.json`

to arduino-cli.yaml as instructed. However, the arduino-cli tree has 3 arduino-cli.yaml files in these locations:

go/src/github.com/arduino/arduino-cli/configs/testdata/navigate/inheritance/first/arduino-cli.yaml
go/src/github.com/arduino/arduino-cli/configs/testdata/navigate/inheritance/first/second/arduino-cli.yaml
go/src/github.com/arduino/arduino-cli/configs/testdata/navigate/local/first/second/arduino-cli.yaml

I updated all of them like so:

`sketchbook_path: /tmp
board_manager:
  additional_urls:
    - https://downloads.arduino.cc/package_index_mraa.json
    - https://adafruit.github.io/arduino-board-index/package_adafruit_index.json`

...but this did not seem to download the package_adafruit_index.json.

I also added arduino-cli.yaml to the path 

> go/src/github.com/arduino/arduino-cli/configs

...but this did not seem to get the adafruit package index downloaded, either.

gdsports, 2019-03-10:
This might help.

https://github.com/arduino/arduino-cli/issues/5

BobCochran, 2019-03-10:
@gdsports Thank you! I looked in yaml_serializer.go but could not find anything in the format mentioned in  Issue #5. That issue references a file .cli_config.yml that does not exist. I did notice a configuration.go file that is in /go/src/github.com/arduino/arduino-cli/configs and that seems to have a function NewConfiguration which, in part, returns this:

> BoardManagerAdditionalUrls: []*url.URL{defaultPackageIndexURL}

I'm thinking of changing that to point to Adafruit's package index file and seeing what happens.

It's still unclear how exactly to add additional URL's in a way that will be parsed and processed properly.

per1234, 2019-03-11:
>However, the arduino-cli tree has 3 arduino-cli.yaml files in these locations:
>
>go/src/github.com/arduino/arduino-cli/configs/testdata/navigate/inheritance/first/arduino-cli.yaml
go/src/github.com/arduino/arduino-cli/configs/testdata/navigate/inheritance/first/second/arduino-cli.yaml
go/src/github.com/arduino/arduino-cli/configs/testdata/navigate/local/first/second/arduino-cli.yaml

Those are used for testing of arduino-cli. They will not be used when you're actually running arduino-cli.

If you want to see the locations that are searched for your arduino-ci.yaml file, run this command:
```
arduino-cli config dump --debug
```
The priority is opposite of the order they are listed in the debug output (i.e. the configuration file in the current folder overrides the configuration file in the .arduino15 folder).

>I also added arduino-cli.yaml to the path
>
>      go/src/github.com/arduino/arduino-cli/configs

Unless you're actually building arduino-cli from source, it doesn't make sense for you to be editing files in the source code.

>That issue references a file .cli_config.yml that does not exist.

The file name has since been changed from .cli_config.yml to arduino-ci.yaml.

>I'm thinking of changing that to point to Adafruit's package index file and seeing what happens.

Unless you just enjoy hacking around in the code, you're better off using arduino-cli as it was designed to be used.

>It's still unclear how exactly to add additional URL's in a way that will be parsed and processed properly.

This is the correct format:
```yaml
board_manager:
  additional_urls:
    - https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
```
Note that the file in your other URL: `https://downloads.arduino.cc/package_index_mraa.json` does not exist and having this invalid URL will cause the installation of Adafruit SAMD Boards to fail.

With the above configuration in your arduino-ci.yaml file, you can now do this:
```
arduino-cli core update-index
arduino-cli core install adafruit:samd
```

---
It seems to me that the root of this issue is the need for documentation of the possible locations of arduino-ci.yaml. The format of the additional Boards Manager URLs in that file is already correctly documented here: https://github.com/arduino/arduino-cli#adding-3rd-party-cores

BobCochran, 2019-03-11:
@per1234, I'm very confused and need some clarity here. I installed arduino-cli within the past 24 hours. So I'm getting the most recent version of the master branch. I'm running it on a Ubuntu 18.04 LTS server box. You say in your post above that I should add my additional_urls to this file:

arduino-ci.yaml

However, it does not exist anywhere under my ~/go/src/github.com/arduino/arduino-cli/ directory. Can you tell me exactly where this file is located? 

Do I need to check out a totally different branch of arduino-cli in order to find the file, because it has not yet been merged into the master branch?

I did in fact change the file configuration.go to add the Adafruit package index to it. It took a bit of research to see how arrays and slices are created in go, and some experimentation. Eventually I was able to run `go install`, get a new binary, and the package indexes downloaded:

```
~/Arduino/MyFirstSketch$ arduino-cli core update-index
Updating index: package_index.json downloaded                                                                                                              
Updating index: package_adafruit_index.json downloaded    
```
This allowed me to install the correct Adafruit core:

```
~/Arduino/MyFirstSketch$ arduino-cli core list
ID                 	Installed	Latest	Name                                        
adafruit:samd@1.3.0	1.3.0    	1.3.0 	Adafruit SAMD Boards                        
arduino:avr@1.6.23 	1.6.23   	1.6.23	Arduino AVR Boards                          
arduino:samd@1.6.20	1.6.20   	1.6.20	Arduino SAMD Boards (32-bits ARM Cortex-M0+)
```

I was able to successfully compile MyFirstSketch. Unfortunately the upload failed due to an issue with the bossac path:

~$ arduino-cli upload -p /dev/ttyACM0 --fqbn adafruit:samd:adafruit_itsybitsy_m0 Arduino/MyFirstSketch
Error: fork/exec {runtime.tools.bossac-1.7.0.path}/bossac: no such file or directory
Could not execute upload tool.

The real point of this particular issue is to ask you: 

What is the path and name of the configuration file that needs to be edited in order to add additional URLs? The file name given in README.md appears to be wrong. No path to the file is specified. 

Thanks so much!

gdsports, 2019-03-11:
The Adafruit SAMD board package depends on files in the Arduino SAMD board package so be sure to install both. The missing bossac error is almost always because the Arduino SAMD board package is not installed.

per1234, 2019-03-11:
I do have Arduino SAMD Boards installed.

BobCochran, 2019-03-11:
I have the following board packages installed. Are these sufficient to avoid the "no such file or directory" error I experienced with bossac? 

```
$ arduino-cli core list
ID                 	Installed	Latest	Name                                        
adafruit:samd@1.3.0	1.3.0    	1.3.0 	Adafruit SAMD Boards                        
arduino:avr@1.6.23 	1.6.23   	1.6.23	Arduino AVR Boards                          
arduino:samd@1.6.20	1.6.20   	1.6.20	Arduino SAMD Boards (32-bits ARM Cortex-M0+)
```

gdsports, 2019-03-12:
@BobCochran Looks good to me. If missing files messages are still appearing you may have found a bug.

gdsports, 2019-03-12:
@BobCochran The problem might be considered a bug in Adafruit's board package or the new arduino-cli, I do not know which. I use the older Arduino IDE command line interface which works fine with Adafruit, ESP, SparkFun, etc. boards.

Here is an example.

```
#!/bin/bash
# Arduino command line build using the Arduino IDE command line interface
# https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc
IDEVER="1.8.8"
WORKDIR="/tmp/autobuild_$$"
mkdir -p ${WORKDIR}
# Install Ardino IDE in work directory
tar xf ~/Downloads/arduino-${IDEVER}-linux64.tar.xz -C ${WORKDIR}
# Create portable sketchbook and library directories
IDEDIR="${WORKDIR}/arduino-${IDEVER}"
LIBDIR="${IDEDIR}/portable/sketchbook/libraries"
mkdir -p "${LIBDIR}"
export PATH="${IDEDIR}:${PATH}"
cd ${IDEDIR}
which arduino
# Install Adafruit SAMD board package
arduino --pref "boardsmanager.additional.urls=https://adafruit.github.io/arduino-board-index/package_adafruit_index.json" --save-prefs
arduino --install-boards "adafruit:samd"
# Build Blink.ino for Adafruit Trinket M0
BOARD="adafruit:samd:adafruit_trinket_m0"
CC="arduino --port /dev/ttyACM2 --upload --verbose-upload --board ${BOARD}"
# This step fails with missing bossac
$CC ./examples/01.Basics/Blink/Blink.ino
# This fixs the missing bossac problem 
arduino --install-boards "arduino:samd"
$CC ./examples/01.Basics/Blink/Blink.ino
```

BobCochran, 2019-03-12:
@gdsports thank you so much for the comment and for providing the example. I'm going to dig into this some more and play with your script. I did have the 1.8.8 IDE downloaded to my Ubuntu account and untapped, but then I discovered arduino-cli. I was afraid the IDE might conflict with arduino-cli so I deleted it. That is easily fixed, I can re-download again and keep trying. I'll file an issue about bossac to either the Adafruit package repository or to arduino-cli. I definitely like using arduino devices on the linux command line.

tuna-f1sh, 2019-04-18:
I am also experiencing the issue of the `bossac` path being non-resolvable. It looks like it is an issue with `arduino-cli` not expanding the variable correctly as it seems to be trying to execute the variable name. The tool is apparently 'loaded':

```
INFO[0000] Configuration set
INFO[0000] arduino-cli-0.3.6-alpha.preview
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Reading sketch from dir: /Users/johnwhittington/Documents/Arduino/libraries/Adafruit_DRV2605/examples/basic
INFO[0000] Initializing package manager
INFO[0000] Loading hardware from: /Users/johnwhittington/Library/arduino15/packages
INFO[0000] Loading package adafruit from: /Users/johnwhittington/Library/arduino15/packages/adafruit/hardware
INFO[0000] Loaded platform                               platform="adafruit:samd@1.3.0"
INFO[0000] Loading package arduino from: /Users/johnwhittington/Library/arduino15/packages/arduino/hardware
INFO[0000] Loaded platform                               platform="arduino:samd@1.6.21"
INFO[0000] Checking existence of 'tools' path: /Users/johnwhittington/Library/arduino15/packages/arduino/tools
INFO[0000] Loading tools from dir: /Users/johnwhittington/Library/arduino15/packages/arduino/tools
INFO[0000] Loaded tool                                   tool="arduino:CMSIS@4.5.0"
INFO[0000] Loaded tool                                   tool="arduino:CMSIS-Atmel@1.1.0"
INFO[0000] Loaded tool                                   tool="arduino:CMSIS-Atmel@1.2.0"
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.0"
INFO[0000] Loaded tool                                   tool="arduino:arm-none-eabi-gcc@4.8.3-2014q1"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.7.0"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.8.0-48-gb176eee"
INFO[0000] Loaded tool                                   tool="arduino:openocd@0.9.0-arduino"
INFO[0000] Loaded tool                                   tool="arduino:openocd@0.9.0-arduino6-static"
INFO[0000] Loading package builtin from: /Users/johnwhittington/Library/arduino15/packages/builtin
INFO[0000] Checking existence of 'tools' path: /Users/johnwhittington/Library/arduino15/packages/builtin/tools
INFO[0000] Loading tools from dir: /Users/johnwhittington/Library/arduino15/packages/builtin/tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Searching tools required for board adafruit:samd:adafruit_feather_m0
INFO[0000] Touching port /dev/cu.usbmodem14421401 at 1200bps
INFO[0000] Waiting for upload port...
No new serial port detected.
ERRO[0011] Could not execute upload tool.                error="fork/exec {runtime.tools.bossac-1.7.0.path}/bossac: no such file or directory"
Error: fork/exec {runtime.tools.bossac-1.7.0.path}/bossac: no such file or directory
Could not execute upload tool.
```

Perhaps this issue should be renamed to 'Adafruit SAMD Bossac tool not resolving'

jeff-winn, 2019-06-10:
@BobCochran I had a similar issue with the board detection for the Adafruit Feather M0 SAMD board. That particular issue isn’t with the Arduino CLI, that issue is with the boards.txt file I’m the Adafruit SAMD board package. See [here](https://github.com/adafruit/ArduinoCore-samd/issues/113) for an issue I opened in their repository to fix that problem. 

If you know which board it should be listed as, they can make similar adjustments to the VID and PID values on the appropriate board and that will resolve it being “unknown” even though the package is present.

per1234, 2019-06-15:
>```Error: fork/exec {runtime.tools.bossac-1.7.0.path}/bossac: no such file or directory```

This is a bug on Adafruit's end:
https://github.com/adafruit/arduino-board-index/issues/22
for which a pull request has been submitted (but not yet merged):
https://github.com/adafruit/arduino-board-index/pull/23/files

masci, 2019-08-28:
Upstream issue issue is fixed, any update on this?

per1234, 2019-08-28:
Before https://github.com/adafruit/arduino-board-index/pull/32 was merged, I was able to reproduce the error on both the Arduino IDE and arduino-cli.

Using Adafruit SAMD Boards 1.5.3 (which correctly specifies its tool dependency on arduino's bossac 1.7.0 tool), with the Arduino IDE, the problem is fixed.

However, when I use it with arduino-cli @ fffa451acd60da9341df523c79e10e429c21de22, I get the same error during uploading:
```
Error during Upload: cannot execute upload tool: exec: "{runtime.tools.bossac-1.7.0.path}/bossac": file does not exist
```

The installation of Adafruit SAMD Boards 1.5.3 via `arduino-cli core install adafruit:samd` did install bossac 1.7.0 under the arduino_data folder at `packages/arduino/tools/bossac/1.7.0`, the same as was done by the installation in the Arduino IDE Boards Manager.

So it seems to me that, in addition to the fixed bug in the Adafruit SAMD Boards Boards Manager JSON file that caused the error, there is also a bug in arduino-cli that causes the same error even after the problem in Adafruit SAMD Boards was fixed.

armsp, 2019-09-02:
I think I have a similar issue. I am able to search for Adafruit SAMD boards and install them too using the following input from `arduino-cli.yaml` - 
```
board_manager:
  additional_urls:
    - https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
    - https://www.adafruit.com/package_adafruit_index.json
```
After that `search` and `update-index` and `install` works.  

However the error comes with the bossac files -
```
$ arduino-cli upload  --fqbn adafruit:samd:adafruit_feather_m0_express -p /dev/ttyACM0 blinky
No new serial port detected.
Error: cannot execute upload tool: fork/exec {runtime.tools.bossac-1.7.0.path}/bossac: no such file or directory
Error during Upload
```

cmaglie, 2019-09-05:
In https://adafruit.github.io/arduino-board-index/package_adafruit_index.json there are two bossac tools listed as required:

```
            {
              "packager": "arduino",
              "name": "bossac",
              "version": "1.7.0"
            },
            {
              "packager": "arduino",
              "name": "bossac",
              "version": "1.8.0-48-gb176eee"
            },
```

I don't know if the Adafruit core requires both version, anyway this "confuses" the CLI (that accepts only the last one). If I manually remove the extra tool from the index it works.

At a first glance I though that the index.json should be fixed by removing the extra instance of `bossac`, but in reality we already have enough expressivity in place to allow multiple versions of the same tool to be used, so maybe the fix could be to allow it.

per1234, 2019-09-05:
>I don't know if the Adafruit core requires both version

It does. They have two different upload tools defined in their platform.txt and used in boards.txt: `bossac` ([which uses bossac 1.7.0](https://github.com/adafruit/ArduinoCore-samd/blob/master/platform.txt#L123)), and `bossac18` ([which uses 1.8.0-48-gb176eee](1.8.0-48-gb176eee))

armsp, 2019-09-05:
@cmaglie which one did you remove? Or either of them work?  
What do you recommend we do? Wait for the changes you mention or manually edit the file and get it working?

armsp, 2019-09-05:
@per1234  @cmaglie  Well in that case I guess its not advisable to manually edit the file, is it?

cmaglie, 2019-09-05:
Nope, a fix for the cli is coming.

closed 2019-09-09 by cmaglie

armsp, 2019-09-14:
@facchinm @cmaglie any tentative release date for this fix ?

cmaglie, 2019-09-16:
It will be in the next 0.6.0 release, but not timeline yet.
You can try it using the nightly build: https://github.com/arduino/arduino-cli#download-a-nightly-build

armsp, 2019-09-16:
@cmaglie  Okay, great. Checked it with the nightly build....it worked on Adafruit Feather M0 Express : SAMD21 .

photomoose, 2019-12-19:
Has there been a regression of this? I'm seeing the same issue in version 0.6.0:

```
$ arduino-cli version
arduino-cli Version: 0.6.0 Commit: 3a08b07d458ef65fdefe88a7d71056227f626f5a

$ arduino-cli upload -p /dev/cu.usbmodem14301 --fqbn adafruit:samd:adafruit_feather_m0 test
No new serial port detected.
Error during Upload: cannot execute upload tool: fork/exec {runtime.tools.bossac-1.7.0-arduino3.path}/bossac: no such file or directory
```

PR #394 (2019-09-09): Allow multiple instances of the same 'required_tool' in cores
Fix #166
commit 52642458 (2019-09-09): Allow multiple instances of the same 'required_tool' in cores

Fix #166