arduino/arduino-cli #173: CLI hijacks IDE binaries etc

NormanDunbar, 2019-03-23:
**Arduino IDE: 1.8.5
CLI: 0.3.6 Alpha.
Linux Mint 19.1 64bit.**

I installed the IDE, cleanly, by downloading the Linux compressed tar.gz file. I untarred it and started the IDE from within the extracted directory. After setting my preferrences to show verbose compilation and upload messages, compiling sketches used compilers etc from within the directory I was running the IDE from - where I had untarred it to.

NOTE: I did not run the `install.sh` script, I just run the IDE from wherever it happens to be.

I then downloaded the latest CLI, version 0.3.6, for Linux and installed the `arduino:avr` platform which downloaded a few files into a freshly created `~/.arduino15` directory. I then created and compiled a test sketch to ensure all was well. It was.

However, whenever I compile a sketch in the IDE, it lists the compiler and source files from `~/arduino15` which belong to the CLI and not the IDE. The CLI has hijacked the IDE's files it seems.

Cheers,
Norm.

per1234, 2019-03-23:
It is the expected behavior for the Arduino IDE to use packages installed to `~/.arduino15`. The reason is that is the location the Arduino IDE's Boards Manager installs packages. You may not have noticed this before because the Arduino IDE comes with a bundled version of Arduino AVR Boards.

If you want the Arduino IDE and arduino-cli to use separate data folders, you have a couple of options:
- Set the `arduino_data` configuration property in arduino-cli.yaml to another location. This will leave `~/.arduino15` for exclusive use by the Arduino IDE.
- Run the Arduino IDE in portable mode: https://www.arduino.cc/en/Guide/PortableIDE. That will cause the Arduino IDE to use packages from the `portable` folder instead of `~/.arduino15`, leaving `~/.arduino15` for exclusive use by arduino-cli.

closed 2019-03-23 by per1234

NormanDunbar, 2019-03-23:
Hi @per1234 and thank you.

Do you think I haven't seen this behaviour before because I never run the `install.sh` script? I suspect maybe this is the case.

Happy to close this, and thanks again.

Cheers,
Norm.

per1234, 2019-03-24:
I think you haven't seen this behavior before because you haven't had a hardware core installed in `~/.arduino15` before. The Arduino IDE uses `~/.arduino15` whether or not install.sh is run. That's also where the Arduino IDE's preference settings are installed so the Arduino IDE will create that folder the first time you run it if it doesn't already exist.

You can run an experiment like this:

1. Delete `~/.arduino15` to clear the data that was saved there by arduino-cli.
2. Start the Arduino IDE.
3. Tools > Board > Boards Manager
4. Wait for downloads to finish
5. Click on "Arduino AVR Boards".
6. Select any other version from the dropdown version menu.
7. Click "Install".
8. Wait for the installation to finish.
9. Click "Close".

If you run a compilation with verbose output enabled, you should see that the Arduino IDE is now using the tools installed in the `~/.arduino15` folder. There is no mechanism for the IDE to know whether the package was installed by Boards Manager or arduino-cli. It just automatically gives preference to the Arduino AVR Boards package installed to `~/.arduino15` over the one bundled with the Arduino IDE (though it will display a warning on startup if the one in `~/.arduino15` is older than the bundled one).

NormanDunbar, 2019-03-24:
Hi @per1234 and thank you again.

I did as you advised and yes, the updated packages etc were indeed found in the `~/.arduino15` folder. This is fine and I understand it now. My original problem was simply that I had not updated the IDE, I had been using it happily on whatever version of the Arduino AVR Boards package that 1.8.5 had installed and then, the cli version downloaded a separate version and "forced" the IDE to use that instead of what it had been using earlier.

The reason it caused me a problem was because I'd been amending a couple of files under my original IDE location to "fix" something that was broken (I have a separate issue on that matter) and all of a sudden, my changes were never being picked up. That drove me nuts trying to see why my changes were apparently being ignored - until I turned verbose compilations on, and it was obvious. 

At least now I know what's happening, I can do the changes, as necessary, in the correct area now. :-)

Cheers,
Norm.