arduino/arduino-cli #174: JSON format not working properly with compile command

Renaud11232, 2019-03-24:
When using `--format json` with the `compile` command, the output is not properly formatted as JSON.

## Successful compilation
### Running
```bash
./arduino-cli compile --fqbn aruino:avr:uno Arduino/Test --format json
```
### Outputs
On stdout
```
Sketch uses 444 bytes (1%) of program storage space. Maximum is 30720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
```

## Failed compilation
### Running
```bash
./arduino-cli compile --fqbn aruino:avr:uno Arduino/Test --format json
```
### Outputs
On stdout
```
Build options changed, rebuilding all
{"Message":"Compilation failed.","Cause":"exit status 1"}
```
On stderr
```
/home/renaud/Arduino/Test/Test.ino: In function 'void setup()':
/home/renaud/Arduino/Test/Test.ino:3:1: error: 'a' was not declared in this scope
 a
 ^
```

thekunalsaini, 2020-03-03:
Firstly declare 'a'

Renaud11232, 2020-03-03:
Yes I know why it doesn't compile the issue isn't there.

The issue is that the output provided by `arduino-cli` is not properly formatted in json even though I specified the `--format json` flag.

thekunalsaini, 2020-03-03:
As when I used the json format it produces correctly

thekunalsaini, 2020-03-03:
Can you tell what steps are you following

Renaud11232, 2020-03-03:
I created two sketches. One that should compile correctly and one that shouldn't :

`cat Success/Success.ino`
```cpp
void setup() {
}

void loop() {
}
```
`cat Fail/Fail.ino`
```cpp
void setup() {
a;
}

void loop() {
}
```

Then I ran the following commands : 

```bash
./arduino-cli compile --format json --fqbn arduino:avr:uno Success
```
which gave me this 
```
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
```

And this : 
```bash
./arduino-cli compile --format json --fqbn arduino:avr:uno Fail
```
Which gave me this
```
/home/renaud/Fail/Fail.ino: In function 'void setup()':
/home/renaud/Fail/Fail.ino:3:1: error: 'a' was not declared in this scope
 a;
 ^
Error during build: exit status 1
```

The behavior seem to have slightly changed since I originally created this issue but the problem is still present in 0.9.0

```
$ ./arduino-cli version
arduino-cli Version: 0.9.0 Commit: e4d02b4
```

ubidefeo, 2020-03-03:
@Renaud11232 
I think the confusion originates from you expecting output in JSON format for everything, but this isn't the case.
you can get JSON output for a number of things such as
`arduino-cli board list(all) --format json`
`arduino-cli core list --format json`
but you won't get the result of compile formatted into a JSON string.
you only get the compiler's final output (unless you add `-v` to your command for verbosity) or, in case you have a C/C++ related error, the error from the compiler.
That's why you get the straigh compiler error when it finds you using a symbol which is not defined.

We will remove that `--format` option from some commands as we move forward, sorry for the confusion

(@rsora this is another one of those sections in which we should update the help text)

Renaud11232, 2020-03-03:
Oh OK
I thought it would be available for all commands as it's a global flag.

Now I know :)

Thanks for the answer

masci, 2020-03-03:
Just to clarify: while I confirm all @ubidefeo said, we also have a bug here.

We're not in full control of the output from the lower layers of the CLI and some informations that should be collected by the logger and correctly displayed through the `-v` flag leak to the stderr/stdout instead.

ubidefeo, 2020-03-03:
@masci I had this conversation with @cmaglie in the past when I had some stuff not showing up in my `stdout` and he said it's because of the way `avrdude` pipes output to different places.
I believe we should handle the type of output from the compiler such as @Renaud11232 's case, because an integration would need to parse the final output.
That aside, are we able to funnel output from both `stdout` and `stderr` into our final output?
There must be some way of issuing a redirect directive (sorry for the cumbersomeness of the terms)

masci, 2020-03-03:
Yes we can and should get control of stdout/stderr for any subprocess the CLI starts, there was a tentative mitigation you can see here https://github.com/arduino/arduino-cli/pull/364 but we decided not to proceed because the real fix consists in refactoring the low-level module we use to handle info and error messages "from below" (for the curious, I'm talking about this nasty module https://github.com/arduino/arduino-cli/tree/master/legacy/builder/i18n).