arduino/arduino-cli #177: SIGSEGV when compiling for custom boards

Renaud11232, 2019-03-27:
I noticed a problem when I try compiling a sketch for a custom board.
```
~$ ./arduino-cli compile --fqbn anet:avr:anet Marlin/
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x34 pc=0x2cf560]

goroutine 1 [running]:
github.com/arduino/arduino-cli/arduino/cores.(*PlatformRelease).GetLibrariesDir(0x0, 0x17230)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/cores/cores.go:210 +0x2c
github.com/arduino/arduino-cli/arduino/libraries/librariesmanager.(*LibrariesManager).AddPlatformReleaseLibrariesDir(0x12a17600, 0x0, 0x4)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/libraries/librariesmanager/librariesmanager.go:138 +0x1c
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*LibrariesLoader).Run(0x818d70, 0x12a461e0, 0x818d70, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/libraries_loader.go:62 +0x3f0
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x818d70, 0x12a461e0, 0x818d70, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/container_setup.go:63 +0x2a8
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.runCommands(0x12a461e0, 0x12843b9c, 0x22, 0x22, 0x1, 0x0, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:191 +0xc4
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*Builder).Run(0x12843cc0, 0x12a461e0, 0x58b470, 0x12a127d0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:109 +0x98c
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.RunBuilder(0x12a461e0, 0x0, 0x58b470)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:207 +0x24
github.com/arduino/arduino-cli/commands/compile.run(0x128d5400, 0x129600e0, 0x1, 0x3)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/compile/compile.go:240 +0x1090
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0x128d5400, 0x129600a0, 0x3, 0x4, 0x128d5400, 0x129600a0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:766 +0x20c
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0x128d4c80, 0x1295c3c0, 0x128d5900, 0x128d5540)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:852 +0x20c
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0x128d4c80, 0x3cecc, 0x440a14)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:800 +0x1c
main.main()
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x18
```

Here is the output with the `--debug` flag:
```
~$ ./arduino-cli compile --fqbn anet:avr:anet Marlin/ --debug
INFO[0000] Checking for config file in: /home/renaud/.arduino15/arduino-cli.yaml
INFO[0000] Reading configuration from /home/renaud/.arduino15/arduino-cli.yaml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /home/renaud
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Checking for config in: /arduino-cli.yaml
INFO[0000] error loading                                 error="open /arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/renaud/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/renaud/arduino-cli.yaml: no such file or directory"
INFO[0000] Configuration set
INFO[0000] arduino-cli-0.3.6-alpha.preview
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Executing `arduino compile`
INFO[0000] Initializing package manager
INFO[0000] Loading hardware from: /home/renaud/.arduino15/packages
INFO[0000] Loading package Sanguino from: /home/renaud/.arduino15/packages/Sanguino/hardware
INFO[0000] Loading package arduino from: /home/renaud/.arduino15/packages/arduino/hardware
INFO[0000] Checking existence of 'tools' path: /home/renaud/.arduino15/packages/arduino/tools
INFO[0000] Loading tools from dir: /home/renaud/.arduino15/packages/arduino/tools
INFO[0000] Loading package builtin from: /home/renaud/.arduino15/packages/builtin
INFO[0000] Checking existence of 'tools' path: /home/renaud/.arduino15/packages/builtin/tools
INFO[0000] Loading tools from dir: /home/renaud/.arduino15/packages/builtin/tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loading hardware from: /home/renaud/Arduino/hardware
INFO[0000] Loading package anet from: /home/renaud/Arduino/hardware/anet
INFO[0000] Loaded platform                               platform="anet:avr@"
Ts: 1553687869 - Running: GenerateBuildPathIfMissing
Setting build path to /tmp/arduino-sketch-E96CAEBD9ABAE19B046369917BB51116
Ts: 1553687869 - Running: EnsureBuildPathExists
Ts: 1553687869 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1553687869 - Running: AddAdditionalEntriesToContext
[... 14 lines cut ...]
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/cores/cores.go:210 +0x2c
github.com/arduino/arduino-cli/arduino/libraries/librariesmanager.(*LibrariesManager).AddPlatformReleaseLibrariesDir(0x12c035e0, 0x0, 0x4)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/libraries/librariesmanager/librariesmanager.go:138 +0x1c
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*LibrariesLoader).Run(0x818d70, 0x12c341e0, 0x818d70, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/libraries_loader.go:62 +0x3f0
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x818d70, 0x12c341e0, 0x818d70, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/container_setup.go:63 +0x2a8
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.runCommands(0x12c341e0, 0x12a43b9c, 0x22, 0x22, 0x1, 0x0, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:191 +0xc4
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*Builder).Run(0x12a43cc0, 0x12c341e0, 0x58b470, 0x12c084a0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:109 +0x98c
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.RunBuilder(0x12c341e0, 0x0, 0x58b470)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:207 +0x24
github.com/arduino/arduino-cli/commands/compile.run(0x12ac9540, 0x12b4e0a0, 0x1, 0x4)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/compile/compile.go:240 +0x1090
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0x12ac9540, 0x12b4e060, 0x4, 0x4, 0x12ac9540, 0x12b4e060)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:766 +0x20c
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0x12ac8dc0, 0x12b4a500, 0x12ac9a40, 0x12ac9680)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:852 +0x20c
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0x12ac8dc0, 0x3cecc, 0x440a14)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:800 +0x1c
main.main()
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x18
```
The problem does not appear when `arduino:avr` is installed.

per1234, 2019-03-28:
Please provide instructions for how you installed the `anet:avr` package so we can attempt to reproduce the issue.

Does the issue only occur when you're compiling Marlin, or does it also occur if you are compiling a minimal sketch like examples/01.Basics/BareMinimum?

Renaud11232, 2019-03-31:
Hello,
I've downloaded the `anet:avr` package [here](https://github.com/SkyNet3D/anet-board) and placed the content of the `hardware` folder into `~/Arduino/hardware`
Once that was installed i just ran `./arduino-cli compile --fqbn anet:avr:anet Marlin/` wich gave me the error.
The issue also occur when I try to compile `BareMinimum`.
The problem also exists on Windows with the package installed in `C:\Users\gaspa\Documents\Arduino\hardware`, also failing with `BareMinimum`.
```
PS C:\Users\gaspa\Downloads> .\arduino-cli.exe compile --fqbn anet:avr:anet .\BareMinimum
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x34 pc=0x6a58ed]

goroutine 1 [running]:
github.com/arduino/arduino-cli/arduino/cores.(*PlatformRelease).GetLibrariesDir(0x0, 0x20)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/cores/cores.go:210 +0x2d
github.com/arduino/arduino-cli/arduino/libraries/librariesmanager.(*LibrariesManager).AddPlatformReleaseLibrariesDir(0x11c1e840, 0x0, 0x4)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/libraries/librariesmanager/librariesmanager.go:138 +0x25
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*LibrariesLoader).Run(0xc1248c, 0x11c4e1e0, 0xc1248c, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/libraries_loader.go:62 +0x3fc
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0xc1248c, 0x11c4e1e0, 0xc1248c, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/container_setup.go:63 +0x323
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.runCommands(0x11c4e1e0, 0x11a99ba4, 0x22, 0x22, 0x1, 0x0, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:191 +0xc1
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.(*Builder).Run(0x11a99cc8, 0x11c4e1e0, 0x97a960, 0x11c146b8)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:109 +0xb3a
github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder.RunBuilder(0x11c4e1e0, 0x0, 0x97a960)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/arduino-builder/builder.go:207 +0x29
github.com/arduino/arduino-cli/commands/compile.run(0x11ad7180, 0x11b317a0, 0x1, 0x3)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/compile/compile.go:240 +0x116f
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0x11ad7180, 0x11b31760, 0x3, 0x4, 0x11ad7180, 0x11b31760)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:766 +0x20e
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0x11ad6a00, 0x11b4e140, 0x11ad7680, 0x11ad72c0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:852 +0x28a
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0x11ad6a00, 0x81f6d8, 0x82099c)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:800 +0x21
main.main()
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x1f
```
I also tried with custom boards installed with the boad manager. To try that I added the sanguino package index from [here](https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json) into `arduino-cli.yaml`, then updated the index with `./arduino-cli core update-index`, then installed the sanguino boards with `./arduino-cli core install Sanguino:avr`.
Once it was installed I ran `./arduino-cli compile --fqbn Sanguino:avr:sanguino BareMinimal/`. This gave me the same error.

I hope it will help.

zmarcantel, 2019-04-17:
I can confirm the  same behavior, using multiple custom boards/cores, both with Marlin and the basic example.

I tracked down the source of the segfault but still trying to fully understand (and hopefully get a PR up, but fighting dep with forks and such).

To further complicate things -- a workaround is to install any `arduino:{proc}` core.
After running `arduino-cli core install arduino:avr` all the `nil`s below are valid (visual inspection, have not tried flashing). So there's clearly something with silently erroring or not being checked around custom board directories.

The issue starts here:

```
github.com/arduino/arduino-builder.(*LibrariesLoader).Run(ctx *types.Context) error
```

The context passed is `nil` for `ctx.ActualPlatform` which ends up being used (in that function) here:

```
59:		actualPlatform := ctx.ActualPlatform
60:		platform := ctx.TargetPlatform
61:		if actualPlatform != platform {
62:			lm.AddPlatformReleaseLibrariesDir(actualPlatform, libraries.ReferencedPlatformBuiltIn)
63:		}
```

where the call to `AddPlatformReleaseLibrariesDir` uses the `actualPlatform` arg immediately, assuming non-nil:

```
func (sc *LibrariesManager) AddPlatformReleaseLibrariesDir(plaftormRelease *cores.PlatformRelease, location libraries.LibraryLocation) {
   138:		path := plaftormRelease.GetLibrariesDir()
```

which triggers the segfault (as `platformRelease` is `nil`.

zmarcantel, 2019-04-17:
Some more tracking down makes it seem the Sanguino *requires* the `arduino` package to be installed -- which makes the above workaround make sense at least.

```
func (pm *PackageManager) ResolveFQBN(fqbn *cores.FQBN) (....)

// sanguino has "arduino:arduino" for "build.core"
184:		coreParts := strings.Split(buildProperties.Get("build.core"), ":")

...

// this is valid, and looks like the standard "arduino" package
192:			buildPlatform := buildPackage.Platforms[fqbn.PlatformArch]
193:			buildPlatformRelease = pm.GetInstalledPlatformRelease(buildPlatform)
```

... but then we try to get the installed packages:

```
func (pm *PackageManager) GetInstalledPlatformRelease(platform *cores.Platform) *cores.PlatformRelease {
322:		releases := platform.GetAllInstalled()
323:		if len(releases) == 0 {
324:			return nil
325:		}

(dlv) p releases
[]*github.com/arduino/arduino-cli/arduino/cores.PlatformRelease len: 0, cap: 0, []
```

which bubbles *all* the way out to `LibraryLoader`s context not having an `ActualPlatform` set.

QuinnFreedman, 2019-08-23:
I can confirm that I'm getting the same problem trying to work with Adafruit boards on a clean install. As @zmarcantel suggested, running `arduino-cli core install arduino:avr` fixed it. This definitely needs a better error message at the very least though

masci, 2019-10-01:
I wasn't able to track the exact commits but this is fixed now:

```
» ./arduino-cli compile --fqbn anet:avr:anet Marlin/
Error during build: platform not installed
```

Renaud11232, 2019-11-14:
Hello, is the fix for this issue included in 0.6.0 ?

Because I tried the same thing I decribed above and got this : 
```
PS C:\Users\gaspa\Downloads> .\arduino-cli-0.6.0.exe version
arduino-cli-0.6.0.exe Version: 0.6.0 Commit: 3a08b07
PS C:\Users\gaspa\Downloads> .\arduino-cli-0.6.0.exe compile --fqbn anet:avr:anet .\BareMinimum\
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x68 pc=0x7ef175]

goroutine 1 [running]:
github.com/arduino/arduino-cli/arduino/cores.(*PlatformRelease).GetLibrariesDir(0x0, 0x2030000)
        /__w/arduino-cli/arduino-cli/arduino/cores/cores.go:224 +0x45
github.com/arduino/arduino-cli/arduino/libraries/librariesmanager.(*LibrariesManager).AddPlatformReleaseLibrariesDir(0xc000e4e400, 0x0, 0x2)
        /__w/arduino-cli/arduino-cli/arduino/libraries/librariesmanager/librariesmanager.go:143 +0x4a
github.com/arduino/arduino-cli/legacy/builder.(*LibrariesLoader).Run(0x1262df8, 0xc000070a80, 0x1262df8, 0x0)
        /__w/arduino-cli/arduino-cli/legacy/builder/libraries_loader.go:62 +0x45d
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x1262df8, 0xc000070a80, 0x1262df8, 0x40c91f)
        /__w/arduino-cli/arduino-cli/legacy/builder/container_setup.go:60 +0x217
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000070a80, 0xc000095408, 0x20, 0x20, 0x1, 0x0, 0xc0000953e8)
        /__w/arduino-cli/arduino-cli/legacy/builder/builder.go:209 +0xe6
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0xc000095660, 0xc000070a80, 0xd574c0, 0xc00145e2a0)
        /__w/arduino-cli/arduino-cli/legacy/builder/builder.go:116 +0xba2
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
        /__w/arduino-cli/arduino-cli/legacy/builder/builder.go:225
github.com/arduino/arduino-cli/commands/compile.Compile(0xd52360, 0xc00002a120, 0xc000095ca8, 0xd419e0, 0xc000006018, 0xd419e0, 0xc000006020, 0xc00016c630, 0xc0000a1c00, 0x9df0c2, ...)
        /__w/arduino-cli/arduino-cli/commands/compile/compile.go:172 +0x1172
github.com/arduino/arduino-cli/cli/compile.run(0xc00019aa00, 0xc000181650, 0x1, 0x3)
        /__w/arduino-cli/arduino-cli/cli/compile/compile.go:96 +0x285
github.com/spf13/cobra.(*Command).execute(0xc00019aa00, 0xc0001815c0, 0x3, 0x3, 0xc00019aa00, 0xc0001815c0)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x2b1
github.com/spf13/cobra.(*Command).ExecuteC(0x123ab60, 0x0, 0xb126e0, 0xc000076058)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x302
github.com/spf13/cobra.(*Command).Execute(...)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
        /__w/arduino-cli/arduino-cli/main.go:28 +0x34
```

And as mentioned earlier, installing ` arduino:avr` prevents it from crashing : 
```
PS C:\Users\gaspa\Downloads> .\arduino-cli-0.6.0.exe core install arduino:avr
Downloading packages...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5 already downloaded
arduino:avrdude@6.3.0-arduino17 already downloaded
arduino:arduinoOTA@1.3.0 already downloaded
arduino:avr@1.8.1 already downloaded
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5 installed
Installing arduino:avrdude@6.3.0-arduino17...
arduino:avrdude@6.3.0-arduino17 installed
Installing arduino:arduinoOTA@1.3.0...
arduino:arduinoOTA@1.3.0 installed
Installing arduino:avr@1.8.1...
arduino:avr@1.8.1 installed
PS C:\Users\gaspa\Downloads> .\arduino-cli-0.6.0.exe compile --fqbn anet:avr:anet .\BareMinimum\
Sketch uses 510 bytes (0%) of program storage space. Maximum is 126976 bytes.
Global variables use 9 bytes of dynamic memory.
```

masci, 2019-11-14:
Thanks @Renaud11232 we must have a regression because the same command from my previous comment now crashes, reopening

galou, 2019-11-20:
Same issue here.

## Bug Report

### Current behavior

Commands (after running install.sh from 2019-11-19):
```bash
arduino-cli core update-index --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json
cd /tmp && arduino-cli sketch new MyFirstSketch
arduino-cli -v compile --fqbn arbotix:avr:arbotixAd MyFirstSketch
```
Result:
```
Using board 'arbotixAd' from platform in folder: /home/gael/Arduino/hardware/arbotix/avr
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x68 pc=0xa61c6c]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.(*TargetBoardResolver).Run(0x126c780, 0xc001ea2380, 0x126c780, 0x0)
	/__w/arduino-cli/arduino-cli/legacy/builder/target_board_resolver.go:61 +0x47c
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x126c780, 0xc001ea2380, 0x126c780, 0x40c618)
	/__w/arduino-cli/arduino-cli/legacy/builder/container_setup.go:60 +0x210
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc001ea2380, 0xc0015d3408, 0x20, 0x20, 0x2aaaaaa01, 0x22, 0x28)
	/__w/arduino-cli/arduino-cli/legacy/builder/builder.go:209 +0xdf
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0xc0015d3660, 0xc001ea2380, 0xd53280, 0xc0014e30a0)
	/__w/arduino-cli/arduino-cli/legacy/builder/builder.go:116 +0xb9b
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/__w/arduino-cli/arduino-cli/legacy/builder/builder.go:225
github.com/arduino/arduino-cli/commands/compile.Compile(0xd4e300, 0xc000038148, 0xc0015d3ca8, 0xd3da40, 0xc00000e018, 0xd3da40, 0xc00000e020, 0xc00017a6c0, 0xc0001a9c00, 0x9e0dcb, ...)
	/__w/arduino-cli/arduino-cli/commands/compile/compile.go:172 +0x116b
github.com/arduino/arduino-cli/cli/compile.run(0xc0001aa500, 0xc000193300, 0x1, 0x4)
	/__w/arduino-cli/arduino-cli/cli/compile/compile.go:96 +0x27e
github.com/spf13/cobra.(*Command).execute(0xc0001aa500, 0xc000193280, 0x4, 0x4, 0xc0001aa500, 0xc000193280)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0x1242ec0, 0x0, 0xb08e00, 0xc00003c0b8)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x2fb
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
	/__w/arduino-cli/arduino-cli/main.go:28 +0x2d
```

### Expected behavior

This should compile.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli Version: 0.6.0 Commit: 3a08b07`
- OS and platform: Ubuntu 18.04, 64bit

### Additional context

PR #491 (2019-11-28): Avoid panics when compiling for custom boards
Fixes #177 

When working with custom boards some corner cases might arise and provide an inconsistent in-memory representation of the cores. This PR introduces more checks for structs and fields that might be `nil` at some point during compilation.
closed 2019-11-28 by masci