arduino/arduino-cli #179: arduino upload sketch panic

ololosha228, 2019-03-30:
So, i tried to upload sketch on my arduino nano, but i catch up panic. here it is:

```
=>: arduino upload -p /dev/ttyUSB0 MyFirstSketch.ino 

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x8c47e5]

goroutine 1 [running]:
github.com/arduino/arduino-cli/commands/upload.run(0xc4201c0f00, 0xc420184930, 0x1, 0x3)
	/home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/upload/upload.go:93 +0xc5
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0xc4201c0f00, 0xc4201848d0, 0x3, 0x3, 0xc4201c0f00, 0xc4201848d0)
	/home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:766 +0x2c1
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0xc42016bb80, 0xc4201bec80, 0xc4201b7680, 0xc4201b6f00)
	/home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:852 +0x30a
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0xc42016bb80, 0x8c7fbc, 0xc420199f78)
	/home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:800 +0x2b
main.main()
	/home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x27
```

i will try to understand where is the problem, but arduino-cli is **SO** raw as much as it can be. so sad 😞😞😞😞😞

ololosha228, 2019-03-30:
found reason: runtime panic not working while you use `--fqbn` argument like:
`arduino upload -p /dev/ttyUSB0 --fqbn arduino:avr MyFirstSketch.ino`

so i have 2 questions to maintainers:
what is fqbn? is this an abbriveation, brand, or what?
even if i understand what it is: what if i have nano without fqbn data like
```
arduino board list 
FQBN    Port        	ID       	Board Name
    	/dev/ttyUSB0	0403:6001	unknown  
```

if somebody have this problem, please, contact me, we will fix this bug 🙃

per1234, 2019-03-31:
>what if i have nano without fqbn data like

For an explanation of why there is no FQBN for your nano see:
https://github.com/arduino/arduino-cli#why-the-arduino-unomegaduemilanove-is-not-detected-when-i-run-arduino-cli-board-list

There is actually a specific mention of the Nano's FQBN in the FAQ:
https://github.com/arduino/arduino-cli#what-is-the-fqbn-for-

I provided some more general information on determining the FQBN for any board here:
https://github.com/arduino/arduino-cli/issues/138#issuecomment-459169051

ololosha228, 2019-04-04:
Hey, @per1234, thanks for answering!

Okay, may be fqbn is important cause different platforms have different build and upload systems, i understand it.
but is it **really** clearly mechanics? if not so, may be i can help to fix this thing and make it more obviously, plus write some docs or slt? i ask you about it, cause i did not found contributing guides.

per1234, 2019-04-04:
>is it really clearly mechanics?

I don't understand what you're asking.

>may be i can help to fix this thing and make it more obviously, plus write some docs or slt? i ask you about it, cause i did not found contributing guides.

I'm sure contributions to this project are welcome and will be considered for acceptance.

Although this repository doesn't have a contributing guide, there is one in the arduino/Arduino repository which contains information relevant to contributing in general:
https://github.com/arduino/Arduino/blob/4c9da50e57030aebd4320fcc24e3f497f504b663/CONTRIBUTING.md

ololosha228, 2019-04-04:
Thanks so much for contributing guide!
I promise, i close issue when i make pull request, that fix this bug.