arduino/arduino-cli #182: user: Current not implemented on linux/arm

Renaud11232, 2019-04-02:
When the program is run without the `USER` environment variable set, it shows : 
```
./arduino-cli board listall
Error: getting default arduino data dir: retrieving user home dir: user: Current not implemented on linux/arm
Error creating default configuration
```
This can happen for instance, when `arduino-cli` is used by a program launched using `start-stop-daemon`

Here is the debug output:
```
./arduino-cli board listall --debug
ERRO[0000] Error creating default configuration          error="getting default arduino data dir: retrieving user home dir: user: Current not implemented on linux/arm"
ERRO[0000] Error creating default configuration          error="getting default arduino data dir: retrieving user home dir: user: Current not implemented on linux/arm"
Error: getting default arduino data dir: retrieving user home dir: user: Current not implemented on linux/arm
Error creating default configuration
```

It fails with the other commands as well.

To reproduce the issue simply do : 
```bash
unset USER
./arduino-cli board listall
```

According to [this issue](https://github.com/golang/go/issues/14625) `cgo` might be missing for the linux/* targets.

SConaway, 2019-04-14:
If you're here for docker, run `export USER=root` on ubuntu.

masci, 2019-07-17:
Fixed in https://github.com/arduino/arduino-cli/pull/239

closed 2019-07-17 by masci