arduino/arduino-cli #185: List All Libraries

dilin-MS, 2019-04-03:
Hi @cmaglie ,

Can arduino-cli support listing all available libraries? Like you select the "all" type in Arduino IDE's Library Manager. Currently we can only use `arduino-cli lib search XXX` to search some libraries related to one core. Cases are that I have a device A, and want to use libraries from other core, like Arduino or something. Listing all available libraries can give information to user so he can check the library.

Thx.

ololosha228, 2019-04-04:
i think, this is not neccessary, but it can be easily implemented. can you write some cases, how to use it?

ololosha228, 2019-04-04:
it works like you say:
```
$ arduino-cli lib search
Name: "I2C Temperature Sensors derived from the LM75"
  Author:  Jeremy Cole <jeremy@jcole.us>, Liam Cole <liam@liamiam.com>
  Maintainer:  Jeremy Cole <jeremy@jcole.us>
...
```

try it! :full_moon_with_face:

closed 2019-04-05 by per1234

dilin-MS, 2019-04-06:
Thanks! That's what I need. :)