arduino/arduino-cli #191: Please specify instructions better

johnmorrisbeck, 2019-04-10:
On step 4 of the instructions it says to use the command:

arduino-cli core search mkr1000

That's all fine and dandy if you use the mkr1000 but I do not know if that is the arduino I have.

When I google the phrase mkr1000 I get arduino mkr1000 wifi. So I assume that if you had that board you would come across a page that tells you the codeword for that is simply "mkr1000".

I have the arduino uno r3.

When I run the comand:

arduino-cli core search uno

I get: 

Searching for platforms matching 'uno'                                                                                                                                                          
                                                                                                                                                                                                
ID              Version Name                                                                                                                                                                    
arduino:avr     1.6.23  Arduino AVR Boards                                                                                                                                                      
arduino:megaavr 1.6.26  Arduino megaAVR Boards 

___________

Do I install both of these? Just one? Will either one work?

Please specify in the instructions how to take the brand name of my arduino and look up the model number or whatnot I am supposed to type into the search command OR specify what to do if multiple results come up.

NormanDunbar, 2019-04-10:
Install arduino:avr for an Uno R3. Or indeed, any other Uno. 

I agree with your second paragraph.

Cheers,
Norm.

per1234, 2019-05-24:
Related:
- https://github.com/arduino/arduino-cli/issues/138
- https://github.com/arduino/Arduino/issues/8716
- https://github.com/arduino/arduino-cli/issues/146

This sort of confusion seems to be coming up regularly. I'm not sure how I feel about adding the amount of documentation it would take to make people fully understand how the FQBN system works when that documentation is already present in https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification (though interspersed with additional information that is not relevant). I'm leaning towards the opinion that arduino-cli is for more advanced users who have an understanding of how Arduino core packages work or can gain that understanding from reading the hardware specification. For everyone else, there is the IDE. So the documentation would only need a link to the hardware specification with a short explanation of why that information is important.

NormanDunbar, 2019-05-26:
Morning @per1234. Having used an Arduino Duemilanove for some 10 years, looking up docs on the Arduino.cc site as and when required, my first encounter with an FQBN came when I started testing the new CLI. It's fine having docs on Github, but then need to be linked from the official site. I tried searching for it, this was the result:

Your search - Fqbn site:https://www.arduino.cc - did not match any documents.

So, yes, your final sentence is correct, there must be a link to the appropriate docs.

Cheers,
Norm.

endorama, 2019-07-11:
@per1234 A note on 
> or can gain that understanding from reading the hardware specification

The page you linked [https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification](arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification) do **not** mention the word "fqbn" once. Searching that repo for "fqbn" yields only results in code, commits and issues.  
I found pretty difficult to "gain an understanding" and I even opened #271 because I had the same issue with a Nano 33. 

I agree with yout about the cli requiring a deeper knowledge of Arduino internals, but we may help a bit in smoothing some edges (in particular the FQBN bit, which is very present in the cli and a bit obscure)

PR #395 (2019-09-10): Show core name when listing boards
Fixes #191

Now that #336 has been released, the vicious cycle of not knowing the board name if the corresponding core is not available, and not knowing which core to install without the board name, is finally broken.

To help users who are not familiar with the FQBN concept to better understand which core has to be installed, a new column in the output of `board list` was added to show the name of the core that  supports the detected board.

The README was updated accordingly. 

Note: the core name wasn't added to the JSON output because that would require a [... cut]
closed 2019-09-10 by masci