arduino/arduino-cli #193: {recipe.output.tmp_file} missing.

AppDevMichael, 2019-04-13:
Hi all,

I am having a bit of a problem with compiling.
This is my error:

```
Sketch uses 3308 bytes (55%) of program storage space. Maximum is 6012 bytes.
Global variables use 92 bytes of dynamic memory.
Error: open C:\Users\USERNAME\AppData\Local\Temp\arduino-sketch-2A2571335EBEE0BEF9DF3C1808A4F396/{recipe.output.tmp_file}: The system cannot find the file specified.
Error copying output file.
```

Thanks in advance.

per1234, 2019-04-13:
Please post the arduino-cli command you used when you got this error.

AppDevMichael, 2019-04-13:
arduino-cli compile --fqbn digistump:avr:digispark-tiny sketch.ino

per1234, 2019-04-13:
It's really a bug in the `digistump:avr` hardware core, rather than in arduino-cli. The bug has actually been [fixed](https://github.com/digistump/DigistumpArduino/pull/51), but Digistump hasn't made a release since that time. Probably the best solution for you to be able to use arduino-cli with the Digispark is to implement this patch on your installation of Digistump AVR Boards:
https://github.com/digistump/DigistumpArduino/commit/d208cb7669cb975ac9eb8f99fa59a70fda62514c
The platform.txt file is most likely at `C:\Users\USERNAME\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\platform.txt`.

It is possible this issue is present in other 3rd party hardware cores, since a missing `recipe.output.tmp_file` definition only affects the rarely used **Sketch > Export Compiled Binary** in the Arduino IDE. I'm leaning toward just closing this as not being our problem, but I'll leave it open for now in case someone else here wants to consider whether there is a way to provide compatibility with outdated hardware cores that are missing `recipe.output.tmp_file`. In the case of Digistump, the whole Digispark project seems to have been abandoned, so I don't expect them to make a release with the updated platform.txt.

AppDevMichael, 2019-04-14:
When I do apply this patch I get another error;

```
Sketch uses 3320 bytes (55%) of program storage space. Maximum is 6012 bytes.
Global variables use 92 bytes of dynamic memory.
Error: open TestCmdOpen\TestCmdOpen.ino\TestCmdOpen.ino.digistump.avr.digispark-tiny.hex: The system cannot find the path specified.
Error copying output file.
```

per1234, 2019-04-14:
You should not specify the .ino file to the `compile` command. Only specify the sketch folder which contains that .ino file:
```
arduino-cli compile --fqbn digistump:avr:digispark-tiny TestCmdOpen
```
The reason is that sketches may contain multiple files, so the entire folder is the sketch, rather than just a single .ino file.

closed 2019-11-23 by github-actions[bot]