arduino/arduino-cli #194: [Question] How do you install this on the raspberry pi?

NovaViper, 2019-04-14:
I'm trying to install this library in order to use the Marlin Firmware flash plugin for Octoprint.

per1234, 2019-04-14:
1. Go to the "Releases" page of this repository: https://github.com/arduino/arduino-cli/releases
2. From the "Assets" list, click on the link that has "linuxarm" in the filename.
3. Wait for the download to finish.
4. Extract the downloaded file.

You can copy the extracted file to any convenient location on your drive. There is no additional installation required.

This issue tracker is intended to be used for reporting bugs and making feature requests. In the future, please ask this type of question on the [Arduino Forum](http://forum.arduino.cc/). I'm sure we'll be able to help you out over there.

closed 2019-04-14 by per1234

NovaViper, 2019-04-14:
How do I download the file on the Pi though? I don't have the desktop installed (only the command line)

per1234, 2019-04-14:
Use wget:
https://www.gnu.org/software/wget/