arduino/arduino-cli #197: Build output path customization

dilin-MS, 2019-04-15:
Hi, I was wondering whether arduino-cli support customization of build path?

Currently I can only set `sketchbook_path` and `arduino_data` through `arduino-cli.yaml` file. I am using arduino-cli within docker container. 

In order to realize **incremental build**, I need to preserve intermediate files in a specific folder, so next time these files can be found by the arduino-cli comiler and thus reduce the compilation time. However, the `sketchbook_path` only store the `XX.ino`, `XX.elf` and `XX.bin` file. Other files needed to reduce the next compilation time is stored automatically in `/tmp/` directory. Is there any way that I can change the default intermediate files output path `/tmp` to some customized path, like `/root/Arduino/MyFirstSketch/.build`? Where should I configure this value?
When I talk about **intermediate files**, I mean something like:
```bash
Not found: /tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/core/NTPClient/NTPClient.cpp.o
Not found: /tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/core/wiring_analog.cpp.o
Not found: /tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/core/httpclient/http_parsed_url.cpp.o
Not found: /tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/core/system/SystemWiFi.cpp.o
Not found: /tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/core/httpclient/http_c_response.cpp.o
Not found: /tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/core/Telemetry/Telemetry.cpp.o
Not found: /tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/core/cli/console_cli.cpp.o
...
```

## Reproduce the scenario

* `/root/.arduino15/arduino-cli.yaml` file
```bash
board_manager:
  additional_urls:
    - https://raw.githubusercontent.com/VSChina/azureiotdevkit_tools/master/package_azureboard_index.json
sketchbook_path: /root/Arduino
arduino_data: /root/.arduino15
```

When I ran `arduino-cli compile --fqbn AZ3166:stm32f4:MXCHIP_AZ3166 /root/Arduino/devkit/devkit.ino --debug` command, I got:
```bash
################################ Compile Application #################################
INFO[0000] Checking for config file in: /root/.arduino15/arduino-cli.yaml
INFO[0000] Reading configuration from /root/.arduino15/arduino-cli.yaml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /usr/bin
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Checking for config in: /arduino-cli.yaml
INFO[0000] error loading                                 error="open /arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /work/arduino-cli.yaml
INFO[0000] Configuration set
INFO[0000] arduino-cli-0.3.6-alpha.preview
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Executing `arduino compile`
INFO[0000] Initializing package manager
INFO[0000] Loading hardware from: /root/.arduino15/packages
INFO[0000] Loading package AZ3166 from: /root/.arduino15/packages/AZ3166/hardware
INFO[0000] Loaded platform                               platform="AZ3166:stm32f4@1.6.2"
INFO[0000] Checking existence of 'tools' path: /root/.arduino15/packages/AZ3166/tools
INFO[0000] Loading tools from dir: /root/.arduino15/packages/AZ3166/tools
INFO[0000] Loaded tool                                   tool="AZ3166:arm-none-eabi-gcc@5_4-2016q3"
INFO[0000] Loaded tool                                   tool="AZ3166:openocd@0.10.0"
INFO[0000] Loading package builtin from: /root/.arduino15/packages/builtin
INFO[0000] Checking existence of 'tools' path: /root/.arduino15/packages/builtin/tools
INFO[0000] Loading tools from dir: /root/.arduino15/packages/builtin/tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
Ts: 1555324955 - Running: GenerateBuildPathIfMissing
Setting build path to /tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128
Ts: 1555324955 - Running: EnsureBuildPathExists
Ts: 1555324955 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1555324955 - Running: AddAdditionalEntriesToContext
Ts: 1555324955 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1555324955 - Running: HardwareLoader
Ts: 1555324955 - Running: PlatformKeysRewriteLoader
Ts: 1555324955 - Running: RewriteHardwareKeys
Ts: 1555324955 - Running: TargetBoardResolver
Ts: 1555324955 - Running: ToolsLoader
INFO[0000] Searching tools required for board AZ3166:stm32f4:MXCHIP_AZ3166
INFO[0000] Required tool                                 tool="AZ3166:arm-none-eabi-gcc@5_4-2016q3"
INFO[0000] Required tool                                 tool="AZ3166:openocd@0.10.0"
[... 173 lines cut ...]
Ts: 1555324978 - Running: ExportProjectCMake
Ts: 1555324978 - Running: Sizer
Sketch uses 215260 bytes (20%) of program storage space. Maximum is 1048576 bytes.
Global variables use 43896 bytes (16%) of dynamic memory, leaving 218248 bytes for local variables. Maximum is 262144 bytes.
INFO[0022] copying sketch build output                   from=/tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/devkit.ino.bin to=/root/Arduino/devkit/devkit.AZ3166.stm32f4.MXCHIP_AZ3166.bin
INFO[0022] copying sketch build output                   from=/tmp/arduino-sketch-EE403C8C9656741E279CF77C06E4F128/devkit.ino.elf to=/root/Arduino/devkit/devkit.AZ3166.stm32f4.MXCHIP_AZ3166.elf
```
* `ls -al /work/devkit/`
```bash
total 12
drwxr-xr-x 1 root root 4096 Apr 15 10:42 .
drwxr-xr-x 1 root root 4096 Apr 15 10:42 ..
-rwxrwxrwx 1 root root  129 Apr 15 04:00 devkit.ino
```
* `ls -al  /root/Arduino/devkit/`
```bash
total 1993
drwxrwxrwx 2 root root       0 Apr 15 10:23 .
drwxrwxrwx 2 root root       0 Apr 15 10:22 ..
-rwxr-xr-x 1 root root  216656 Apr 15 10:42 devkit.AZ3166.stm32f4.MXCHIP_AZ3166.bin
-rwxr-xr-x 1 root root 1818900 Apr 15 10:42 devkit.AZ3166.stm32f4.MXCHIP_AZ3166.elf
-rwxr-xr-x 1 root root     129 Apr 15 10:42 devkit.ino
```

Wait for the reply. Thanks a lot!

per1234, 2019-04-15:
Although you can't set it via arduino-cli.yaml, you can accomplish this via `arduino-cli compile`'s `--build-path` and `--build-cache-path` options.

dilin-MS, 2019-04-16:
Hi, @per1234 , I try the `arduino-cli compile`'s `--build-path` and `--build-cache-path` options. The result confuses me. Could you help explain?
## Background
I mount the folder `/root` to a docker volume., which means everytime I run the image, only `/root` will be preserved. Other folders will be flushed out(for example, `/tmp`). To preserve last time compilation result, I will set the compiled files to be stored in `/root` folder.

## Command: 
```bash
project_path=/root/Arduino/device
tmp_path=/root/tmp
```
1. `arduino-cli compile --fqbn AZ3166:stm32f4:MXCHIP_AZ3166 ${project_path} --debug`
2. `arduino-cli compile --fqbn AZ3166:stm32f4:MXCHIP_AZ3166 ${project_path} --build-cache-path ${tmp_path} --build-path ${tmp_path} --debug`
3. `arduino-cli compile --fqbn AZ3166:stm32f4:MXCHIP_AZ3166 ${project_path} --build-path ${tmp_path} --debug`
4. `arduino-cli compile --fqbn AZ3166:stm32f4:MXCHIP_AZ3166 ${project_path} --build-cache-path ${tmp_path} --debug`

I ran the above four commands separately. Only **the fourth command** achieves incremental build, which means re-compilation takes little time. Other commands' second-time compilation time consumption is as same as the first time. They didn't achieve incremental build. Why?
Why the `--build-cache-path` didn't save the time and even worse the time?

By running the fourth command **`arduino-cli compile --fqbn AZ3166:stm32f4:MXCHIP_AZ3166 ${project_path} --build-cache-path ${tmp_path} --debug`**, the recompile action only suggest "devkit.ino.cpp.o" as "Not Found".
```bash
Not found: /tmp/arduino-sketch-3E00829D3080742371472E93F1888B0F/sketch/device.ino.cpp.o
```

Any other best practice?

dilin-MS, 2019-04-23:
@per1234 ,Hi, how can I specify the build output file `XX.elf` and `XX.bin` to a customized folder, such as a folder named `.build/`? We do not want to place the file directly in the skech folder.
Currently if I set `--build-path=.build/`, the `arduino-cli compile` command will first store them in `.build` folder, and later copy them into my skech folder. 

* Command: 
```bash
root@93b792d24baa:/workspaces/workflow# arduino-cli compile --fqbn AZ3166:stm32f4:MXCHIP_AZ3166 /workspaces/workflow/ --build-path /workspaces/workflow/.build --debug
```
* Output:
```bash
...
Sketch uses 215372 bytes (20%) of program storage space. Maximum is 1048576 bytes.
Global variables use 43896 bytes (16%) of dynamic memory, leaving 218248 bytes for local variables. Maximum is 262144 bytes.
INFO[0002] copying sketch build output                   from=/workspaces/workflow/.build/workflow.ino.bin to=/workspaces/workflow/workflow.AZ3166.stm32f4.MXCHIP_AZ3166.bin
INFO[0002] copying sketch build output                   from=/workspaces/workflow/.build/workflow.ino.elf to=/workspaces/workflow/workflow.AZ3166.stm32f4.MXCHIP_AZ3166.elf
```

Looking for the solution. Thanks!

per1234, 2019-04-23:
You can use `arduino-cli compile`'s --output option for this purpose:
```
arduino-cli compile --fqbn AZ3166:stm32f4:MXCHIP_AZ3166 /workspaces/workflow/ --build-path /workspaces/workflow/.build --output /workspaces/workflow/.build/output --debug
```
will save the output files as `/workspaces/workflow/.build/output/output.elf` and `/workspaces/workflow/.build/output/output.bin`.

lhanson, 2019-10-17:
It seems like this should really be configurable via `arduino-cli.yaml`.

closed 2019-11-23 by github-actions[bot]

diegormsouza, 2023-01-09:
Hi everyone! I wonder if the latest IDE version is already supporting the configuration of the output path via arduino-cli.yaml.

I'm using the hex files in a simulator and this would facilitate a lot.

Thanks!