arduino/arduino-cli #203: Cannot compile blink code for Arduino Uno

shebet, 2019-05-02:
I am trying to get the Arduino-cli to upload the sample blink code to Arduino UNO board. 
I am running this on a  Nvidia TX2 board running Ubuntu 16.04(aarch64) .
Here is the list of board installed:

 ```
$arduino-cli core list
ID                      Installed       Latest  Name
arduino:avr@1.6.23      1.6.23          1.6.23  Arduino AVR Boards
arduino:megaavr@1.6.26  1.6.26          1.6.26  Arduino megaAVR Boards
```
When I try to compile it it says : 
```  
$ arduino-cli compile --fqbn arduino:avr:uno ~/Arduino/MyFirstSketch
Build options changed, rebuilding all
Error: fork/exec /home/nvidia/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-g++: no such file or directory
Compilation failed.```

Here is the same compile with the debug enabled: 

```
avr-g++: no such file or directory Compilation failed. arduino-cli --debug  compile --fqbn arduino:avr:uno ~/Arduino/MyFirstSketch
INFO[0000] Checking for config file in: /home/nvidia/.arduino15/arduino-cli.yaml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /home/nvidia/Documents/cli
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Checking for config in: /arduino-cli.yaml
INFO[0000] error loading                                 error="open /arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/nvidia/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/nvidia/arduino-cli.yaml: no such file or directory"
INFO[0000] Configuration set
INFO[0000] arduino-cli-0.3.6-alpha.preview
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Executing `arduino compile`
INFO[0000] Initializing package manager
INFO[0000] Loading hardware from: /home/nvidia/.arduino15/packages
INFO[0000] Loading package arduino from: /home/nvidia/.arduino15/packages/arduino/hardware
INFO[0000] Loaded platform                               platform="arduino:avr@1.6.23"
INFO[0000] Loaded platform                               platform="arduino:megaavr@1.6.26"
INFO[0000] Checking existence of 'tools' path: /home/nvidia/.arduino15/packages/arduino/tools
INFO[0000] Loading tools from dir: /home/nvidia/.arduino15/packages/arduino/tools
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.1"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino14"
INFO[0000] Loading package builtin from: /home/nvidia/.arduino15/packages/builtin
INFO[0000] Checking existence of 'tools' path: /home/nvidia/.arduino15/packages/builtin/tools
INFO[0000] Loading tools from dir: /home/nvidia/.arduino15/packages/builtin/tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
Ts: 1556811750 - Running: GenerateBuildPathIfMissing
Setting build path to /tmp/arduino-sketch-2952FBE27FB011D6DF4404B7737E1334
Ts: 1556811750 - Running: EnsureBuildPathExists
Ts: 1556811750 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1556811750 - Running: AddAdditionalEntriesToContext
Ts: 1556811750 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1556811750 - Running: HardwareLoader
Ts: 1556811750 - Running: PlatformKeysRewriteLoader
Ts: 1556811750 - Running: RewriteHardwareKeys
Ts: 1556811750 - Running: TargetBoardResolver
Ts: 1556811750 - Running: ToolsLoader
INFO[0000] Searching tools required for board arduino:avr:uno
INFO[0000] Required tool                                 tool="arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2"
INFO[0000] Required tool                                 tool="arduino:avrdude@6.3.0-arduino14"
INFO[0000] Required tool                                 tool="arduino:arduinoOTA@1.2.1"
Ts: 1556811750 - Running: AddBuildBoardPropertyIfMissing
Ts: 1556811750 - Running: LibrariesLoader
INFO[0000] Adding libraries dir                          dir=/home/nvidia/.arduino15/packages/arduino/hardware/avr/1.6.23/libraries location=platform
INFO[0000] Adding libraries dir                          dir=/home/nvidia/Arduino/libraries location=sketchbook
Ts: 1556811750 - Running: SketchLoader
Ts: 1556811750 - Running: SetupBuildProperties
Ts: 1556811750 - Running: LoadVIDPIDSpecificProperties
Ts: 1556811750 - Running: SetCustomBuildProperties
Ts: 1556811750 - Running: AddMissingBuildPropertiesFromParentPlatformTxtFiles
Ts: 1556811750 - Running: ContainerBuildOptions
Ts: 1556811750 - Running: CreateBuildOptionsMap
Ts: 1556811750 - Running: LoadPreviousBuildOptionsMap
Ts: 1556811750 - Running: WipeoutBuildPathIfBuildOptionsChanged
Build options changed, rebuilding all
Ts: 1556811750 - Running: StoreBuildOptionsMap
Ts: 1556811750 - Running: WarnAboutPlatformRewrites
Ts: 1556811750 - Running: RecipeByPrefixSuffixRunner
Looking for recipes like recipe.hooks.prebuild*.pattern
Ts: 1556811750 - Running: ContainerMergeCopySketchFiles
Ts: 1556811750 - Running: SketchSourceMerger
Ts: 1556811750 - Running: SketchSaver
Ts: 1556811750 - Running: AdditionalSketchFilesCopier
Ts: 1556811750 - Running: loggerAction
Ts: 1556811750 - Running: ContainerFindIncludes
Checking previous results for /tmp/arduino-sketch-2952FBE27FB011D6DF4404B7737E1334/sketch/MyFirstSketch.ino.cpp (result = /tmp/arduino-sketch-2952FBE27FB011D6DF4404B7737E1334/sketch/MyFirstSketch.ino.cpp.o, dep = /tmp/arduino-sketch-2952FBE27FB011D6DF4404B7737E1334/sketch/MyFirstSketch.ino.cpp.d)
Not found: /tmp/arduino-sketch-2952FBE27FB011D6DF4404B7737E1334/sketch/MyFirstSketch.ino.cpp.o
Ts: 1556811750 - Running: PrintUsedAndNotUsedLibraries
Ts: 1556811751 - Running: PrintUsedLibrariesIfVerbose
Ts: 1556811751 - Running: ExportProjectCMake
Ts: 1556811751 - Running: Sizer
ERRO[0000] Compilation failed.                           error="fork/exec /home/nvidia/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-g++: no such file or directory"
Error: fork/exec /home/nvidia/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-g++: no such file or directory
Compilation failed.
```

Any idea what I am missing here?

/ASP

facchinm, 2019-05-06:
@cmaglie is `linux-aarch64` target already supported in the CLI? The json lists the toolchain as
```
            {
              "size": "33141295",
              "checksum": "SHA-256:0fa9e4f2d6d09782dbc84dd91a302849cde2f192163cb9f29484c5f32785269a",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avr-gcc-5.4.0-atmel3.6.1-arduino2-aarch64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-5.4.0-atmel3.6.1-arduino2-aarch64-pc-linux-gnu.tar.bz2"
            },
```

cmaglie, 2019-05-06:
nope, will add

mpovel, 2019-06-18:
Just tried with arduino-cli-0.3.6-alpha.preview-linuxaarch64 and got:
root@orangepipc:~/Arduino# ./arduino-cli core update-index
Updating index: package_index.json downloaded                                   
root@orangepipc:~/Arduino# ./arduino-cli core install arduino:avr
The tool arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2 is not available for the current OS

cmaglie, 2019-06-18:
it's not yet in 0.3.6-alpha, please try this build --> https://github.com/arduino/arduino-cli/pull/204#issuecomment-490067165

mpovel, 2019-06-21:
Hi, sorry there is no ARM64 executable ;-(
root@orangepipc2:~/Arduino# ./arduino-cli-48-PR204-linuxarm 
-bash: ./arduino-cli-48-PR204-linuxarm: No such file or directory
root@orangepipc2:~/Arduino# file ./arduino-cli-48-PR204-linuxarm 
./arduino-cli-48-PR204-linuxarm: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, not stripped
root@orangepipc2:~/Arduino# file arduino-cli-0.3.6-alpha.preview-linuxaarch64 
arduino-cli-0.3.6-alpha.preview-linuxaarch64: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, not stripped

PR #204 (2019-07-11): Added arm64 support
Fix #203

/cc @shebet
commit a17ecb9f (2019-07-11): Added arm64 support (#204)

* Added arm64 support

Fix #203

* Fixed typo in test
closed 2019-07-11 by masci