arduino/arduino-cli #209: arduino-cli upload fail

usijeff, 2019-05-16:
Hi all, 
after i use arduino-cli compile 
=========================================
[root@localhost Arduino]# arduino-cli compile --fqbn arduino:avr:uno /root/Arduino/LED_ON/
Alternatives for Wire.h: [Wire@1.0]
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
Sketch uses 3766 bytes (11%) of program storage space. Maximum is 32256 bytes.
Global variables use 412 bytes (20%) of dynamic memory, leaving 1636 bytes for local variables. Maximum is 2048 bytes.
=============================================

looks compile success, and upload 
=============================================
[root@localhost Arduino]# arduino-cli upload -p /dev/ttyACM0 --fqbn arduino:avr:uno root/Arduino/LED_ON
Compiled sketch not found: root/Arduino/LED_ON/LED_ON.arduino.avr.uno.hex. Please compile first.
==============================================

it say no .hex, but .hex is exist
==============================================
[root@localhost LED_ON]# ls
LED_ON.arduino.avr.uno.elf  LED_ON.arduino.avr.uno.hex  LED_ON.ino
==============================================

where is the problem?
Thanks all!

NormanDunbar, 2019-05-16:
Looks like you compiled in /root/... but tried to upload in just root/... with no leading '/'.

````
arduino-cli compile --fqbn arduino:avr:uno /root/Arduino/LED_ON/
````
and
````
arduino-cli upload -p /dev/ttyACM0 --fqbn arduino:avr:uno root/Arduino/LED_ON
````
_However_, what are you doing running as root? That's really a bad idea!

HTH

Cheers,
Norm.

usijeff, 2019-05-17:
Yes, you are right,  solve.
thanks!
and i will running not as root.

closed 2019-05-17 by per1234

NormanDunbar, 2019-05-17:
@usijeff You are welcome.