arduino/arduino-cli #210: Set custom build flags

manasIU, 2019-05-20:
Hi, in my boards.txt file I am using my own build flags (eg. -DFLAGNAME). Arduino-cli recognises these flags but sets these to their default values (0). How do I tell arduino-cli to set certain flags?

per1234, 2019-05-20:
>Arduino-cli recognises these flags but sets these to their default values (0).

I can't reproduce this. Could you please provide more details, including the exact line from your boards.txt file and a minimal demonstration sketch so that I can reproduce your setup exactly?

>How do I tell arduino-cli to set certain flags?

You can use arduino-cli's `--build-properties` option to do things like:
```
arduino-cli compile --fqbn arduino:avr:uno --build-properties build.extra_flags=-DFLAGNAME SketchName
```
but be aware this will override the value of the `build.extra_flags` property that might have been set in your boards.txt.

bxparks, 2019-05-20:
I would like to make a feature request for the Arduino-CLI to explicitly support user-defined `-D` macros without resorting to this hack (which also happens to work in the [Arduino IDE CommandLine](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) ). You mentioned in https://forum.arduino.cc/index.php?topic=537500.0 that the Arduino devs have previously been reluctant to do this, but maybe they have changed their minds due to the ongoing work on the Arduino CLI.

manasIU, 2019-05-21:
@per1234  My boards.txt file looks something like this : 
```
# Dragonfly-L496RG
Dragonfly-L496RG.menu.debugOptions.none=None
Dragonfly-L496RG.menu.debugOptions.none.build.debugEnable=1
```
When I compile with verbose on, I see the -DDEBUGENABLE=0 in the output even after using your suggestion of using the `--build-properties build.extra_flags=-DFLAGNAME`

per1234, 2019-05-21:
Is this an existing unmodified hardware core? If so, post a link to it. If you customized it yourself, then you need to provide the full boards.txt and platform.txt.

I also need the complete arduino-cli command you're running.

per1234, 2019-05-24:
>I would like to make a feature request for the Arduino-CLI to explicitly support user-defined -D macros

We can consider https://github.com/arduino/arduino-cli/issues/159 as the dedicated issue for that feature request and use this issue to investigate @manasIU's problem.

manasIU, 2019-05-24:
It's a slightly modified Dragonfly-l496RG board with custom debug flag options added ([link](https://github.com/GrumpyOldPizza/arduino-STM32L4)).
Full arduino-cli command : 
`arduino-cli compile  --fqbn grumpyoldpizza:stm32l4:Dragonfly-L496RG --build-properties build.extra_flags=-DDEBUGENABLE=1 sketch_name --verbose`
When I run the above command, I see `-DDEBUGENABLE=0` in the verbose output.

nerdralph, 2020-01-08:
> > How do I tell arduino-cli to set certain flags?
> 
> You can use `arduino-cli compile`'s `--build-properties` option to do things like:
> 
> ```
> arduino-cli compile --fqbn arduino:avr:uno --build-properties build.extra_flags=-DFLAGNAME SketchName
> ```
> 
> but be aware this will override the value of the `build.extra_flags` property that might have been set in your boards.txt.

Is there a way to save the fqbn and build-properties in the global config so they don't have to be repeated on the command line?  I skimmed the source but couldn't find anything obvious.  For now I'm using shell aliases.

sayanee, 2020-07-04:
Any idea how to set 2 flags?

The edited `boards.txt` file is:

```
arduino_zero_edbg.build.extra_flags=-DCRYSTALLESS -D__SAMD21G18A__ {build.usb_flags}

```

What will be the equivalent `arduino-cli` command?

I have tried:

```
arduino-cli compile --fqbn arduino:samd:arduino_zero_native --build-properties "build.extra_flags=-DCRYSTALLESS -D__SAMD21G18A__ {build.usb_flags}" ./ --verbose
```

or

```
arduino-cli compile --fqbn arduino:samd:arduino_zero_native --build-properties build.extra_flags="-DCRYSTALLESS -D__SAMD21G18A__ {build.usb_flags}" ./ --verbose
```

but the USB port cannot be detected when I use `arduino-cli` vs when I edited the `boards.txt` directly.