arduino/arduino-cli #215: Attiny85 not compiling, but works in Arduino-IDE

yesimxev, 2019-05-22:
Hopefully this is just a missing library or so. I'm trying to compile a C source, as I need the CLI to make my process flawless. At the moment, I have to edit the source in Arduino IDE, and compile there, which works without anything really installed.  I have the Attiny core/board installed in both of them. 

```
root@kali:/usr/share/arduino/hardware/arduino/cores/arduino# arduino-cli compile -b attiny:avr:ATtinyX5:cpu=attiny85,clock=internal8 /root/Arduino/Magspoof
Build options changed, rebuilding all
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c: In function 'setup':
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c:52:18: error: 'OUTPUT' undeclared (first use in this function)
   pinMode(PIN_A, OUTPUT);
                  ^
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c:52:18: note: each undeclared identifier is reported only once for each function it appears in
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c:55:23: error: 'INPUT_PULLUP' undeclared (first use in this function)
   pinMode(BUTTON_PIN, INPUT_PULLUP);
                       ^
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c: In function 'blink':
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c:68:23: error: 'HIGH' undeclared (first use in this function)
     digitalWrite(pin, HIGH);
                       ^
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c:70:23: error: 'LOW' undeclared (first use in this function)
     digitalWrite(pin, LOW);
                       ^
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c: In function 'playTrack':
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c:115:28: error: 'HIGH' undeclared (first use in this function)
   digitalWrite(ENABLE_PIN, HIGH);
                            ^
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c:179:23: error: 'LOW' undeclared (first use in this function)
   digitalWrite(PIN_A, LOW);
                       ^
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c: In function 'loop':
/tmp/arduino-sketch-12E8601B7903A1C11826B3F60370E025/sketch/magspoof.c:290:37: error: 'LOW' undeclared (first use in this function)
   while (digitalRead(BUTTON_PIN) == LOW);
                                     ^
Error: exit status 1
Compilation failed.
```

per1234, 2019-05-22:
I can't reproduce the issue. Please provide a [minimal, complete, verifiable example](https://stackoverflow.com/help/reprex) sketch that demonstrates the problem.

yesimxev, 2019-05-22:
Attached the .ino renamed to txt
[sketch_mar21a.txt](https://github.com/arduino/arduino-cli/files/3206932/sketch_mar21a.txt)

per1234, 2019-05-22:
Did you try compiling that .ino file with arduino-cli? Your arduino-cli output shows errors from a .c file, but you haven't provided a .c file.

In order to make programming easier for beginners, special preprocessing is done on .ino files that is not done on any other type of source file. One of the preprocessing steps is to add `#include <Arduino.h>` to the .ino file. Arduino.h contains the declarations for the Arduino core API, including `OUTPUT`, `INPUT_PULLUP`, `HIGH`, and `LOW`. The sketch preprocessing allows you to use the Arduino core API in a .ino file without doing anything special. However, if you want to use the Arduino core API in a .c file, you need to make sure to add an `#include` directive for Arduino.h. This is true whether you're using the Arduino IDE or arduino-cli.

yesimxev, 2019-05-23:
Ah ok left the .c there by accident, moved that out, and the .ino compiled succesfully! Thanks

closed 2019-05-23 by yesimxev