arduino/arduino-cli #218: Multiple libraries were found, compilation failed.

filipgoc, 2019-05-30:
I'm compiling the Marlin firmware, which uses U8glib.h library

When I try to compile, I get the following error: 
```
arduino-cli compile --fqbn arduino:avr:mega Marlin    
Build options changed, rebuilding all
Alternatives for U8glib.h: [U8glib@1.19.1]
ResolveLibrary(U8glib.h)
  -> candidates: [U8glib@1.19.1]

Multiple libraries were found for "U8glib.h"
Used: /Users/filipgoc/Documents/Arduino/libraries/U8glib
Error: open /dev/null: too many open files

Compilation failed.
```

1. I had this working already. I successfully compiled the code and even uploaded it to Mega2560 board. 
2. Then it stopped working. I was trying to make sure I have the latest libraries, and I ran `arduino-cli lib install U8glib`, which worked fine. 
3. However, now I can't compile. I tried
    - restarting the computer and terminal
    - Uninstalling the U8glib from cli (lib uninstall), then I get that there is no U8glib library and thus error. 
    - Uninstalling the U8glib from cli and cli reinstalling - same error (too many files...)
    - Removing the U8glib manually from Documents/Arduino/Libraries. This behaved the same as cli uninstall. Not resolved.
4. BTW Arduino IDE compiles the same sketch with no problems. 

Any troubleshooting pointers appreciated. What happened?! Why was it working and now it's not?

filipgoc, 2019-05-30:
PS: I _can_ compile sketches that do not use U8glib (like blink).

filipgoc, 2019-05-30:
Curious. I was able to compile when I raised the limit on how many open files terminal can have. Thanks to https://superuser.com/questions/433746/is-there-a-fix-for-the-too-many-open-files-in-system-error-on-os-x-10-7-1 

`ulimit -S -n 1000`

So... it works, but something strange is happening. 

It seems that Arduino needs to open a lot of files to resolve which library it should use when there are multiple or something. When the limit is raised, it does resolve:

```
Alternatives for U8glib.h: [U8glib@1.19.1]
ResolveLibrary(U8glib.h)
  -> candidates: [U8glib@1.19.1]
```

umbynos, 2022-10-12:
Hi @filipgoc Could you please try with the latest version of the cli? https://github.com/arduino/arduino-cli/releases/latest