arduino/arduino-cli #219: Installed and not installed cores. How to tell apart?

Ahsan2, 2019-05-31:
readme file states that command "core search" returns a list of available packages with fields ID, Version, Installation and Name. But it seems the current version of arduino-cli doesnt return the Installation field. Is there any other way of telling apart the installed and not-installed packages?

Ahsan2, 2019-06-25:
Still stuck in the issue.
To be more clear, how to find out only the installed cores and libs?

endorama, 2019-07-10:
Do `arduino-cli core list` and `arduino-cli lib list` help?

I'm using version `arduino-cli Version: 0.3.7-alpha.preview Commit:  BuildDate: 2019-07-10 15:39:56.926258169 +0000 UTC`

masci, 2019-08-05:
@Ahsan2 I can confirm docs in the README are wrong, the `Installed` fields isn't shown anywhere and it's been like this since https://github.com/arduino/arduino-cli/pull/48, I don't know that part well enough to say wether it was an oversight or a design choice.

I'll treat this issue as a documentation bug and fix it in the README. In the case `arduino-cli core list` doesn't fit your use case feel free to open a feature request to get the `Installed` column back.

PR #330 (2019-08-08): [skip changelog] Fix examples in README
Fixes #219 

There's no `Installed` column is search output
closed 2019-08-08 by masci