arduino/arduino-cli #244: go build fails

mastrolinux, 2019-06-12:
On the current master, aftere a fresh `go get` I got:
```
arduino-cli$ go build
# github.com/facchinm/go-serial-native
In file included from /usr/include/x86_64-linux-gnu/bits/libc-header-start.h:33:0,
                 from /usr/include/string.h:26,
                 from libserialport_internal.h:25,
                 from linux.c:23:
/usr/include/features.h:184:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
linux.c: In function ‘list_ports’:
linux.c:180:2: warning: ‘readdir_r’ is deprecated [-Wdeprecated-declarations]
  while (!readdir_r(dir, &entry, &result) && result) {
  ^~~~~
In file included from libserialport_internal.h:60:0,
                 from linux.c:23:
/usr/include/dirent.h:183:12: note: declared here
 extern int readdir_r (DIR *__restrict __dirp,
            ^~~~~~~~~
# github.com/facchinm/go-serial-native
In file included from /usr/include/x86_64-linux-gnu/bits/libc-header-start.h:33:0,
                 from /usr/include/string.h:26,
                 from libserialport_internal.h:25,
                 from serialport.c:25:
/usr/include/features.h:184:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
In file included from serialport.c:25:0:
serialport.c: In function ‘sp_list_ports’:
libserialport_internal.h:168:5: warning: this statement may fall through [-Wimplicit-fallthrough=]
  if (sp_debug_handler) \
     ^
libserialport_internal.h:172:31: note: in expansion of macro ‘DEBUG_FMT’
 #define DEBUG_ERROR(err, msg) DEBUG_FMT("%s returning " #err ": " msg, __func__)
                               ^~~~~~~~~
serialport.c:345:3: note: in expansion of macro ‘DEBUG_ERROR’
   DEBUG_ERROR(SP_ERR_SUPP, "Enumeration not supported on this platform");
   ^~~~~~~~~~~
serialport.c:346:2: note: here
  default:
  ^~~~~~~
# golang.org/x/net/idna
../../../golang.org/x/net/idna/idna10.0.0.go:46:6: ToASCII redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:44:33
../../../golang.org/x/net/idna/idna10.0.0.go:51:6: ToUnicode redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:49:35
../../../golang.org/x/net/idna/idna10.0.0.go:56:6: Option redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:54:6
../../../golang.org/x/net/idna/idna10.0.0.go:63:6: Transitional redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:61:38
../../../golang.org/x/net/idna/idna10.0.0.go:69:6: VerifyDNSLength redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:67:35
../../../golang.org/x/net/idna/idna10.0.0.go:78:6: RemoveLeadingDots redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:76:37
../../../golang.org/x/net/idna/idna10.0.0.go:85:6: ValidateLabels redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:83:34
../../../golang.org/x/net/idna/idna10.0.0.go:106:6: StrictDomainName redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:104:33
../../../golang.org/x/net/idna/idna10.0.0.go:119:6: BidiRule redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:117:17
../../../golang.org/x/net/idna/idna10.0.0.go:125:6: ValidateForRegistration redeclared in this block
        previous declaration at ../../../golang.org/x/net/idna/idna.go:123:32
../../../golang.org/x/net/idna/idna10.0.0.go:125:6: too many errors
# google.golang.org/genproto/googleapis/rpc/status
../../../google.golang.org/genproto/googleapis/rpc/status/status.pb.go:23:11: undefined: proto.ProtoPackageIsVersion3
../../../google.golang.org/genproto/googleapis/rpc/status/status.pb.go:118:28: undefined: proto.InternalMessageInfo
# google.golang.org/grpc/binarylog/grpc_binarylog_v1
../../../google.golang.org/grpc/binarylog/grpc_binarylog_v1/binarylog.pb.go:206:34: undefined: proto.InternalMessageInfo
../../../google.golang.org/grpc/binarylog/grpc_binarylog_v1/binarylog.pb.go:474:34: undefined: proto.InternalMessageInfo
../../../google.golang.org/grpc/binarylog/grpc_binarylog_v1/binarylog.pb.go:534:34: undefined: proto.InternalMessageInfo
../../../google.golang.org/grpc/binarylog/grpc_binarylog_v1/binarylog.pb.go:581:29: undefined: proto.InternalMessageInfo
../../../google.golang.org/grpc/binarylog/grpc_binarylog_v1/binarylog.pb.go:645:29: undefined: proto.InternalMessageInfo
../../../google.golang.org/grpc/binarylog/grpc_binarylog_v1/binarylog.pb.go:711:30: undefined: proto.InternalMessageInfo
../../../google.golang.org/grpc/binarylog/grpc_binarylog_v1/binarylog.pb.go:751:35: undefined: proto.InternalMessageInfo
../../../google.golang.org/grpc/binarylog/grpc_binarylog_v1/binarylog.pb.go:800:29: undefined: proto.InternalMessageInfo
# github.com/arduino/arduino-cli/arduino/resources
arduino/resources/install.go:61:27: too many arguments in call to extract.Archive
        have (context.Context, *os.File, string, nil)
        want (io.Reader, string, extract.Renamer)
```

mastrolinux, 2019-06-12:
One less warning if we have this merged: https://github.com/facchinm/go-serial-native/pull/2

cmaglie, 2019-06-12:
uhm, we removed the dep on github.com/facchinm/go-serial-native some time ago, are you sure you're building master?

masci, 2019-06-17:
Can confirm the failing module isn't a dependency anymore and a `go build` from a clean state (empty `pkg/mod` folder) works on a linux64 env.

closed 2019-07-04 by masci