arduino/arduino-cli #252: Installed boards disappear when entry is missing from index file

masci, 2019-06-17:
## Bug Report

This is for a bug afflicting the IDE, see https://github.com/arduino/Arduino/issues/8936 for details. Let's double check how the same issue impacts the CLI and whether the solution being proposed might work here as well.

per1234, 2019-06-17:
Using arduino-cli 5cca8e278bee6f1c77aec8abe7f96af55011bcc7

### Regarding the issue of the installed boards disappearing if the download of their Boards Manager JSON file fails

What I did to test:
1. Publish a Boards Manager JSON file as a Git Gist.
1. Install a hardware core from that JSON file using `arduino-cli core install`.
1. Delete the Gist to simulate an outage of the JSON file.
1. `arduino-cli core update-index` now outputs the following:
    ```
    Updating index: package_index.json downloaded                                                                                                                                            
    Updating index: package_index.json downloaded                                                                                                                                            
    Updating index: package_asdf_index.json downloaded                                                                                                                                       
    Error: invalid package index in https://gist.githubusercontent.com/per1234/ac3efc0419d3c3a73c262edb7e350ee7/raw/f90a073a1b24823639f211249c01446e8bce5874/package_asdf_index.json: invalid character ':' after top-level value
    Error updating index
    ```

After that, package_asdf_index.json is still present in the data folder and `arduino-cli core list` still shows the hardware core from that JSON file is installed and I'm able to compile for the boards.

So it seems to me that arduino-cli has the correct behavior in this case.

### Regarding the issue of the installed boards disappearing if the entry for the installed release is removed from the JSON file

1. Install the hardware core
2. Remove the entry for the installed version from the JSON file
3. `arduino-cli core update-index`
4. Verify the updated version of the JSON file with the entry for the installed version of the hardware core removed was downloaded.

After that, `arduino-cli core list` still shows the hardware core from that JSON file is installed and I'm able to compile for the boards.

---
Despite these positive findings, @matthijskooijman's idea presented in https://github.com/arduino/Arduino/issues/8936#issuecomment-498203693 of storing the JSON at the time of the board installation is still useful in the event of the JSON file author later removing information relevant to the installation from their file because the `toolsDependencies` information is needed when checking whether any other core has a dependency on tools that that might be removed during a hardware core uninstallation.

masci, 2019-06-17:
@per1234 thanks for the detailed report, going to turn this issue into a feature request and schedule some work in our backlog to implement the `installed.json` feature.

PR #998 (2020-10-20): Enhances tracking of installed platforms
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [x] Docs have been added / updated (for bug fixes / features)

* **What kind of change does this PR introduce?**
Enhances tracking of installed platforms. [... cut]
closed 2020-10-20 by silvanocerza