arduino/arduino-cli #253: Error downloading index

itechnocrat, 2019-06-22:
## Bug Report

### Current behavior
Oh shit!
Does not work!
On Step 3. "Connect the board to your PC", I get the message:
Updating index: package_index.json downloaded
Error: Get: unsupported protocol scheme ""
Error downloading index

What the fuck?

### Expected behavior
Updating index: package_index.json downloaded

### Environment

- CLI version (output of `ardunio-cli version`): version 0.3.6-alpha.preview
- Go version (if building from sources):
- OS version: Ubuntu 16.04.6 LTS

### Additional context

per1234, 2019-06-22:
Are you using an arduino-cli.yaml file? If so, please post the contents here.

I can reproduce the error with this:
```yaml
board_manager:
  additional_urls:
    -
```

itechnocrat, 2019-06-22:
Where should arduino-cli.yaml be located?
In my case, it was generated in ~/.arduino15/

```
My arduino-cli.yaml:
proxy_type: auto
sketchbook_path: /home/technocrat/import/projects/programming/arduino
arduino_data: /home/technocrat/app/arduino
board_manager:
   additional_urls:
   - ""
```
Or
```
...
arduino_data: /home/technocrat/.arduino15
...
```
In both cases, the result is:

> 
> Updating index: package_index.json downloaded                                                                                                  
> Error: Get : unsupported protocol scheme ""
> Error downloading index

per1234, 2019-06-24:
>it was generated in ~/.arduino15/

What do you mean by "generated"? In my experience, that file is not generated. The user must create the file, otherwise arduino-cli uses the default configuration values.

>My arduino-cli.yaml:

Adding empty elements under `additional_urls` will result in the error message you received. The solution is to not add empty elements.

>Or

What is the meaning of the ellipses? Do they indicate there is more to the file that you didn't post?

itechnocrat, 2019-06-24:
Generated means that this is the result of the `arduino-cli config init`

What are ellipses?

I showed a variant of the arduino-cli.yaml fragment.

per1234, 2019-06-24:
I understand what you mean about generated now. Thanks for the explanation. However, when I run `arduino-cli config init` the default arduino-cli.yaml file generated looks like this:
```yaml
proxy_type: auto
sketchbook_path: /home/per/Arduino
arduino_data: /home/per/.arduino15
board_manager: {}
```
which will not cause the error you got.

However, I discovered that `arduino-cli config init` has an unexpected and undocumented behavior: arduino-cli recognizes arduino-cli.yaml files in several locations on your computer. If an arduino-cli.yaml file is located in any of those locations when you run `arduino-cli config init`, that file will be copied to your `.arduino15` folder instead of a default arduino-cli.yaml file being generated.

My suspicion is that you have a malformed arduino-cli.yaml file on your computer, which is being copied to your .arduino15 folder when you run `arduino-cli config init`. You can see the locations arduino-cli checks for the arduino-cli.yaml by running `arduino-cli --debug config init`. You'll see errors for every folder that didn't contain an arduino-cli.yaml file and no error when one was found. Please check that output and investigate whether you have a malformed arduino-cli.yaml in one of those folders which is the source of this error.

>What are ellipses?

This is an ellipsis: ...

itechnocrat, 2019-06-25:
I understand everything.
Thank you very much!

closed 2019-06-25 by per1234