arduino/arduino-cli #255: board list causes panic: runtime-error

Joshuao95, 2019-06-25:
## Bug Report
### Current behavior
arduino-cli board list fails to discover and has a runtime error

### Environment
arduino-cli version 0.3.6-alpha.preview
Installed from Arch package

```
INFO[0000] Checking for config file in: /home/josh/.arduino15/arduino-cli.yaml
INFO[0000] Reading configuration from /home/josh/.arduino15/arduino-cli.yaml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /usr/bin
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Checking for config in: /arduino-cli.yaml
INFO[0000] error loading                                 error="open /arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/josh/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/josh/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/josh/documents/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/josh/documents/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/josh/documents/work/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/josh/documents/work/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/josh/documents/work/genough/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/josh/documents/work/genough/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/josh/documents/work/genough/telematics/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/josh/documents/work/genough/telematics/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/josh/documents/work/genough/telematics/Freematics/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/josh/documents/work/genough/telematics/Freematics/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/josh/documents/work/genough/telematics/Freematics/firmware_v4/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/josh/documents/work/genough/telematics/Freematics/firmware_v4/arduino-cli.yaml: no such file or directory"
INFO[0000] Checking for config in: /home/josh/documents/work/genough/telematics/Freematics/firmware_v4/sim5360test/arduino-cli.yaml
INFO[0000] error loading                                 error="open /home/josh/documents/work/genough/telematics/Freematics/firmware_v4/sim5360test/arduino-cli.yaml: no such file or directory"
INFO[0000] Configuration set
INFO[0000] arduino-cli-0.3.6-alpha.preview
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Initializing package manager
INFO[0000] Loading hardware from: /home/josh/.arduino15/packages
INFO[0000] Loading package arduino from: /home/josh/.arduino15/packages/arduino/hardware
INFO[0000] Loaded platform                               platform="arduino:avr@1.6.21"
INFO[0000] Checking existence of 'tools' path: /home/josh/.arduino15/packages/arduino/tools
INFO[0000] Loading tools from dir: /home/josh/.arduino15/packages/arduino/tools
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.1.1"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@4.9.2-atmel3.5.4-arduino2"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino9"
Discovering.  2019/06/25 17:40:45 [ERR] bonjour: Failed to bind to udp6 port: listen udp6 [ff02::]:5353: socket: address family not supported by protocol
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x55fd598bc88f]

goroutine 21 [running]:
net.(*conn).ok(...)
        /usr/lib/go/src/net/net.go:168
net.(*UDPConn).SyscallConn(...)
        /usr/lib/go/src/net/udpsock.go:98
github.com/arduino/arduino-cli/vendor/golang.org/x/net/internal/socket.NewConn(0x55fd59cfbe60, 0x0, 0x0, 0x55fd59cfbe60, 0x0)
        src/github.com/arduino/arduino-cli/vendor/golang.org/x/net/internal/socket/rawconn.go:32 +0x9f
github.com/arduino/arduino-cli/vendor/golang.org/x/net/ipv6.NewPacketConn(0x55fd59cfb440, 0x0, 0xc000367e50)
        src/github.com/arduino/arduino-cli/vendor/golang.org/x/net/ipv6/endpoint.go:122 +0x5f
github.com/arduino/arduino-cli/vendor/github.com/oleksandr/bonjour.newClient(0x0, 0xc0003b2030, 0xc0003a0c70, 0x10)
        src/github.com/arduino/arduino-cli/vendor/github.com/oleksandr/bonjour/client.go:100 +0x1b8
github.com/arduino/arduino-cli/vendor/github.com/oleksandr/bonjour.NewResolver(0x0, 0x55fd59cda3a8, 0xc0003b0000, 0xc0003b0060)
        src/github.com/arduino/arduino-cli/vendor/github.com/oleksandr/bonjour/client.go:23 +0x31
github.com/arduino/arduino-cli/vendor/github.com/arduino/board-discovery.listEntries(0x0, 0x0, 0x0, 0x0, 0x0)
        src/github.com/arduino/arduino-cli/vendor/github.com/arduino/board-discovery/network.go:121 +0xdd
github.com/arduino/arduino-cli/vendor/github.com/arduino/board-discovery.(*Monitor).networkDiscover(0xc0003196a0, 0x0, 0x0)
        src/github.com/arduino/arduino-cli/vendor/github.com/arduino/board-discovery/network.go:95 +0x39
github.com/arduino/arduino-cli/vendor/github.com/arduino/board-discovery.(*Monitor).Start.func1.1(0xc000225ce0)
        src/github.com/arduino/arduino-cli/vendor/github.com/arduino/board-discovery/discovery.go:209 +0x34
github.com/arduino/arduino-cli/vendor/github.com/codeclysm/cc.Run.func1(0xc0000a2610, 0xc000241050, 0xc000225da0)
        src/github.com/arduino/arduino-cli/vendor/github.com/codeclysm/cc/cc.go:86 +0x39
created by github.com/arduino/arduino-cli/vendor/github.com/codeclysm/cc.Run
        src/github.com/arduino/arduino-cli/vendor/github.com/codeclysm/cc/cc.go:85 +0x129

```

masci, 2019-08-05:
@Joshuao95 can you try again using 0.4.0?

Joshuao95, 2019-08-08:
That works, thanks.

closed 2019-08-08 by Joshuao95