arduino/arduino-cli #257: go get does not work

wizzi86, 2019-06-27:
Hello, I am trying to install the arduino-cli using go get and it returns a lot of errors.
go version go1.7.4 linux/arm
System:
**PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"**

```
pi@hana-pi:~/go/bin $ go get github.com/arduino/arduino-cli
# google.golang.org/grpc/credentials/internal
../src/google.golang.org/grpc/credentials/internal/syscallconn.go:29: syntax error: unexpected = in type declaration
# github.com/arduino/arduino-cli/arduino/libraries
../src/github.com/arduino/arduino-cli/arduino/libraries/librarieslist.go:58: undefined: sort.Slice
# github.com/arduino/arduino-cli/configs
../src/github.com/arduino/arduino-cli/configs/directories.go:39: undefined: os.UserHomeDir
../src/github.com/arduino/arduino-cli/configs/directories.go:60: undefined: os.UserHomeDir
../src/github.com/arduino/arduino-cli/configs/preferences_txt_serializer.go:42: undefined: os.Executable
# google.golang.org/grpc/naming
../src/google.golang.org/grpc/naming/dns_resolver.go:41: undefined: net.DefaultResolver in net.DefaultResolver.LookupHost
../src/google.golang.org/grpc/naming/dns_resolver.go:42: undefined: net.DefaultResolver in net.DefaultResolver.LookupSRV
# google.golang.org/grpc/internal/syscall
../src/google.golang.org/grpc/internal/syscall/syscall_linux.go:79: tcpconn.SyscallConn undefined (type *net.TCPConn has no field or method SyscallConn)
../src/google.golang.org/grpc/internal/syscall/syscall_linux.go:100: tcpconn.SyscallConn undefined (type *net.TCPConn has no field or method SyscallConn)
# google.golang.org/grpc/resolver/dns
../src/google.golang.org/grpc/resolver/dns/dns_resolver.go:68: undefined: net.DefaultResolver
../src/google.golang.org/grpc/resolver/dns/dns_resolver.go:89: undefined: net.Resolver
# github.com/miekg/dns/vendor/golang.org/x/net/internal/socket
../src/github.com/miekg/dns/vendor/golang.org/x/net/internal/socket/rawconn.go:17: undefined: syscall.RawConn
# golang.org/x/net/http2
../src/golang.org/x/net/http2/server.go:214: h1.IdleTimeout undefined (type *http.Server has no field or method IdleTimeout)
../src/golang.org/x/net/http2/server.go:215: h1.IdleTimeout undefined (type *http.Server has no field or method IdleTimeout)
../src/golang.org/x/net/http2/server.go:220: s.RegisterOnShutdown undefined (type *http.Server has no field or method RegisterOnShutdown)
../src/golang.org/x/net/http2/server.go:2109: undefined: http.ErrAbortHandler
../src/golang.org/x/net/http2/server.go:2683: undefined: http.Pusher
../src/golang.org/x/net/http2/server.go:2685: undefined: http.PushOptions
../src/golang.org/x/net/http2/server.go:2697: undefined: http.PushOptions
../src/golang.org/x/net/http2/transport.go:497: undefined: http.NoBody
../src/golang.org/x/net/http2/transport.go:503: req.GetBody undefined (type *http.Request has no field or method GetBody)
../src/golang.org/x/net/http2/transport.go:505: req.GetBody undefined (type *http.Request has no field or method GetBody)
../src/golang.org/x/net/http2/transport.go:505: too many errors
# golang.org/x/net/internal/socket
../src/golang.org/x/net/internal/socket/rawconn.go:17: undefined: syscall.RawConn

```

cmaglie, 2019-06-27:
You must install the latest version of go.

jnorgan, 2019-07-11:
Was using go version go1.11 darwin/amd64, had the same issue.
Updated to go1.12.7 darwin/amd64, problem solved.

closed 2019-07-15 by masci

twillert02, 2020-08-15:
update to go1.15   same shit arduino-cli does not instal!!!
trying for days now..  it is sssssssooooo immature.

ubidefeo, 2020-08-15:
@twillert02 
I would like to invite you to use a more civilised language.
You have joined GitHub less than a month and your contributions add up to two complaints on our repos for two projects marked as **alpha**.

In order to submit an issue or a comment you should be so kind to list your Operating System and its version.
Please do not expect an **alpha** product to be _mature_, because you'll set yourself to be disappointed.