arduino/arduino-cli #258: Download specific release

hgrecco, 2019-06-27:
It does not seem to be a way now to download specific alpha releases (only the latest) which make it hard to use it certain times (For example, now the macos link is broken now). 

In the original [blog post](https://blog.arduino.cc/2018/08/24/announcing-the-arduino-command-line-interface-cli/) there are some specific links (which are broken now). Is there a repository of previous builds?

per1234, 2019-06-27:
Builds for every release are available on the Releases page of this repository:
https://github.com/arduino/arduino-cli/releases
just click the "Assets" link to see the build downloads for each OS.

closed 2019-06-27 by per1234

per1234, 2019-06-27:
You're welcome.

I found the problem with the macOS download link and have submitted a pull request to fix it (https://github.com/arduino/arduino-cli/pull/259). The correct download link is:
https://downloads.arduino.cc/arduino-cli/arduino-cli-latest-macosx.zip