arduino/arduino-cli #261: go get error

KMACEL, 2019-06-28:
## Bug Report
### Environment
- CLI version : arduino-cli Version: 0.3.7-alpha.preview Commit:  BuildDate: 2019-06-28 06:44:10.27600698 +0000 UTC
- Go version (if building from sources): go version go1.12.5 linux/amd64
- OS version : Ubuntu 16.04.6 LTS

### Expected behavior
"go get -u github.com/arduino/arduino-cli" error
if "GO111MODULE = on" gets the following error.

# github.com/cpuguy83/go-md2man/md2man
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/md2man.go:11:16: undefined: blackfriday.EXTENSION_NO_INTRA_EMPHASIS
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/md2man.go:12:16: undefined: blackfriday.EXTENSION_TABLES
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/md2man.go:13:16: undefined: blackfriday.EXTENSION_FENCED_CODE
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/md2man.go:14:16: undefined: blackfriday.EXTENSION_AUTOLINK
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/md2man.go:15:16: undefined: blackfriday.EXTENSION_SPACE_HEADERS
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/md2man.go:16:16: undefined: blackfriday.EXTENSION_FOOTNOTES
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/md2man.go:17:16: undefined: blackfriday.EXTENSION_TITLEBLOCK
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/md2man.go:19:29: too many arguments to conversion to blackfriday.Markdown: blackfriday.Markdown(doc, renderer, extensions)
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/roff.go:19:9: cannot use &roffRenderer literal (type *roffRenderer) as type blackfriday.Renderer in return argument:
	*roffRenderer does not implement blackfriday.Renderer (missing RenderFooter method)
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/roff.go:102:11: undefined: blackfriday.LIST_TYPE_ORDERED
go/pkg/mod/github.com/cpuguy83/go-md2man@v1.0.10/md2man/roff.go:102:11: too many errors

masci, 2019-07-17:
Hi @KMACEL 
the problem is upstream, I think you might have this same issue as https://github.com/golang/go/issues/32495 where you might have accidentally `go get` the wrong version of `https://github.com/russross/blackfriday`.

I've opened https://github.com/arduino/arduino-cli/pull/284 to consolidate the mod state but at glance it won't help you if you have blackfriday v2 installed. Can you try again from a clean state and let us know if things are different with these new info?

masci, 2019-07-24:
I'm closing since this seems to work now, feel free to reopen if you still have the issue

closed 2019-07-24 by masci

rogercoll, 2019-07-25:
There's a problem with `go-md2man@v1.0.10/md2man` library. I have also found this in different projects, hope is solved soon.