arduino/arduino-cli #263: [build][protoc] Compiling the protobuf definitions results in a dirty Git state

kittaakos, 2019-07-04:
## Bug Report

### Current behavior

Running [`protoc --proto_path=rpc --go_out=plugins=grpc,paths=source_relative:rpc ./rpc/commands/*.proto`](https://github.com/kittaakos/arduino-cli/blob/519d87a8dbb23c711c4df77b927e9878cb5ce693/Taskfile.yml#L7) results in a dirty Git state. I have outging changes in the generated [`rpc/commands/commands.pb.go`](https://github.com/arduino/arduino-cli/blob/c2d9c1b96fe996eee07ca7e5c9a9f4fce7872fd6/rpc/commands/commands.pb.go) file. Is this expected?

```diff
diff --git a/rpc/commands/commands.pb.go b/rpc/commands/commands.pb.go
index 49bf146..5c857d1 100644
--- a/rpc/commands/commands.pb.go
+++ b/rpc/commands/commands.pb.go
@@ -8,8 +8,6 @@ import (
 	fmt "fmt"
 	proto "github.com/golang/protobuf/proto"
 	grpc "google.golang.org/grpc"
-	codes "google.golang.org/grpc/codes"
-	status "google.golang.org/grpc/status"
 	math "math"
 )
 
@@ -1305,83 +1303,6 @@ type ArduinoCoreServer interface {
 	LibraryList(context.Context, *LibraryListReq) (*LibraryListResp, error)
 }
 
-// UnimplementedArduinoCoreServer can be embedded to have forward compatible implementations.
-type UnimplementedArduinoCoreServer struct {
-}
-
-func (*UnimplementedArduinoCoreServer) Init(req *InitReq, srv ArduinoCore_InitServer) error {
-	return status.Errorf(codes.Unimplemented, "method Init not implemented")
-}
-func (*UnimplementedArduinoCoreServer) Destroy(ctx context.Context, req *DestroyReq) (*DestroyResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method Destroy not implemented")
-}
-func (*UnimplementedArduinoCoreServer) Rescan(ctx context.Context, req *RescanReq) (*RescanResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method Rescan not implemented")
-}
-func (*UnimplementedArduinoCoreServer) UpdateIndex(req *UpdateIndexReq, srv ArduinoCore_UpdateIndexServer) error {
-	return status.Errorf(codes.Unimplemented, "method UpdateIndex not implemented")
-}
-func (*UnimplementedArduinoCoreServer) UpdateLibrariesIndex(req *UpdateLibrariesIndexReq, srv ArduinoCore_UpdateLibrariesIndexServer) error {
-	return status.Errorf(codes.Unimplemented, "method UpdateLibrariesIndex not implemented")
-}
-func (*UnimplementedArduinoCoreServer) Version(ctx context.Context, req *VersionReq) (*VersionResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method Version not implemented")
-}
-func (*UnimplementedArduinoCoreServer) BoardDetails(ctx context.Context, req *BoardDetailsReq) (*BoardDetailsResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method BoardDetails not implemented")
-}
-func (*UnimplementedArduinoCoreServer) BoardAttach(req *BoardAttachReq, srv ArduinoCore_BoardAttachServer) error {
-	return status.Errorf(codes.Unimplemented, "method BoardAttach not implemented")
-}
-func (*UnimplementedArduinoCoreServer) BoardList(ctx context.Context, req *BoardListReq) (*BoardListResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method BoardList not implemented")
-}
-func (*UnimplementedArduinoCoreServer) BoardListAll(ctx context.Context, req *BoardListAllReq) (*BoardListAllResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method BoardListAll not implemented")
-}
-func (*UnimplementedArduinoCoreServer) Compile(req *CompileReq, srv ArduinoCore_CompileServer) error {
-	return status.Errorf(codes.Unimplemented, "method Compile not implemented")
-}
-func (*UnimplementedArduinoCoreServer) PlatformInstall(req *PlatformInstallReq, srv ArduinoCore_PlatformInstallServer) error {
-	return status.Errorf(codes.Unimplemented, "method PlatformInstall not implemented")
-}
-func (*UnimplementedArduinoCoreServer) PlatformDownload(req *PlatformDownloadReq, srv ArduinoCore_PlatformDownloadServer) error {
-	return status.Errorf(codes.Unimplemented, "method PlatformDownload not implemented")
-}
-func (*UnimplementedArduinoCoreServer) PlatformUninstall(req *PlatformUninstallReq, srv ArduinoCore_PlatformUninstallServer) error {
-	return status.Errorf(codes.Unimplemented, "method PlatformUninstall not implemented")
-}
-func (*UnimplementedArduinoCoreServer) PlatformUpgrade(req *PlatformUpgradeReq, srv ArduinoCore_PlatformUpgradeServer) error {
-	return status.Errorf(codes.Unimplemented, "method PlatformUpgrade not implemented")
-}
-func (*UnimplementedArduinoCoreServer) Upload(req *UploadReq, srv ArduinoCore_UploadServer) error {
-	return status.Errorf(codes.Unimplemented, "method Upload not implemented")
-}
-func (*UnimplementedArduinoCoreServer) PlatformSearch(ctx context.Context, req *PlatformSearchReq) (*PlatformSearchResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method PlatformSearch not implemented")
-}
-func (*UnimplementedArduinoCoreServer) PlatformList(ctx context.Context, req *PlatformListReq) (*PlatformListResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method PlatformList not implemented")
-}
-func (*UnimplementedArduinoCoreServer) LibraryDownload(req *LibraryDownloadReq, srv ArduinoCore_LibraryDownloadServer) error {
-	return status.Errorf(codes.Unimplemented, "method LibraryDownload not implemented")
-}
-func (*UnimplementedArduinoCoreServer) LibraryInstall(req *LibraryInstallReq, srv ArduinoCore_LibraryInstallServer) error {
-	return status.Errorf(codes.Unimplemented, "method LibraryInstall not implemented")
-}
-func (*UnimplementedArduinoCoreServer) LibraryUninstall(req *LibraryUninstallReq, srv ArduinoCore_LibraryUninstallServer) error {
-	return status.Errorf(codes.Unimplemented, "method LibraryUninstall not implemented")
-}
-func (*UnimplementedArduinoCoreServer) LibraryUpgradeAll(req *LibraryUpgradeAllReq, srv ArduinoCore_LibraryUpgradeAllServer) error {
-	return status.Errorf(codes.Unimplemented, "method LibraryUpgradeAll not implemented")
-}
-func (*UnimplementedArduinoCoreServer) LibrarySearch(ctx context.Context, req *LibrarySearchReq) (*LibrarySearchResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method LibrarySearch not implemented")
-}
-func (*UnimplementedArduinoCoreServer) LibraryList(ctx context.Context, req *LibraryListReq) (*LibraryListResp, error) {
-	return nil, status.Errorf(codes.Unimplemented, "method LibraryList not implemented")
-}
-
 func RegisterArduinoCoreServer(s *grpc.Server, srv ArduinoCoreServer) {
 	s.RegisterService(&_ArduinoCore_serviceDesc, srv)
 }
```

### Expected behavior

The Git repository state is clean. (`git status -s`)

### Environment

- CLI version (output of `arduino-cli version`):
- Go version (if building from sources):
```
go version && protoc --version
go version go1.12.6 darwin/amd64
libprotoc 3.7.1
```
- OS version:
macOS `10.14.4 (18E226)`

### Additional context

I am currently on 69d86a4.

masci, 2019-07-04:
Hi @kittaakos thanks for the detailed report.

I can confirm it's not expected but I've generated the Go modules with `protoc` 3.8.0 (and the same version [is used in the test pipeline](https://github.com/arduino/arduino-cli/blob/master/Dockerfiles/CI/Dockerfile#L8). Can you try upgrading `protoc` to 3.8 to see if the behaviour is the same?

kittaakos, 2019-07-05:
Thank you for the hints, @masci.

 - I wiped `protobuf` from [`brew`](https://brew.sh), as the [latest is the `3.7.1`](https://formulae.brew.sh/formula/protobuf) version.
 - I manually downloaded `3.8.0` from [here](https://github.com/protocolbuffers/protobuf/releases/download/v3.8.0/protoc-3.8.0-osx-x86_64.zip).
 - I have copied the `protoc` binary to `/usr/local/bin/` and also put the `google` folder under `/usr/locale/include/` as suggested in the `readme.txt` from the downloaded archive.
 - Unfortunately, I still have outgoinng changes.

I did the followings:
```
Akoss-MacBook-Pro:arduino-cli akos.kitta$ git rev-parse --short HEAD
4a8523b
Akoss-MacBook-Pro:arduino-cli akos.kitta$ which protoc
/usr/local/bin/protoc
Akoss-MacBook-Pro:arduino-cli akos.kitta$ protoc --version
libprotoc 3.8.0
Akoss-MacBook-Pro:arduino-cli akos.kitta$ git status -s
Akoss-MacBook-Pro:arduino-cli akos.kitta$ protoc --proto_path=rpc --go_out=plugins=grpc,paths=source_relative:rpc ./rpc/commands/*.proto
Akoss-MacBook-Pro:arduino-cli akos.kitta$ git status -s
 M rpc/commands/commands.pb.go
Akoss-MacBook-Pro:arduino-cli akos.kitta$ 
```

masci, 2019-07-17:
@kittaakos can you try to do `go get -u github.com/golang/protobuf/protoc-gen-go` and try again? There was a bug in the service generator on OSX, your library might be affected despite the `protoc` executable is brand new.

kittaakos, 2019-07-17:
@masci, thank you for the update. It works as expected.

I did the followings for the verification:
```
Akoss-MacBook-Pro:~ akos.kitta$ cd ~
Akoss-MacBook-Pro:~ akos.kitta$ go get -u github.com/golang/protobuf/protoc-gen-go
Akoss-MacBook-Pro:~ akos.kitta$ cd ~/git/arduino-cli/
Akoss-MacBook-Pro:arduino-cli akos.kitta$ git rev-parse --short HEAD
1628b39
Akoss-MacBook-Pro:arduino-cli akos.kitta$ git status -s
Akoss-MacBook-Pro:arduino-cli akos.kitta$ protoc --proto_path=rpc --go_out=plugins=grpc,paths=source_relative:rpc ./rpc/commands/*.proto
Akoss-MacBook-Pro:arduino-cli akos.kitta$ git status -s
Akoss-MacBook-Pro:arduino-cli akos.kitta$ 
```

I also deleted a couple of `go` files under the `rpc/commands` folder and ran the generation. The git status was clean.

I am closing this as the generation does not result in outgoing git changes.

closed 2019-07-17 by kittaakos