arduino/arduino-cli #270: snap[Feature Request]

mikeroyal, 2019-07-10:
I was wondering if you might have an interest in a [snap](https://snapcraft.io/) version. I would help make the snap and submit a PR if there's any interest.
[Crafting your first snap](https://tutorials.ubuntu.com/tutorial/create-your-first-snap#0)

gvarisco, 2019-07-11:
This is something we have been discussing internally (cc @mastrolinux @endorama) and definitely worth exploring. Would you mind submitting a PR for it? We're happy to give it a try!

endorama, 2019-07-12:
Hi @mikeroyal, I'm very interested in a snap, tag me in the PR so that I can have a look and review it :)

What type of [confinement](https://docs.snapcraft.io/snap-confinement) do you plan to set if the snap?

mikeroyal, 2019-07-12:
Hi @gvarisco and @endorama, I'll start working on it this weekend.

masci, 2019-07-17:
PR is here https://github.com/arduino/arduino-cli/pull/281

umbynos, 2022-10-12:
https://github.com/arduino/arduino-cli/pull/281#issuecomment-1209195561

closed 2022-10-12 by umbynos