arduino/arduino-cli #271: Unknown board, which core should I install?

endorama, 2019-07-10:
Hello, I have a Arduino Nano 33 IoT, I attach it to the computer and it shows in the board list:
```
$ arduino-cli board list
Port         Type              Board Name FQBN
/dev/ttyACM1 Serial Port (USB) Unknown
```
I suspect this is due to a missing core. Core list:
```
arduino-cli core list
ID           Installed Latest Name                                        
arduino:avr  1.6.23    1.6.23 Arduino AVR Boards                          
arduino:samd 1.6.22    1.6.22 Arduino SAMD Boards (32-bits ARM Cortex-M0+)
```

I tried searching cores for `nano`, `33`, `iot`, I installed the core found (`arduino:avr`) but still the board is unknown.

Which core should I install? How should I know the core to install?

Thank you!

mastrolinux, 2019-07-11:
by reading VID and PID we should check in the vid:pid arduino API and propose to install the right core. 
The arduino-cli should install at least the serial and the usb discoverer by default. The logic is already here:
https://github.com/arduino/arduino-cli/blob/master/arduino/cores/packagemanager/package_manager.go#L93
@endorama are you sure you attached a board with a proper boot loader?

endorama, 2019-07-11:
> @endorama are you sure you attached a board with a proper boot loader?

I would say yes, as I was able to flash on it the Blink sketch from Create Editor.

Create Editor does recognise it properly, performing this request:
```
$ curl 'https://builder.arduino.cc/v3/boards/byVidPid/0x2341/0x8057' -H 'Authorization: Bearer [REDACTED]'
{"architecture":"samd","fqbn":"arduino:samd:nano_33_iot","href":"/v3/boards/arduino:samd:nano_33_iot","id":"nano_33_iot","name":"Arduino NANO 33 IoT","package":"arduino","plan":"create-free"}
```

endorama, 2019-07-11:
Upgrading to the latest `samd` core fixed this issue. Core index was outaded, so I missed the possible new version.

Use `arduino-cli core update-index` to get update core list.

After update I discovered a new `samd` version:
```
arduino-cli core list
ID           Installed Latest Name                                        
arduino:avr  1.6.23    1.6.23 Arduino AVR Boards                          
arduino:samd 1.6.22    1.8.3  Arduino SAMD Boards (32-bits ARM Cortex-M0+)
```

Trying to upgrade with `arduino-cli core upgrade` resulted in issue #273 

Single core update did work (`arduino-cli core upgrade arduino:samd`)

With `samd` core `1.8.3` the board is properly recognised:
```
$ arduino-cli board list
Port         Type              Board Name          FQBN                    
/dev/ttyACM0 Serial Port (USB) Arduino NANO 33 IoT arduino:samd:nano_33_iot
```

Searching for `33` now yield proper result:
```
$ arduino-cli core search 33
Searching for platforms matching '33'
ID           Version Name                                        
arduino:samd 1.8.3   Arduino SAMD Boards (32-bits ARM Cortex-M0+)
```

This issue is resolved, but I think we should improve this interaction. Probably a line suggesting to perform `update-index` may be enough (I did not read the entire getting started in the main README, but went directly for [step 4](https://github.com/arduino/arduino-cli#step-4-find-and-install-the-right-core), while this information was in step 3)

PR #274 (2019-07-11): improve find FQBN FAQ
As reported in multiple issues (#138 #191 #271) finding the right FQBN for a board can be difficult.

Improving the documentation may help users to find the relevant information.

Specifically, this PR closes #271 and add little improvement as suggested in #191.
commit 24fe3c61 (2019-07-11): improve find fqbn FAQ (#274)

As reported in multiple issues (#138 #191 #271) finding the right FQBN for a board can be difficult.

Improving the documentation may help users to find the relevant information.

Specifically, this PR closes #271 and add little improvement as suggested in #191.
closed 2019-07-11 by rsora