arduino/arduino-cli #275: [Feature request] include in homebrew/linuxbrew (OSX/Linux)

endorama, 2019-07-12:
Would be nice to have the cli released through homebrew/linuxbrew also.

Efrat19, 2019-09-25:
Coming soon... 🤞 
https://github.com/Homebrew/homebrew-core/pull/44575

Efrat19, 2019-09-25:
Coming soon... 🤞

endorama, 2019-10-07:
@Efrat19 thank you! I've just installed it, and everything went perfectly on Linux. 

@gvarisco can you please give it a go under MacOSX?

gvarisco, 2019-10-07:
As a reference, PR for arduino-cli inclusion in homebrew is https://github.com/Homebrew/homebrew-core/pull/44948 and got merged in https://github.com/Homebrew/homebrew-core/commit/cdf88f38253db3a05a4c6ef2f075a53a6c1d04d6

Will let you know once I run a couple of tests.

dawidd6, 2019-10-07:
Could someone confirm that running `brew test arduino-cli` does not produce an exception on macOS?

Running this command on linux results in an exeception as `arduino-cli` writes the new sketch in `~/Arduino/test_sketch` and in the formula test definition it is actually `~/Documents/Arduino/test_sketch` ($HOME is set to testpath during test).

Efrat19, 2019-10-07:
@dawidd6  as far as I saw this was fixed at arduino-cli0.5.0-showports, but you preferred not to use that version because it was a pre-release

dawidd6, 2019-10-07:
@Efrat19 I see, thanks. Actually I was in favor of this `-showports` release, but anyway I workarounded it on linux for now.

masci, 2019-10-22:
Closing as fixed by https://formulae.brew.sh/formula/arduino-cli

closed 2019-10-22 by masci