arduino/arduino-cli #276: Possible outdated nightly binary

kittaakos, 2019-07-12:
## Bug Report
Can you please verify whether the downloadable `nightly` is built from a17ecb9?

### Current behavior
The problem is the following:
 - I downloaded the nightly from macOS from this link: https://downloads.arduino.cc/arduino-cli/nightly/arduino-cli-nightly-latest-macosx.zip.
 - Start the daemon and initialize it.
 - When I execute `grpcurl -d '{"instance":{"id":1}}' -import-path ./rpc/ -proto commands/commands.proto --plaintext localhost:50051 cc.arduino.cli.commands.ArduinoCore.PlatformSearch` from the `arduino/arduino-cli` repository root, I get back the following results:
```json
{
  "searchOutput": [
    {
      "ID": "Intel:i586",
      "Installed": "1.6.7+1.0",
      "Latest": "Intel i586 Boards"
    },
    {
      "ID": "Intel:i686",
      "Installed": "1.6.7+1.0",
      "Latest": "Intel i686 Boards"
    },
    {
      "ID": "Intel:arc32",
      "Installed": "2.0.4",
      "Latest": "Intel Curie Boards"
    },
    {
      "ID": "atmel-avr-xminis:avr",
      "Installed": "0.6.0",
      "Latest": "Atmel AVR Xplained-minis"
    },
    {
      "ID": "littleBits:avr",
      "Installed": "1.0.0",
      "Latest": "littleBits Arduino AVR Modules"
    },
    {
      "ID": "Microsoft:win10",
      "Installed": "1.1.2",
      "Latest": "Windows 10 Iot Core"
    },
    {
      "ID": "Arrow:samd",
      "Installed": "2.1.0",
      "Latest": "Arrow Boards"
    },
    {
      "ID": "emoro:avr",
      "Installed": "3.2.2",
      "Latest": "EMORO 2560"
    },
    {
      "ID": "arduino:avr",
      "Installed": "1.6.23",
      "Latest": "Arduino AVR Boards"
    },
    {
      "ID": "arduino:megaavr",
      "Installed": "1.8.2",
      "Latest": "Arduino megaAVR Boards"
    },
    {
      "ID": "arduino:sam",
      "Installed": "1.6.12",
      "Latest": "Arduino SAM Boards (32-bits ARM Cortex-M3)"
    },
    {
      "ID": "arduino:samd",
      "Installed": "1.8.3",
      "Latest": "Arduino SAMD Boards (32-bits ARM Cortex-M0+)"
    },
    {
      "ID": "arduino:samd_beta",
      "Installed": "1.6.25",
      "Latest": "Arduino SAMD Beta Boards (32-bits ARM Cortex-M0+)"
    },
    {
      "ID": "arduino:stm32f4",
      "Installed": "1.0.1",
      "Latest": "Arduino STM32F4 Boards"
    },
    {
      "ID": "arduino:nrf52",
      "Installed": "1.0.2",
      "Latest": "Arduino nRF52 Boards"
    },
    {
      "ID": "industruino:samd",
      "Installed": "1.0.1",
      "Latest": "Industruino SAMD Boards (32-bits ARM Cortex-M0+)"
    }
  ]
}
```

However, when I get the source (a17ecb9) and build the `arduino-cli` locally, start the daemon and execute the same command I get back different result:
```json
{
  "searchOutput": [
    {
      "ID": "arduino:sam",
      "Latest": "1.6.12",
      "Name": "Arduino SAM Boards (32-bits ARM Cortex-M3)",
      "Maintainer": "Arduino",
      "Website": "http://www.arduino.cc/",
      "Email": "packages@arduino.cc"
    },
    {
      "ID": "arduino:samd",
      "Latest": "1.8.3",
      "Name": "Arduino SAMD Boards (32-bits ARM Cortex-M0+)",
      "Maintainer": "Arduino",
      "Website": "http://www.arduino.cc/",
      "Email": "packages@arduino.cc"
    },
    {
      "ID": "arduino:samd_beta",
      "Latest": "1.6.25",
      "Name": "Arduino SAMD Beta Boards (32-bits ARM Cortex-M0+)",
      "Maintainer": "Arduino",
      "Website": "http://www.arduino.cc/",
      "Email": "packages@arduino.cc"
    },
    {
      "ID": "arduino:stm32f4",
      "Latest": "1.0.1",
      "Name": "Arduino STM32F4 Boards",
      "Maintainer": "Arduino",
      "Website": "http://www.arduino.cc/",
      "Email": "packages@arduino.cc"
    },
    {
      "ID": "arduino:nrf52",
      "Latest": "1.0.2",
      "Name": "Arduino nRF52 Boards",
      "Maintainer": "Arduino",
      "Website": "http://www.arduino.cc/",
      "Email": "packages@arduino.cc"
    },
    {
      "ID": "arduino:avr",
      "Latest": "1.6.23",
      "Name": "Arduino AVR Boards",
      "Maintainer": "Arduino",
      "Website": "http://www.arduino.cc/",
      "Email": "packages@arduino.cc"
    },
    {
      "ID": "arduino:megaavr",
      "Latest": "1.8.2",
      "Name": "Arduino megaAVR Boards",
      "Maintainer": "Arduino",
      "Website": "http://www.arduino.cc/",
      "Email": "packages@arduino.cc"
    },
    {
      "ID": "Intel:arc32",
      "Latest": "2.0.4",
      "Name": "Intel Curie Boards",
      "Maintainer": "Intel",
      "Website": "http://maker.intel.com/",
      "Email": "support@intel.com"
    },
    {
      "ID": "Intel:i586",
      "Latest": "1.6.7+1.0",
      "Name": "Intel i586 Boards",
      "Maintainer": "Intel",
      "Website": "http://maker.intel.com/",
      "Email": "support@intel.com"
    },
    {
      "ID": "Intel:i686",
      "Latest": "1.6.7+1.0",
      "Name": "Intel i686 Boards",
      "Maintainer": "Intel",
      "Website": "http://maker.intel.com/",
      "Email": "support@intel.com"
    },
    {
      "ID": "littleBits:avr",
      "Latest": "1.0.0",
      "Name": "littleBits Arduino AVR Modules",
      "Maintainer": "littleBits Electronics",
      "Website": "http://www.littlebits.cc/bits/w6-arduino",
      "Email": "support@littlebits.cc"
    },
    {
      "ID": "Microsoft:win10",
      "Latest": "1.1.2",
      "Name": "Windows 10 Iot Core",
      "Maintainer": "Microsoft.IoT",
      "Website": "https://github.com/ms-iot/iot-utilities/tree/master/IotCoreAppDeployment/ArduinoIde/",
      "Email": "bfjelds@microsoft.com"
    },
    {
      "ID": "industruino:samd",
      "Latest": "1.0.1",
      "Name": "Industruino SAMD Boards (32-bits ARM Cortex-M0+)",
      "Maintainer": "Industruino",
      "Website": "https://industruino.com/",
      "Email": "connect@industruino.com"
    },
    {
      "ID": "atmel-avr-xminis:avr",
      "Latest": "0.6.0",
      "Name": "Atmel AVR Xplained-minis",
      "Maintainer": "Atmel University France",
      "Website": "https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/wiki"
    },
    {
      "ID": "Arrow:samd",
      "Latest": "2.1.0",
      "Name": "Arrow Boards",
      "Maintainer": "Axel Elettronica S.r.l.",
      "Website": "http://axelelettronica.it/",
      "Email": "development@axelelettronica.it"
    },
    {
      "ID": "emoro:avr",
      "Latest": "3.2.2",
      "Name": "EMORO 2560",
      "Maintainer": "Inovatic-ICT",
      "Website": "http://www.emoro.eu/shop/system/download/EMoRo2560_and_GLAM_brochure.pdf.0b83da8ad90db30722cca3a5d2529494",
      "Email": "support@inovatic-ict.com"
    }
  ]
}
```

Please note, the `Boards` are missing from the result from the local build as well. The issue leads [here](https://github.com/arduino/arduino-cli/pull/268/files#diff-4fe499fff4708a0e2f12285bfa64e2faR29); I want to tackle it in a separate issue.

### Expected behavior

It seams https://github.com/arduino/arduino-cli/pull/268 is not part of the downloadble `nightly`.

### Environment

- CLI version (output of `arduino-cli version`):
`arduino-cli-nightly-13` vs. local build.
- Go version (if building from sources):
`go version go1.12.6 darwin/amd64`
- OS version:
`macOS 10.14.4 (18E226)`

### Additional context

rsora, 2019-07-12:
Hi @kittaakos, 
Thanks for the heads up, I can confirm that the nightly builds were outdated, as you reported in your issue the result of `arduino-cli version` was:
`arduino-cli Version: nightly-13 Commit: 4a8523b BuildDate: 2019-07-12 09:31:39.666193 +0000 UTC`
Both the BuildDate and the and the Version were right but the commit was clearly outdated.
This led me to find remove an unwanted git ref filter in our CI that was causing a wrong git fetch.
You can now download a fresh (and updated to the latest commit from master) nightly build that should report:
`arduino-cli Version: nightly-18 Commit: 1628b39 BuildDate: 2019-07-12 10:24:51.971145978 +0000 UTC`
Can you please confirm that this solves your issue?

kittaakos, 2019-07-12:
I can confirm; the nightly works as expected.

Thank you for the prompt fix, @rsora! 🙏

closed 2019-07-12 by kittaakos