arduino/arduino-cli #285: Uninstall does not remove lib binaries

mastrolinux, 2019-07-17:
## Bug Report
Lib uninstall should not only remove libs from the index but also remove the corresponding binary in .arduino15/staging/libraries/ or at least have an option to avoid removing the lib binary. Default behaviour should be to remove the bin too.

### Current behavior
```
$ arduino-cli lib install 'WiFiNINA'
Downloading WiFiNINA@1.4.0...
WiFiNINA@1.4.0 already downloaded
Installing WiFiNINA@1.4.0...
Installed WiFiNINA@1.4.0
$ arduino-cli lib uninstall 'WiFiNINA'
Uninstalling WiFiNINA@1.4.0...
$ ls -l ~/.arduino15/staging/libraries/
total 232
-rw-r--r-- 1 luca luca  83192 Jul 15 14:49 WiFiNINA-1.4.0.zip
$ arduino-cli lib download 'WiFiNINA'
WiFiNINA@1.4.0 already downloaded
```

### Expected behavior
Uninstall should remove the lib file

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.3.7-alpha.preview Commit:  BuildDate: 2019-07-17 10:37:06.0191213 +0000 UTC
- Go version (if building from sources): 1.12
- OS version: Windows 10, Windows WSL

masci, 2019-07-29:
Proposal:
 * rename any reference to `staging` as `cache` so it's more clear for the user what that directory tree contains
 * provide a set of `cache` subcommands the user can run to free up some space when needed, something on the lines of `arduino-cli cache clean`

per1234, 2019-07-30:
I think it's a reasonable proposal, but there are a couple of things that should be considered:

---
Since https://github.com/arduino/Arduino/pull/9023, there is already a `.arduino15/cache` file in use by the Arduino IDE to store the cached Boards and Library Manager index files. The folder structure currently looks something like this:

```
.arduino15
|_cache
   |_arduino.esp8266.com
   |_downloads.arduino.cc
   |_raw.githubusercontent.com
   |_cache.json
```
So it might make sense to move the existing contents of the cache folder to a dedicated subfolder (something like `.arduino15/cache/indexes`), like what is currently done with libraries and packages in the `staging` folder. This will keep the folder organized nicely and prevent any chance of unwanted interactions between the different uses of the `cache` folder:

```
.arduino15
|_cache
   |_indexes
   |_libraries
   |_packages
```

---
The `.arduino15/staging` folder has been in use since Arduino IDE 1.6.4 so it's possible that changing the folder name might cause breakage of 3rd party applications. I don't have any evidence that anything is reliant on that specific path and can't think of a hypothetical case other than maybe some users having a custom script that clears it from time to time so likely this is a non-issue.

---
The library binaries in `.arduino15/staging/libraries` are unlikely to take up a significant amount of disk space, but the `.arduino15/staging/packages` folder can get pretty huge!

masci, 2019-07-30:
Thanks @per1234 for the insights, I think there's enough to rethink my approach, I'd rather avoid messing with folder names if this means breaking the old IDE, specially now that development iterations on the CLI are so fast we've no chance to keep up with the IDE.

Another option would be keep going with the `staging` terminology (albeit in all honesty the name is puzzling) and still give users the opportunity to clean `libraries` and `packages` with explicit commands like `staging clean`.

@mastrolinux thoughts?

cmaglie, 2019-07-30:
We can still use `cache clean` (or whatever) on the command line but keep `staging` as the internal folder name.

masci, 2019-09-26:
In case somebody wants to contribute, to sum up the discussion what's need to be done here is a new `cache clean` command that would wipe the contents of the `.arduino15/staging` folder.

HowJMay, 2020-01-08:
May I try this issue
?

masci, 2020-01-09:
@HowJMay yes please!

HowJMay, 2020-01-17:
I currently remove the default path of the caching file, but I think I may need a way out to remove caching file in other user-assigned path.
In order to do this, I think I may need to declare one more global variable to record the path of the caching file. I am still thinking if there is any other way out.

masci, 2020-02-17:
Fix released with 0.8.0, closing...

closed 2020-02-17 by masci