arduino/arduino-cli #292: Is this specific to a particular ARM version??

Zeus916, 2019-07-19:
## Bug Report
I downloaded the Linux ARM version and moved it over to /usr/local/bin
then i renamed the executable to arduino-cli.
Then i type the command "arduino-cli" and  i get "No such file or directory" error.

im not building it from source.
### Environment

- Nvidia Jetson Nano
- Go version (if building from sources):
- OS version: Ubuntu 18.04 on ARMv8

masci, 2019-07-19:
Can you double check file permissions for `/usr/local/bin/arduino-cli` with something like `ls -l/usr/local/bin`? The file `arduino-cli` has to be executable, if it's not, try with:
```
chmod +x /usr/local/bin/arduino-cli
```

facchinm, 2019-07-19:
I think this is due to ARM64 architecture; `No such file or directory` error normally happens when trying to run an executable for a different arch.
Are binaries for ARM64 (or aarch64) available in nightly? Do we simply need to add the proper links in README?

rsora, 2019-07-19:
Currently only ARM32 is available, since we merged https://github.com/arduino/arduino-cli/pull/204, in the next day I'll make the ARM64 binary available after configuring our CI pipeline to produce it.

Zeus916, 2019-07-23:
> I think this is due to ARM64 architecture; `No such file or directory` error normally happens when trying to run an executable for a different arch.
> Are binaries for ARM64 (or aarch64) available in nightly? Do we simply need to add the proper links in README?

closed 2019-07-23 by Zeus916

askkvn, 2019-10-15:
> ## Bug Report
>
> I downloaded the Linux ARM version and moved it over to /usr/local/bin
> then i renamed the executable to arduino-cli.
> Then i type the command "arduino-cli" and i get "No such file or [...]

I have tried same thing, now it not giving any error.

nestukh, 2020-09-27:
I'm having the same problem using the current arm64 build on a Vero 4K+ (by OSMC), which has Debian with aarch64 architecture.
The binary from the `arduino-cli-152-PR307-linuxarm64.tar.bz2` archive (mentioned in another issue) works.