arduino/arduino-cli #312: Add unit testing

user-name-is-taken, 2019-07-28:
This project should support unit testing as [arduino_ci](https://github.com/ianfixes/arduino_ci/tree/master/SampleProjects/DoSomething) does

user-name-is-taken, 2019-07-28:
This might be as simple as compiling for the host.

mastrolinux, 2019-07-29:
Hello, there are quite a bit of test: 
https://github.com/arduino/arduino-cli/tree/master/legacy/builder/test 
https://github.com/arduino/arduino-cli/blob/master/version/version_test.go
https://github.com/arduino/arduino-cli/blob/master/configs/navigate_test.go

and some integration tests too: 
https://github.com/arduino/arduino-cli/tree/master/test

Tests and builds are available here too:
https://cloud.drone.io/arduino/arduino-cli

We know it is not complete but we keep working on it. Feel free to propose your contribution to improving the coverage.

closed 2019-07-29 by mastrolinux

user-name-is-taken, 2019-07-30:
How could I use arduino-cli to test [my arduino project](https://gitlab.com/robotics_fun/DC_motors/blob/master/.gitlab-ci.yml)?