arduino/arduino-cli #313: No ability to pin download binary versions

petewarden, 2019-07-29:
## Bug Report

### Current behavior

- We're downloading the latest binary packages at https://downloads.arduino.cc/arduino-cli/arduino-cli-latest-linux64.tar.bz2 as part of the TensorFlow continuous integration system
- The contents of this package recently changed the name of the executable from arduino-cli-*linux64 binary to arduino-cli.
- This broke our tests.

### Expected behavior

- It should be possible to download binary packages at a particular version from a URL that is never overwritten, so we can pin our testing to a known-good version.

### Environment

- CLI version: N/A
- OS version: Ubuntu 16.04

### Additional context

The CI code we used to install the arduino-cli binary is at https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/experimental/micro/tools/ci_build/install_arduino_cli.sh
Our code for testing that our generated projects building using the CLI is at https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/experimental/micro/tools/ci_build/test_arduino_library.sh
We are able to work around this immediate issue with a change to the install script, but we'd like to make the process more robust to external changes, since we're using it to block all code submits if the Arduino libraries don't compile.

masci, 2019-08-02:
Starting with upcoming `0.4.0` you'll be able to pin versions of the supported packages. RC-1 is available at https://github.com/arduino/arduino-cli/releases/tag/0.4.0-rc1 we'll release on Monday 8/5 if we don't find any blocker while testing the rc.

masci, 2019-08-05:
I'm happy to announce `0.4.0` is out, please checkout the release page for the changelog and the binaries https://github.com/arduino/arduino-cli/releases/tag/0.4.0

closed 2019-08-05 by masci