arduino/arduino-cli #325: `core install` fails when Library/Arduino15 directory is a symlink

dandee, 2019-08-05:
## Bug Report

### Current behavior

Attempt to install a new board fails:

```
> arduino-cli core install arduino:stm32f4
Tool arduino:arm-none-eabi-gcc@4.8.3-2014q1 already installed
Downloading packages...
arduino:dfu-util@0.9.0-arduino1 already downloaded
arduino:arduinoSTM32load@2.0.0 already downloaded
arduino:stm32f4@1.0.1 already downloaded
Installing arduino:dfu-util@0.9.0-arduino1...
arduino:dfu-util@0.9.0-arduino1 installed
Installing arduino:arduinoSTM32load@2.0.0...
arduino:arduinoSTM32load@2.0.0 installed
Installing arduino:stm32f4@1.0.1...
Error: searching package root dir: no unique root dir in archive, found '/Users/mooncake/Library/Arduino15/tmp/package-683071466/cores' and '/Users/mooncake/Library/Arduino15/tmp/package-683071466/drivers'
Error during install
```

### Expected behavior

Board to be successfully installed.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli Version: 0.4.0 Commit: =98b7be9`
- Go version (if building from sources): `N/A`
- OS version: `macOS Mojave 10.14.6 (18G84)`

### Additional context

`/Users/mooncake/Library/Arduino15` is a symbolic link to a directory on another disk (volume?)

dandee, 2019-08-05:
List of boards before `core install`:

```
>arduino-cli core list
ID              Installed Latest Name
arduino:megaavr 1.8.3     1.8.3  Arduino megaAVR Boards
arduino:sam     1.6.12    1.6.12 Arduino SAM Boards (32-bits ARM Cortex-M3)
```

when `core install` failed I've had launched Arduino.app and installed "stm32f4" board through app's GUI, then quit the app. After that `core list` has given me this (with the same symlink in place as before):

```
>arduino-cli core list
ID              Installed Latest Name
arduino:megaavr 1.8.3     1.8.3  Arduino megaAVR Boards
arduino:sam     1.6.12    1.6.12 Arduino SAM Boards (32-bits ARM Cortex-M3)
arduino:stm32f4 1.0.1     1.0.1  Arduino STM32F4 Boards
```

masci, 2019-08-06:
This is the same as https://github.com/arduino/arduino-cli/issues/153

The problem is in the package layout; while the IDE performs some logic to address common problems like this one, the CLI hasn't anything like that. The fix should primarily consist in fixing the package but we're discussing internally whether we should do or not in the CLI what the IDE does. I personally think we shouldn't start adding exceptions or it'll be hard to draw a line when they'll become too many. 

Thoughts and feedback welcome.

dandee, 2019-08-06:
Oh, I didn't know it was a particular package problem, I assumed it's the tool's issue. I agree with you in such a case that adding an exception to the tool is not the best approach.

facchinm, 2019-08-07:
Hi @dandee ,
which board were you trying to target by installing `arduino:stm32f4` core? 

That core only contains StarOtto board from Arduino.org (which was never released AFAIK) so we just decided to remove it to avoid further confusion.

In case you need a generic package for STM32, the [community core](https://github.com/stm32duino/Arduino_Core_STM32#getting-started) is much much better

dandee, 2019-08-07:
Hi @facchinm ,

I've got a couple of NUCLEO-F411RE boards from STM to play with. Is the suggested package would work with it?

facchinm, 2019-08-07:
Yes, it's supported by the community package

dandee, 2019-08-07:
Great! Thanks for letting me know!

facchinm, 2019-08-08:
I'm closing the issue as wontfix since the behaviour is expected (and conforms with all the documentation we have).

closed 2019-08-08 by facchinm