arduino/arduino-cli #340: Requesting ARM6 build for Raspberry Pi Zero

Tatwi, 2019-08-09:
Your install.sh script indicated the following: 

"Sorry, we dont have a dist for your system: Linux ARM
You can request one here: https://github.com/Arduino/arduino-cli/issues"

As such, I would like to request a build for ARM6 that is compatible with the Raspberry Pi Zero (and Zero W). 

Thanks! :)

per1234, 2019-08-09:
Thanks for reporting this!

To tide you over until the installation script is fixed, there is a Raspberry Pi compatible version of arduino-cli available at the "arduino-cli_0.4.0_Linux_32bit.tar.gz" link here:
https://github.com/arduino/arduino-cli/releases/tag/0.4.0

masci, 2019-08-10:
Hi @Tatwi, at this stage we only distribute binaries for ARMv7 and later, this is something that might change if users ask for it but I'm not sure about implications right now. As a workaround, have you tried building the CLI yourself? The [contributing page](https://github.com/arduino/arduino-cli/blob/master/CONTRIBUTING.md#prerequisites) provides some guidance but feel free to reach out should you have any issue with the build.

Let's use this issue to collect feature requests, if anybody else needs an ARMv6 distribution please leave your feedback or upvote OP.

Tatwi, 2019-08-10:
Thanks masci. I'll build it myself when I have time to put together the toolchain (I don't use the software that you use, so I don't have it setup).

My thoughts on why having an ARM6 build would be beneficial are thus:

1. The Raspberry Pi Zero / Zero W will be manufactured and available until at least 2026.

2. Of all the Raspberry Pi products, the Pi Zero best synergies with Arduino products, being that it is often used in embedded systems, robotics, etc. It's less than half the size of an Arduino UNO and it integrates nicely with Arduino Nano devices.

3. Of all the Raspberry Pi products, the ARM6 based Pi Zero is the one that benefits the most from using the command line version of the Arduino compiler, because often the Pi Zero will be used either headless or command line only.

Personally I am just slapping my UNO together with my Zero for the sake of doing so, but I'm sure there are others out there who are enjoying the combination of computing power, small size, low power requirements, and ability to program multiple Arduinos on the fly ("Smart" robots for instance). I think it's worth doing, at least until the Pi Zero is no longer being sold as an ARM6 device.

facchinm, 2019-08-12:
@masci there should be no implications in moving the default ARM 32 bit build to ARMv6 (in fact, `go build` defaults to ARMv5 if I'm not wrong).
All the relevant tools (compilers, avrdude, ...) are already built for ARMv6 too so I'd switch the default target to make everyone happy :slightly_smiling_face:

masci, 2019-08-12:
@Tatwi I don't have a Pi Zero at hand, can you see if https://github.com/arduino/arduino-cli/releases/download/0.4.0-armv6/arduino-cli_0.4.0-armv6_Linux_ARMv6.tar.gz works? If it works I'll merge #341 and ARMv6 builds will be effective starting with the next nightly

Tatwi, 2019-08-12:
I should have some time to test it this week, Massimiliano. I'll let you
know how it goes.

gvarisco, 2019-08-13:
I gave it a try on armv6 (Linux raspberrypi 4.14.79 #1 Sun Jan 27 22:47:58 UTC 2019 armv6l, raspbian_lite). Works like a charm. No errors detected.

Tatwi, 2019-08-14:
I just finished testing https://github.com/arduino/arduino-cli/releases/download/0.4.0-armv6/arduino-cli_0.4.0-armv6_Linux_ARMv6.tar.gz on my Raspberry Pi Zero W (running Debian Buster) and it worked perfectly with my Arduino UNO. Thanks!

FYI for anyone interested: The test sketch took between 5-10 seconds to compile and update the UNO, using a class 10 SD. Might be faster using a RAM Drive for /tmp and anywhere else object code gets splatted during compilation.

closed 2019-08-14 by masci

raj-priyanshu, 2020-06-19:
I tried to install Arduino latest version from installation page of arduino-cli. but it was for arm v7. please release the latest version for arm v6 (for rpi zero w).

PR #341 (2019-08-14): Add support for ARM v6
Fixes #340

Nightly will be available in the archive `arduino-cli_nightly-20190812_Linux_ARMv6.tar.gz`