arduino/arduino-cli #344: Add network-discovery support

cmaglie, 2019-08-13:
Network discovery tool: https://github.com/arduino/mdns-discovery

This should allow OTA and is required to implement #69

thekunalsaini, 2020-02-24:
Hi to all,
I m Kunal Saini . I like to work with you in this project.
What I have planned is given below :

# Implementation
We can attach any WiFi module or bluetooth module or mount ethernet on arduino and then write a functionality through which it can appear in network.

Functionality helps to make available device in network.
Power is given through Battery or any other source.
This is my idea how we can do it.

Thank you

thekunalsaini, 2020-02-24:
@cmaglie Any suggestion regarding implementation most welcome

rtek1000, 2021-04-26:
Hello guys, I don't want to be too boring, but it will be difficult for people to test this IDE 2.0 if they don't have the network port.

Do you have any expectations of being able to add this functionality?

Do you have any priority of adding this network port?

Thank you.

kmorris1077, 2021-04-30:
Not having the network port capability is a total deal breaker for many of us with esp32s in field installations who want to use the 2.0 IDE.

ubidefeo, 2021-05-02:
@kmorris1077 
we are in the peocess of closing our RFC for pluggable discovery.
At that point it would be up to platform implementers to write their discovery layer

StarMiner99, 2021-05-08:
im also not using the arduino 2 ide because i cant programm my esp-01s without cables and an ftdi adapter its to much work to connect it every time.

closed 2021-08-23 by cmaglie