arduino/arduino-cli #348: install.sh fails : eval: Syntax error: "(" unexpected

robbeofficial, 2019-08-14:
## Bug Report

running:
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh

yields:
Installing in /tmp/bin
ARCH=64bit
OS=Linux
Using curl as download tool
Getting https://api.github.com/repos/arduino/arduino-cli/releases/latest
sh: 316: eval: Syntax error: "(" unexpected
Failed to install arduino-cli

### Environment
Ubuntu 18.04.3 LTS

masci, 2019-08-14:
@robbeofficial it's failing for me too, can you try with `curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/massi/noapi/install.sh | sh` and let me know if that works?

fredericmorin-eb, 2019-08-14:
I'm experiencing the same issue @robbeofficial is having but I'm on OSX.
@masci the install script made it further while using your branch but ended with an error still.

```
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/massi/noapi/install.sh | sh
Installing in /Users/fred/qqq/bin
ARCH=64bit
OS=macOS
Using curl as download tool
TAG=0.4.0
CLI_DIST=arduino-cli_0.4.0_macOS_64bit.tar.gz
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.4.0_macOS_64bit.tar.gz
arduino-cli not found. Did you add /Users/fred/qqq/bin to your $PATH?
Failed to install arduino-cli
```

ilesar, 2019-08-14:
@fredericmorin-eb You can install the version which @masci mentioned in the comment by adding the `/Users/fred/qqq/bin` folder to your $PATH or executing the same command with the BINDIR parameter and setting it to a folder which already is in your $PATH.

```
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/massi/noapi/install.sh | BINDIR=/usr/local/bin sh
Installing in /usr/local/bin
ARCH=64bit
OS=macOS
Using curl as download tool
TAG=0.4.0
CLI_DIST=arduino-cli_0.4.0_macOS_64bit.tar.gz
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.4.0_macOS_64bit.tar.gz
arduino-cli Version: 0.4.0 Commit: =98b7be9 installed successfully
```

fredericmorin-eb, 2019-08-14:
Argggg, yup. got it. Sorry for the noise. Failed to read the error message in its entirety

masci, 2019-08-15:
Thanks for confirming it works folks!

@fredericmorin-eb indeed the message is a bit too aggressive I think, the fact that the binary didn't end up in a folder within the PATH doesn't prevent the CLI to work properly, probably it should be a more gentle warning

surfsoft, 2019-08-17:
> @robbeofficial it's failing for me too, can you try with `curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/massi/noapi/install.sh | sh` and let me know if that works?

This also worked for me on MacOs 10.14.6, I got `sh: eval: line 403: syntax error near unexpected token '('` when trying to use the version on master.

masci, 2019-08-19:
I've merged on `master` the fix you guys tested, thanks for the help!
Closing for now, feel free to re-open if you keep getting the syntax error.

closed 2019-08-19 by masci