arduino/arduino-cli #353: gRPC client example was converted to a test but that is not what it was intended for

per1234, 2019-08-21:
https://github.com/arduino/arduino-cli/commit/c2d9c1b96fe996eee07ca7e5c9a9f4fce7872fd6 converted the [gRPC client example program](https://github.com/arduino/arduino-cli/blob/master/commands/daemon/client_test.go) into a test. As @cmaglie stated on Slack, this file was intended to be an example of how to call the gRPC API. I think this is an important component of the gRPC interface documentation and the conversion to a test makes it less useful as an example. For this reason, I would recommend reverting that part of https://github.com/arduino/arduino-cli/commit/c2d9c1b96fe996eee07ca7e5c9a9f4fce7872fd6.

masci, 2019-08-23:
The example code was buried in the repo, if we want that to be a piece of documentation we need to find a better solution than reverting the commit. I'll take care of it.

masci, 2019-08-27:
The idea is to provide examples for the same workflow in different languages but if we go down that path I wouldn't clutter this repo with issues, PRs and ultimately code that's related to an example client we only provide for documentation purposes. This is a personal opinion though, so I guess the last word in this regard should come from the project maintainer.

If we want to keep things simple, I can easily move back here the `main.go` I'm working on under  a `client_example` folder at the root of the repo but note we have no tooling or best practice in place to keep the example code aligned with the gRPC Api, so I wouldn't expect any tangible improvement on that side.

masci, 2019-08-27:
> The documentation should be considered an integral and essential part of the project. Issues and PRs about the documentation are not clutter and this repository is the appropriate place for them.

This is debatable but I do agree docs should stay close to the code when you want devs to help with keeping them up to date. 

I should be done with polishing the client example soon, I'll move it here and archive the other repo to avoid confusion.

PR #361 (2019-08-28): [skip changelog] Revert the gRPC client example to a runnable go program
Fixes #353
closed 2019-08-28 by masci