arduino/arduino-cli #358: Null pointer if target .ino is a folder

facchinm, 2019-08-26:
## Bug Report

### Current behavior
`./arduino-cli --debug compile -b anything:something:meh ~/Arduino/Blink` crashes with null pointer dereference if `~/Arduino/Blink` contains a **folder** called `Blink.ino`

### Expected behavior

Malformed ino files should be recognized and skipped.

### Environment

- CLI version (output of `arduino-cli version`): nightly-4-gcb3c79b 
- OS and platform: linux x86_64

### Additional context

shade34321, 2019-09-25:
Has this already been worked on? If not I'd like to try it.

masci, 2019-09-26:
@shade34321 all yours!

shade34321, 2019-09-28:
Sweet, Thanks!

masci, 2019-10-01:
@shade34321 FYI it seems #42 indirectly fixes this issue

d-a-v, 2019-10-01:
#42 or #421 ?

masci, 2019-10-01:
@d-a-v 421, autocompletion tricked me :)

shade34321, 2019-10-04:
Oh ok. Guess I won't need to work on this. I'll see if I can find something else easy to work on. Thanks!

PR #421 (2019-10-29): Use a local Walk function
Because arduino-cli segfaults when sketchDir is a symbolic link,
a simple walk function is introduced with a twofold effect:
- fix the segfault
- allow to use symlinks

ref: https://www.google.com/search?q=golang+Walk+does+not+follow+symbolic+links

*edit*:
* fixes #424 
* fixes #358
closed 2019-10-29 by rsora