arduino/arduino-cli #362: Libraries installation/uninstallation name inconcistency

Renaud11232, 2019-08-27:
## Bug Report

The name to use when installing or uninstalling libraries is inconsistent

### Current behavior

Installing libraries work as intented. ie:

```
PS C:\Users\gaspa\Downloads> .\arduino-cli.exe lib install "LiquidCrystal I2C"
Downloading LiquidCrystal I2C@1.1.2...
LiquidCrystal I2C@1.1.2 already downloaded
Installing LiquidCrystal I2C@1.1.2...
Installed LiquidCrystal I2C@1.1.2
```
But trying to remove the library using the name used when installing, does not always remove the library.
```
PS C:\Users\gaspa\Downloads> .\arduino-cli.exe lib uninstall "LiquidCrystal I2C"
Library LiquidCrystal I2C is not installed
```
In this case I had to replace the space with a `_`
```
PS C:\Users\gaspa\Downloads> .\arduino-cli.exe lib uninstall "LiquidCrystal_I2C"
Uninstalling LiquidCrystal_I2C@1.1.2...
```

### Expected behavior

The libraries should be installed and uninstalled using the same name

### Environment

#### CLI version
```
PS C:\Users\gaspa\Downloads> .\arduino-cli.exe version
arduino-cli.exe Version: 0.4.0 Commit: =98b7be9
```
#### OS/Platform

Windows 10 Pro build 1903

HowJMay, 2019-09-21:
If possible, could I try this one?

masci, 2019-09-22:
@HowJMay all yours! :smile: 
Thanks!

HowJMay, 2019-09-22:
My pleasure

HowJMay, 2019-09-27:
@masci May I ask that whether all the library with the following link can be downloaded with arduino-cli ?
https://www.arduinolibraries.info/

masci, 2019-09-27:
@HowJMay http://downloads.arduino.cc/libraries/library_index.json contains all the libraries that can be installed out of the box with the CLI, then users can add any number of additional json indexes listing their own "unofficial" libraries.

cmaglie, 2019-09-30:
> May I ask that whether all the library with the following link can be downloaded with arduino-cli ?
https://www.arduinolibraries.info/

AFAIK this website takes the list from our library_index.json so: yes.

PR #443 (2019-12-11): Fix "lib uninstall" when library name contains spaces
Now,  the "_" is replaced with a blanket " ", to reslove the
inconsistency problem. However, use the name of the installing
library may be a better solution.

fixes #362 

BTW, when we uninstall libraries, there are no messages to notify the user libraries are successfully uninstalled.
However, when we install libraries message like `Installed LiquidCrystal I2C@1.1.2` is printed to notify  libraries are successfully installed.
closed 2019-12-11 by masci