arduino/arduino-cli #385: config dump as json outputs  text

piif, 2019-08-31:
## Bug Report

### Current behavior
arduino-cli config dump --format json

outputs config as text :
```
proxy_type: auto
sketchbook_path: /...
```

### Expected behavior
should output valid json like :
```
{
  "proxy_type": "auto",
  "sketchbook_path": "...",
 ...
}
```
### Environment

- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.3.7-alpha.preview Commit: dd6f633
- OS and platform: Linux Fedora

gvarisco, 2019-08-31:
hi @piif, thanks for opening this issue. Could you please give another try by using a newer CLI version? You can find latest releases [here](https://github.com/arduino/arduino-cli/releases). Just follow [this README](https://github.com/arduino/arduino-cli#how-to-install) if you have any questions. 

Let us know if JSON output works fine with any of those newer builds!

piif, 2019-09-01:
Hi,
  It works fine with latest commit, sorry for the needless report

closed 2019-09-01 by piif