arduino/arduino-cli #392: questiong about compilation & build caching

csurf, 2019-09-04:
sorry if this is a dumb question, but does `arduino-cli compile` forcibly recompile everything each time it is run? is there a way to speed this up by only rebuilding object files for those files that have changed? 

Unless there's something wrong with my setup, this doesn't appear to be the case. Each time I run `arduino-cli compile mySketch'`, I see the message `Build options changed, rebuilding all`, and I can see it going through a full compile of all libraries & source files. 

I've tried specifying directories for both the core & sketch builds, but it doesn't appear to be doing much. What am I missing? Is there a way to keep the build process from constantly rebuilding unchanged source files & libraries in order to speed things up?

FYI, in my case, one thing I'm seeing is that the build process finds multiple versions of diferent libraries, and informs me that it's selecting a certain version:
```
Build options changed, rebuilding all
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
Alternatives for LiquidCrystal_I2C.h: [NewLiquidCrystal_lib]
ResolveLibrary(LiquidCrystal_I2C.h)
  -> candidates: [NewLiquidCrystal_lib]
Alternatives for TMC2130Stepper.h: [TMC2130Stepper@2.5.1]
ResolveLibrary(TMC2130Stepper.h)
  -> candidates: [TMC2130Stepper@2.5.1]
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
Alternatives for Cmd.h: [Cmd]
ResolveLibrary(Cmd.h)
  -> candidates: [Cmd]
Sketch uses 21946 bytes (68%) of program storage space. Maximum is 32256 bytes.
Global variables use 1419 bytes (69%) of dynamic memory, leaving 629 bytes for local variables. Maximum is 2048 bytes.
```
please help me figure out how to get some kind of build caching to work...

cmaglie, 2019-09-04:
There was an issue in the build caching, that is solved in `master` but not yet released: https://github.com/arduino/arduino-cli/pull/367

You may want to try a nightly build to see if the problem is fixed: https://github.com/arduino/arduino-cli#download-a-nightly-build

masci, 2019-09-04:
The release candidate for the upcoming 0.5.0 conains the fix, see https://github.com/arduino/arduino-cli/releases/tag/0.5.0-rc1 for a complete changelog

masci, 2019-09-05:
0.5.0 is out, closing

closed 2019-09-05 by masci