arduino/arduino-cli #397: Include library from local folder

tage64, 2019-09-07:
Hello,
How do I install a library from a local folder. I've downloaded grbl and tried to include it as a library, but it seems I cannot install local libraries.

Is this possible? Or have I miss understood something?

Best regards,
Tage

per1234, 2019-09-07:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2019-09-07 by per1234

matthijskooijman, 2019-09-08:
See https://www.arduino.cc/en/Guide/Libraries for guidance on where and how libraries should be installed.