arduino/arduino-cli #398: arduino-builder on Travis-CI Win32 instance error reporting results in another error, "errors.errorString invalid syntax"

earlephilhower, 2019-09-07:
## Bug Report

Just to be up front, I'm not sure if this is an error in arduino-builder(really, inside -cli now) or an issue with the beta Travis-CI Windows VMs.

When trying to report some error, arduino-builder actually ends up with an exception in the error printing routines themselves.  (I can't really identify the first error it's trying to report exactly, since its crashing while trying to print it, but I believe from the trace it's related to a `hardware` directory issue in my CI setup.)  I'm fine with having a bug in my new CI config, but with it crashing I can't actually see what the problem is.

### Current behavior
(taken from an esp8266-Arduino CI run on Travis-CI under Windows at https://api.travis-ci.org/v3/job/582174496/log.txt )

````
C:\Users\travis\arduino_ide\arduino-builder -compile -logger=human -build-path "C:\Users\travis\build\esp8266\Arduino\build.tmp" -tools "C:\Users\travis\arduino_ide\tools-builder" -libraries "C:\Users\travis\Arduino\libraries" -hardware "C:\Users\travis\arduino_ide\hardware" -hardware "C:\Users\travis\build\esp8266\Arduino\tools\..\cores" -fqbn=esp8266com:esp8266:generic:xtal=80,FlashFreq=40,FlashMode=qio,baud=921600,eesz=4M1M,ip=lm2f,ResetMethod=nodemcu -built-in-libraries "C:\Users\travis\arduino_ide\libraries" -ide-version=10607 -warnings=all -verbose C:\Users\travis\build\esp8266\Arduino\libraries\ArduinoOTA\examples\BasicOTA\BasicOTA.ino
travis_time:end:0ee6ce3c:start=1567895616376247900,finish=1567895716459417300,duration=100083169400,event=script

...

*errors.errorString invalid syntax
/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/i18n/errors.go:19 (0x858073)
/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:440 (0x85804a)
/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:249 (0x857910)
/home/jenkins/go1.12.5/src/runtime/proc.go:200 (0x42bc87)
/home/jenkins/go1.12.5/src/runtime/asm_386.s:1321 (0x4526f1)
````

### Expected behavior
I'd expect the appropriate error message to be displayed.

### Environment

- CLI version (output of `arduino-cli version`):   arduino-builder nightly from https://downloads.arduino.cc/arduino-nightly-linux64.tar.xz
- OS and platform:  Win32 on Travis-CI

facchinm, 2019-09-09:
From the log it looks like something horrible is going on with paths handling (unix -> win -> cygwin-like)
```
 ------------ Building /c/Users/travis/build/esp8266/Arduino/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino ------------ 

python3 tools/build.py -b generic -v -w all -s 4M1M -v -k -p C:/Users/travis/build/esp8266/Arduino/build.tmp -n lm2f -l C:/Users/travis/Arduino/libraries  C:/Users/travis/build/esp8266/Arduino/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino
```
Which version of the builder are you using? 1.4.6? 
@cmaglie

facchinm, 2019-09-09:
Ah, and I'm moving back the issue to arduino-builder repo