arduino/arduino-cli #403: Cannot find ports

LepikhinEgor, 2019-09-13:
## Bug Report

### Current behavior

Arduino-cli cannot find connected boards by command "board list". But Arduino IDE easily finds boards and upload scetches.  
Previously used the arduino-cli version 0.3.6-alpha.preview version and everything worked

When i run "arduino-cli board list -v" output is 

> arduino-cli board list -v
INFO[0000] Checking for config file in: C:\Users\Людмила\AppData\Local\Arduino15\arduino-cli.yaml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: D:\git-projects\ard-console
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Checking for config in: C:\arduino-cli.yaml
INFO[0000] error loading                                 error="open C:\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\arduino-cli.yaml
INFO[0000] error loading                                 error="open C:\\Users\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\Людмила\arduino-cli.yaml
INFO[0000] error loading                                 error="open C:\\Users\\Людмила\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\Людмила\AppData\arduino-cli.yaml
INFO[0000] error loading                                 error="open C:\\Users\\Людмила\\AppData\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\Людмила\AppData\Local\arduino-cli.yaml
INFO[0000] error loading                                 error="open C:\\Users\\Людмила\\AppData\\Local\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\Людмила\AppData\Local\Roboblock\arduino-cli.yaml
INFO[0000] error loading                                 error="open C:\\Users\\Людмила\\AppData\\Local\\Roboblock\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Configuration set
INFO[0000] arduino-cli.exe-0.3.7-alpha.preview
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Initializing package manager
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: D:\git-projects\ard-console
INFO[0000] Loading hardware from: C:\Users\Людмила\AppData\Local\Arduino15\packages
INFO[0000] Loading package arduino from: C:\Users\Людмила\AppData\Local\Arduino15\packages\arduino\hardware
INFO[0000] Loaded platform                               platform="arduino:avr@1.6.23"
INFO[0000] Checking existence of 'tools' path: C:\Users\Людмила\AppData\Local\Arduino15\packages\arduino\tools
INFO[0000] Loading tools from dir: C:\Users\Людмила\AppData\Local\Arduino15\packages\arduino\tools
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.1"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino14"
INFO[0000] Loading package builtin from: C:\Users\Людмила\AppData\Local\Arduino15\packages\builtin
INFO[0000] Checking existence of 'tools' path: C:\Users\Людмила\AppData\Local\Arduino15\packages\builtin\tools
INFO[0000] Loading tools from dir: C:\Users\Людмила\AppData\Local\Arduino15\packages\builtin\tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.0.0"
INFO[0000] Loading package esp32 from: C:\Users\Людмила\AppData\Local\Arduino15\packages\esp32\hardware
INFO[0000] Loaded platform                               platform="esp32:esp32@1.0.2"
INFO[0000] Checking existence of 'tools' path: C:\Users\Людмила\AppData\Local\Arduino15\packages\esp32\tools
INFO[0000] Loading tools from dir: C:\Users\Людмила\AppData\Local\Arduino15\packages\esp32\tools
INFO[0000] Loaded tool                                   tool="esp32:esptool_py@2.6.1"
INFO[0000] Loaded tool                                   tool="esp32:mkspiffs@0.2.3"
INFO[0000] Loaded tool                                   tool="esp32:xtensa-esp32-elf-gcc@1.22.0-80-g6c4433a-5.2.0"
INFO[0000] Loading package esp8266 from: C:\Users\Людмила\AppData\Local\Arduino15\packages\esp8266\hardware
INFO[0000] Loaded platform                               platform="esp8266:esp8266@2.5.2"
INFO[0000] Checking existence of 'tools' path: C:\Users\Людмила\AppData\Local\Arduino15\packages\esp8266\tools
INFO[0000] Loading tools from dir: C:\Users\Людмила\AppData\Local\Arduino15\packages\esp8266\tools
INFO[0000] Loaded tool                                   tool="esp8266:esptool@2.5.0-3-20ed2b9"
INFO[0000] Loaded tool                                   tool="esp8266:mkspiffs@2.5.0-3-20ed2b9"
INFO[0000] Loaded tool                                   tool="esp8266:python@3.7.2-post1"
INFO[0000] Loaded tool                                   tool="esp8266:xtensa-lx106-elf-gcc@2.5.0-3-20ed2b9"
INFO[0000] Loading hardware from: C:\Users\Людмила\Documents\Arduino\hardware
INFO[0000] Loading package espressif from: C:\Users\Людмила\Documents\Arduino\hardware\espressif
INFO[0000] Loaded platform                               platform="espressif:esp32-1.0.2@"
INFO[0000] Adding libraries dir                          dir="C:\\Users\\Людмила\\Documents\\Arduino\\libraries" location=sketchbook
INFO[0000] Adding libraries dir                          dir="C:\\Users\\Людмила\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2\\libraries" location=platform
INFO[0000] Adding libraries dir                          dir="C:\\Users\\Людмила\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\libraries" location=platform
INFO[0000] Adding libraries dir                          dir="C:\\Users\\Людмила\\Documents\\Arduino\\hardware\\espressif\\esp32-1.0.2\\libraries" location=platform
INFO[0000] Adding libraries dir                          dir="C:\\Users\\Людмила\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.6.23\\libraries" location=platform
No boards found.

### Environment

- CLI version : arduino-cli.exe Version: 0.3.7-alpha.preview Commit: 032fa05
- OS and platform: Windows 10 Home, 64x

masci, 2019-09-18:
Hi @LepikhinEgor what boards do you have connected? Is the corresponding core installed?

LepikhinEgor, 2019-09-18:
Нes, i try connect esp8266(nodemcu) and and esp8266 core was installled. Then I tried to connect Arduino mega and uno, but they also could not be found

bularond, 2019-09-24:
Also have this problem. Arduino IDE find port without problem.

<details>
bulat@bulat-PC:~$ arduino-cli board list -v <br>
INFO[0000] Checking for config file in: /home/bulat/.arduino15/arduino-cli.yaml <br>
INFO[0000] Checking if CLI is Bundled into the IDE<br>
INFO[0000] Candidate IDE Directory: /bin<br>
INFO[0000] CLI is not bundled into the IDE     <br>         
INFO[0000] Checking for config in: /arduino-cli.yaml    <br>
INFO[0000] error loading                                 error="open /arduino-cli.yaml: no such file or directory" <br>
INFO[0000] Checking for config in: /home/arduino-cli.yaml <br>
INFO[0000] error loading                                 error="open /home/arduino-cli.yaml: no such file or directory" <br>
INFO[0000] Checking for config in: /home/bulat/arduino-cli.yaml <br>
INFO[0000] error loading                                 error="open /home/bulat/arduino-cli.yaml: no such file or directory" <br>
INFO[0000] Configuration set                            <br>
INFO[0000] arduino-cli-0.5.0                            <br>
INFO[0000] Starting root command preparation (`arduino`) <br>
INFO[0000] Formatter set                                <br>
INFO[0000] Initializing package manager               <br>  
INFO[0000] Checking if CLI is Bundled into the IDE      <br>
INFO[0000] Candidate IDE Directory: /bin                <br>
INFO[0000] Loading hardware from: /home/bulat/.arduino15/packages <br>
INFO[0000] Loading package arduino from: /home/bulat/.arduino15/packages/arduino/hardware <br>
INFO[0000] Loaded platform                               platform="arduino:avr@1.8.1" <br>
INFO[0000] Checking existence of 'tools' path: /home/bulat/.arduino15/packages/arduino/tools <br>
INFO[0000] Loading tools from dir: /home/bulat/.arduino15/packages/arduino/tools <br>
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.3.0" <br>
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5"<br>
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino17"<br>
INFO[0000] Loading package builtin from: /home/bulat/.arduino15/packages/builtin <br>
INFO[0000] Checking existence of 'tools' path: /home/bulat/.arduino15/packages/builtin/tools <br>
INFO[0000] Loading tools from dir: /home/bulat/.arduino15/packages/builtin/tools <br>
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"<br>
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.0.0"<br>
INFO[0000] Adding libraries dir                          dir=/home/bulat/Arduino/libraries location=sketchbook <br>
INFO[0000] Adding libraries dir                          dir=/home/bulat/.arduino15/packages/arduino/hardware/avr/1.8.1/libraries location=platform <br>
No boards found.<br>
</details>

----------------

Board: Arduino Nano (non official)
OS: Ubuntu 18.04.3 LTS
arduino-cli: 0.5.0 Commit: 3be2287

bularond, 2019-09-24:
Installed pre-relised version (0.5.0-showports) and it work!

```bash
bulat@bulat-PC:~/programs$ sudo ./arduino-cli board list
Port         Type              Board Name FQBN Core
/dev/ttyUSB0 Serial Port (USB) Unknown
```

masci, 2019-09-24:
Thanks for confirming @bularond merging the branch right now so it'll be available in today's nightly!

masci, 2019-09-24:
Fixed in https://github.com/arduino/arduino-cli/pull/411

closed 2019-10-01 by masci