arduino/arduino-cli #406: Compile still cannot find sketch

NormanDunbar, 2019-09-14:
## Bug Report

### Current behavior

Create a sketch as follows:
````
arduino-cli sketch new Testing123
Sketch created in: /home/norman/Arduino/Testing123
````
Now, after editing, attempt to compile it:
````
arduino-cli compile --fqbn arduino:avr:diecimila:cpu=atmega328 Testing123
````
This fails with the following message:
````
Error during build: build failed: unable to stat Sketch location: 
stat /home/norman/Testing123: 
no such file or directory
````
However, if I specify `Arduino` in the sketch name, as follows, it compiled perfectly:
````
arduino-cli compile --fqbn arduino:avr:diecimila:cpu=atmega328 Arduino/Testing123
Sketch uses 930 bytes (3%) of program storage space. Maximum is 30720 bytes.
````
I would expect the `sketch new` and `compile` or `upload` commands to assume the same default location for a sketch.

As requested by @cmaglie in comment https://github.com/arduino/arduino-cli/issues/51#issuecomment-530859579 I'm raising a new issue.

### Expected behavior
I would expect the default location for a `sketch new` command to be the same one used by a `compile` or `upload` command? At least, this makes sense to me, otherwise, why have a default?

It appears that the default location for `sketch new` is `$HOME/Arduino`, but for `compile` and `upload` it's just `$HOME`.

### Environment
- CLI version (output of `arduino-cli version`): arduino-cli Version: 0.5.0 Commit: 3be2287
- OS and platform: Linux Mint 19.3 64 bit.

Cheers,
Norm.

ozzeruk, 2019-09-24:
I agree fully. The defaults here create confusion. I had an identical situation when first using arduino-cli and I'm sure pretty much all users will run into this.

masci, 2019-09-24:
Thanks for the feedback, I'll consolidate the commands to avoid confusion.

masci, 2019-10-01:
Fixed in https://github.com/arduino/arduino-cli/pull/396

closed 2019-10-01 by masci