arduino/arduino-cli #407: The install.sh install script failing with "unexpected operator"

TomasHubelbauer, 2019-09-15:
I am trying to install the Arduino CLI in a workflow defined in GitHub Actions (beta).

```sh
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh
```

Running this I get:

```
Installing in /home/runner/work/arduino-nmea-checksum/arduino-nmea-checksum/bin
ARCH=64bit
OS=Linux
Using curl as download tool
sh: 81: [: x0.5.0: unexpected operator
TAG=0.5.0
sh: 125: [: Linux: unexpected operator
CLI_DIST=arduino-cli_0.5.0_Linux_64bit.tar.gz
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.5.0_Linux_64bit.tar.gz
arduino-cli not found. Did you add /home/runner/work/arduino-nmea-checksum/arduino-nmea-checksum/bin to your $PATH?
Failed to install arduino-cli
```

The mentioned lines look like normal conditional statements with successful variable substitutions in them, not sure why the would be problematic.

I found #348 and attempted to use the suggested fix:

```sh
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=/home/runner/work/arduino-nmea-checksum/arduino-nmea-checksum/bin sh
```

However with this I just get:

```
/home/runner/work/arduino-nmea-checksum/arduino-nmea-checksum/bin ($BINDIR) folder not found. Please create it before continuing.
Failed to install arduino-cli
```

What could be the problem? You can see my repo here:

https://github.com/TomasHubelbauer/arduino-nmea-checksum

The branch where I'm trying this is `arduino-cli-latest` and the corresponding workflow runs with their captured logs are accessible using the Actions tab. There are two runs titled after the branch and the contain the logs that I've quoted above.

- CLI version (output of `arduino-cli version`): latest
- OS and platform: Ubuntu latest

TomasHubelbauer, 2019-09-15:
BTW I've also noticed that the `checkLatestVersion` function in the install script attempts to determine the latest release by parsing the contents of the GitHub Releases page, but I believe a better approach would be to use the GitHub API. Is there a reason it was not used? I could try to PR the install script to use the API instead.

Also, even sticking to the GitHub Releases page approach, there is no need to determine the latest version this way as you can construct a URL to the latest version easily, as per [Linking to releases](https://help.github.com/en/articles/linking-to-releases):

https://github.com/arduino/arduino-cli/releases/latest

Also it is possible to link directly to an asset of a release, check out the above link to see how to construct that URL. Although in case of Arduino CLI where the asset names include the version name it would be necessary to determine the release version anyway. Which could still be done by going to the URL of the latest release directly and parsing it out of  the document or the URL after the redirect.

Again, willing to take a stab at improving this if there's interest.

per1234, 2019-09-15:
>I believe a better approach would be to use the GitHub API. Is there a reason it was not used?

It was used in the script originally. The problem is that when the script is run from an online CI service, the build is sharing the IP address with many other builds and this caused the installation of arduino-cli to frequently fail due to the IP address having exceeded GitHub's API rate limits, so that approach had to be abandoned:
https://github.com/arduino/arduino-cli/commit/37a8e5eaed00896cb2e348fc95a821e3a0a0b1d9#diff-3fbb47e318cd8802bd325e7da9aaabe8

---
BTW, here's the system I have always used for getting the latest tag:
```shell
git clone --quiet https://github.com/arduino/arduino-cli
cd arduino-cli
# get new tags from the remote
git fetch --tags
# determine the latest tag
echo "Latest tag: $(git describe --tags "$(git rev-list --tags --max-count=1)")"
```

masci, 2019-09-16:
Hi @TomasHubelbauer thanks for the report!

The first issue you encountered is the install script complaining about the cli binary not being available to the PATH, as I mentioned in another issue this should just print a warning and exit with `0` instead of erroring out, my bad I didn't have the time to fix it.

About the second issue, according to the error message you get

```
<snip> ($BINDIR) folder not found. Please create it before continuing.
```
the fix should just be creating the folder where you want the archive to be extracted. For example, this works:

```yaml
name: ci

on: push

jobs:
  ci:
    runs-on: ubuntu-latest

    steps:
      - uses: actions/checkout@v1

      - name: Install Arduino CLI
        env:
          BINDIR: ./bin
        run: |
          mkdir $BINDIR
          export PATH=$PATH:$BINDIR
          curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=$BINDIR sh

      - name: Run Arduino CLI
        run: ./bin/arduino-cli version
```

> Also, even sticking to the GitHub Releases page approach, there is no need to determine the latest version this way as you can construct a URL to the latest version easily, as per Linking to releases:

Unless you hardcode the version into the script itself (and we don't want to), the script doesn't know which is the latest tag, hence the `latest` page scraping. As @per1234 already mentioned, this was preferred over using the GitHub API do avoid forcing newcomers to setup an access token (a-la-brew) to use our install script.

TomasHubelbauer, 2019-09-16:
@masci One thing I am confused about is how to get it to install to default BINDIR. I would like to arrive at a minimal amout of lines which produce the final result of being able to call the Arduino CLI directly using `arduino-cli` not `./arduino-cli` . I assumed the script will install to a default OS location which is already in path. Maybe that even happens but it just doesn't work on the GitHub Actions runner for some reason?

masci, 2019-09-16:
Unfortunately the script doesn't have any logic to detect the PATH, this was by design to avoid chasing corner cases down a rabbit hole.

I think the cleanest approach in this context to have `arduino-cli` available from a job in a workflow is to have a dedicated javascript action, this would look a lot like one of these: https://github.com/arduino/actions. I'll give it a shot tomorrow and post a follow up in this thread, thanks for the feedback!

masci, 2019-09-17:
Hi @TomasHubelbauer I was able to create the action and I can now do this:
```
name: test

on: [push]

jobs:
  test-cli:
    runs-on: ubuntu-latest

    steps:
      - name: Install Arduino CLI
        uses: Arduino/actions/setup-arduino-cli@master
        with:
          version: '0.x'

      - name: Test CLI
        run: arduino-cli version
```
Can you let me know if this fits your use case?

TomasHubelbauer, 2019-09-17:
Yes! Thank you, will check it out soon.

closed 2019-09-17 by TomasHubelbauer

TomasHubelbauer, 2019-09-17:
Verified this works well: https://github.com/TomasHubelbauer/arduino-nmea-checksum
Thank you!

masci, 2019-09-17:
Ah the workflow looks super clear now, happy it worked out.

And thank you for the feedback and for using the CLI!