arduino/arduino-cli #414: fatal error: concurrent map read and map write when working with the daemon

kittaakos, 2019-09-18:
## Bug Report

I do not know how this has happened.

```
arduino-cli daemon -v --log-level info --format json --log-format json
fatal error: concurrent map read and map write

goroutine 29 [running]:
runtime.throw(0x48440d1, 0x21)
        /usr/local/go/src/runtime/panic.go:617 +0x72 fp=0xc00014dca0 sp=0xc00014dc70 pc=0x402dba2
runtime.mapaccess1_faststr(0x47ee540, 0xc0000b5020, 0x4830cbf, 0xa, 0xc00014dd70)
        /usr/local/go/src/runtime/map_faststr.go:21 +0x469 fp=0xc00014dd10 sp=0xc00014dca0 pc=0x4014129
net/http.http2isConnectionCloseRequest(...)
        /usr/local/go/src/net/http/h2_bundle.go:8937
net/http.(*http2clientConnReadLoop).endStreamError(0xc00014dfb8, 0xc000d13b80, 0x492cc40, 0xc000080030)
        /usr/local/go/src/net/http/h2_bundle.go:8571 +0x11a fp=0xc00014dd80 sp=0xc00014dd10 pc=0x431f06a
net/http.(*http2clientConnReadLoop).endStream(...)
        /usr/local/go/src/net/http/h2_bundle.go:8562
net/http.(*http2clientConnReadLoop).processData(0xc00014dfb8, 0xc001d56360, 0xc001d56360, 0x0)
        /usr/local/go/src/net/http/h2_bundle.go:8552 +0x2b2 fp=0xc00014de38 sp=0xc00014dd80 pc=0x431ead2
net/http.(*http2clientConnReadLoop).run(0xc00014dfb8, 0x4863740, 0xc0002647b8)
        /usr/local/go/src/net/http/h2_bundle.go:8137 +0x5eb fp=0xc00014df70 sp=0xc00014de38 pc=0x431c72b
net/http.(*http2ClientConn).readLoop(0xc00009ec00)
        /usr/local/go/src/net/http/h2_bundle.go:8030 +0x76 fp=0xc00014dfd8 sp=0xc00014df70 pc=0x431ba86
runtime.goexit()
        /usr/local/go/src/runtime/asm_amd64.s:1337 +0x1 fp=0xc00014dfe0 sp=0xc00014dfd8 pc=0x405a661
created by net/http.(*http2Transport).newClientConn
        /usr/local/go/src/net/http/h2_bundle.go:7093 +0x637

goroutine 1 [IO wait, 8 minutes]:
internal/poll.runtime_pollWait(0x50b3ea8, 0x72, 0x0)
        /usr/local/go/src/runtime/netpoll.go:182 +0x56
internal/poll.(*pollDesc).wait(0xc0001e4998, 0x72, 0x0, 0x0, 0x482d2fa)
        /usr/local/go/src/internal/poll/fd_poll_runtime.go:87 +0x9b
internal/poll.(*pollDesc).waitRead(...)
        /usr/local/go/src/internal/poll/fd_poll_runtime.go:92
internal/poll.(*FD).Accept(0xc0001e4980, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
        /usr/local/go/src/internal/poll/fd_unix.go:384 +0x1ba
net.(*netFD).accept(0xc0001e4980, 0xc000539401, 0x0, 0x0)
        /usr/local/go/src/net/fd_unix.go:238 +0x42
net.(*TCPListener).accept(0xc0000c25c0, 0xc0000dbb70, 0xc0000dbb78, 0x18)
        /usr/local/go/src/net/tcpsock_posix.go:139 +0x32
net.(*TCPListener).Accept(0xc0000c25c0, 0x4863118, 0xc0000c6180, 0x4945fe0, 0xc0005394c0)
        /usr/local/go/src/net/tcpsock.go:260 +0x48
google.golang.org/grpc.(*Server).Serve(0xc0000c6180, 0x493c0a0, 0xc0000c25c0, 0x0, 0x0)
        /github/home/go/pkg/mod/google.golang.org/grpc@v1.21.1/server.go:587 +0x1e9
github.com/arduino/arduino-cli/cli/daemon.runDaemonCommand(0xc00020b400, 0xc0001f4d20, 0x0, 0x7)
        /__w/arduino-cli/arduino-cli/cli/daemon/daemon.go:74 +0x44e
github.com/spf13/cobra.(*Command).execute(0xc00020b400, 0xc0001f4cb0, 0x7, 0x7, 0xc00020b400, 0xc0001f4cb0)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x2ae
github.com/spf13/cobra.(*Command).ExecuteC(0x4ea4300, 0x46973b1, 0xc0001fdf88, 0xc00003c0b8)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x2fc
github.com/spf13/cobra.(*Command).Execute(...)
        /github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
        /__w/arduino-cli/arduino-cli/main.go:28 +0x2e

goroutine 4 [syscall, 8 minutes]:
os/signal.signal_recv(0x0)
        /usr/local/go/src/runtime/sigqueue.go:139 +0x9f
os/signal.loop()
        /usr/local/go/src/os/signal/signal_unix.go:23 +0x22
created by os/signal.init.0
        /usr/local/go/src/os/signal/signal_unix.go:29 +0x41

goroutine 6 [select, 8 minutes]:
google.golang.org/grpc/internal/transport.(*controlBuffer).get(0xc0000c9bc0, 0x1, 0x0, 0x0, 0x0, 0x0)
        /github/home/go/pkg/mod/google.golang.org/grpc@v1.21.1/internal/transport/controlbuf.go:317 +0x104
google.golang.org/grpc/internal/transport.(*loopyWriter).run(0xc0000c4ba0, 0x0, 0x0)
[... 119 lines cut ...]
github.com/arduino/arduino-cli/rpc/commands._ArduinoCore_BoardList_Handler(0x480f300, 0xc0001c6e00, 0x493eaa0, 0xc003860720, 0xc001d5f9a0, 0x0, 0x493eaa0, 0xc003860720, 0xc002c2dfa0, 0x4)
        /__w/arduino-cli/arduino-cli/rpc/commands/commands.pb.go:1551 +0x23e
google.golang.org/grpc.(*Server).processUnaryRPC(0xc0000c6180, 0x4946b20, 0xc0004af980, 0xc005b3d700, 0xc000220750, 0x4ea3700, 0x0, 0x0, 0x0)
        /github/home/go/pkg/mod/google.golang.org/grpc@v1.21.1/server.go:998 +0x470
google.golang.org/grpc.(*Server).handleStream(0xc0000c6180, 0x4946b20, 0xc0004af980, 0xc005b3d700, 0x0)
        /github/home/go/pkg/mod/google.golang.org/grpc@v1.21.1/server.go:1278 +0xda6
google.golang.org/grpc.(*Server).serveStreams.func1.1(0xc000eea400, 0xc0000c6180, 0x4946b20, 0xc0004af980, 0xc005b3d700)
        /github/home/go/pkg/mod/google.golang.org/grpc@v1.21.1/server.go:717 +0x9f
created by google.golang.org/grpc.(*Server).serveStreams.func1
        /github/home/go/pkg/mod/google.golang.org/grpc@v1.21.1/server.go:715 +0xa1

    at ChildProcess.exithandler (child_process.js:299:12)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:962:16)
    at Socket.stream.socket.on (internal/child_process.js:381:11)
    at Socket.emit (events.js:182:13)
    at Pipe._handle.close (net.js:606:12)
```

- CLI version (output of `arduino-cli version`):
`arduino-cli Version: 0.3.7-alpha.preview Commit: ac85e06`
- OS and platform:
macOS `10.14.6 (18G87)`

### Additional context

cmaglie, 2019-09-27:
There is a (more or less) reliable way to reproduce it?
I have some idea, but I'd like to test the fix too.

kittaakos, 2019-10-07:
> reliable way to reproduce it?

I can give you the steps offline, @cmaglie. I do not have a self-contained way to reproduce it with the CLI only.

PR #438 (2019-10-07): Removed globals.HTTPClientHeader global
Fix #414 

/cc @rsora
closed 2019-10-07 by rsora