arduino/arduino-cli #435: Arduino-cli board list: no boards found using UART GPIO serial tx rx

lucadelu, 2019-10-03:
We are trying to connect an Arduino MKR WAN 1300 device to a Raspberry Pi zero W computer board by wiring the 13(RX), 14(TX) Pins of Arduino and the corresponding GPIO14 (UART0_TXD), GPIO15 (UART0_RXD) on [RPi zero W](https://www.arduino.cc/reference/en/language/functions/communication/serial/).
Following several online tutorials and the arduino-cli installation guide, we were able so far to configure RPi properly and to install all the required software.
Moreover, we were able to successfully test the UART connection by uploading a simple sketch on Arduino
```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

 and testing the serial communication on Raspberry with minicom (minicom -b 9600 -o -D /dev/ttyAMA0)
However, all attempts both to detect the board (arduino-cli board list) or to upload a compiled sketch (arduino-cli board attach serial:///dev/tty/AMA0 MyFirstSketch) to arduino mkr wan 1300 by means of arduino-cli has failed!
We are wondering if the communication via uart is available in the Arduino-CLI code?
Any help would be greatly appreciated!

facchinm, 2019-10-07:
I think it's due to the fact the the cli only lists USB serial ports, while ttyAMA0 is definitely an hardware serial :slightly_smiling_face: 
@AlbyIanna were you taking a look at this (or something similar)?

AlbyIanna, 2019-10-11:
Nope, sorry! :(

facchinm, 2019-10-11:
@cmaglie  do you think we could also list native serial ports (maybe behind a flag)? The change should be at go-serial library level probably

cmaglie, 2019-10-11:
of course we should, but I'm wondering why it's not listed, the cli should list all available serial ports (even if not "recognized")

cmaglie, 2019-10-11:
@lucadelu which version of the cli are you using? please post the output of `arduino-cli version`

lucadelu, 2019-10-14:
@cmaglie arcuino-cli is version 0.5.0 commit 3be3287

cmaglie, 2019-10-16:
Uhm, probably this version still hides the unrecognized ports (I don't know if #411 has been merged before or after 0.5.0 @masci ?)

@lucadelu to be sure can you try the latest nightly? https://github.com/arduino/arduino-cli#download-a-nightly-build

masci, 2019-10-16:
No, the feature didn't make it to 0.5.0. There's this release https://github.com/arduino/arduino-cli/releases/tag/0.5.0-showports which is a 0.5.0 plus just that feature added or a nightly as you mentioned.

lucadelu, 2019-10-17:
@masci @cmaglie we tested with `0.5.0-showports` and it has some improvements, now it list the board has Unknown

```
arduino-cli board list
Port         Type        Board Name FQBN Core
/dev/ttyAMA0 Serial Port Unknown
```

but it is still not possible to upload the sketch

```
arduino-cli upload -p /dev/ttyAMA0 --fqbn arduino:samd:mkrwan1300 test_arduino_connection
No new serial port detected.
No device found on ttyAMA0
Error during Upload: uploading error: exit status 1
```

Using the USB cable instead it is working correctly

```
arduino-cli board list
Port         Type              Board Name           FQBN                    Core
/dev/ttyACM0 Serial Port (USB) Arduino MKR WAN 1300 arduino:samd:mkrwan1300 arduino:samd
/dev/ttyAMA0 Serial Port       Unknown

arduino-cli upload -p /dev/ttyACM0 --fqbn arduino:samd:mkrwan1300 test_arduino_connection
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] Nov 13 2017 18:12:39
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.632 seconds

Write 11560 bytes to flash (181 pages)
[==============================] 100% (181/181 pages)
done in 0.091 seconds

Verify 11560 bytes of flash with checksum.
Verify successful
done in 0.016 seconds
CPU reset.
```

The serial communication using GPIO is working using minicom on /dev/ttyAMA0

facchinm, 2019-10-17:
@lucadelu as per https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/board_definitions_arduino_mkrwan1300.h#L51 we do provide also the UART bootloader, so the problem may just be the fact that you must trigger the bootloader manually (via doubleclicking the reset button for example).
When in bootloader, the upload rule should be quite similar to the USB one (but not the same since `bossac` takes `-Ufalse` when not using the USB. It MAY work but it would be nice if you could report your results

lucadelu, 2019-10-22:
@facchinm yes if we trigger the bootloader manually (via doubleclicking the reset button) it works as expected. Now we will look for a way to do the reset through the RaspberryPi (using the RESET pin could work?)

Thanks for your support.

facchinm, 2019-10-22:
Reset pin definitely works, you just need to simulate the double pressure fast enough to stay in the timeout and slow enough to let the bootloader start and recognize the second event :wink:

lucadelu, 2019-11-04:
Hi all,

here the code used by my colleague to upload an Arduino sketch from Raspberry

```
#!/bin/bash
# Set an Arduino MKR WAN 1300 in bootloader mode with Raspberry Pi zero W and prepare for uploading a sketch by means of arduino-cli 
# usage: /bin/bash /home/pi/scripts/mkrwan_reset_upload.sh <argument = compiled sketch name>
# wiring table:
# RPi zero W      <--> Arduino MKR WAN 1300
# pin 14 (GND)    <--> GND 
# pin 16 (GPIO23) <--> RESET
# pin  8 (GPIO14) <--> 13 RX
# pin 10 (GPIO15) <--> 14 TX
# Author: Roberto Zorer Fondazione Edmund Mach
# email: roberto.zorer@fmach.it
# References: 
# script adapted from: https://medium.com/coinmonks/controlling-raspberry-pi-gpio-pins-from-bash-scripts-traffic-lights-7ea0057c6a90
# Arduino-cli: https://github.com/arduino/arduino-cli/releases/tag/0.5.0-showports

# clear screen
clear

# Arduino-cli default directory for uploading
CLI_PATH=/home/pi/Arduino

# Path to device
MYDEV=/dev/ttyAMA0

# The FQBN to use
MYFQBN=arduino:samd:mkrwan1300

# Check for the existence of an input argument
if [[ $# -eq 0 ]] ; then
    echo 'No argument supplied, the script will exit!'
    echo 'This are the available Arduino sketches for uploading:'
	ls -1 $CLI_PATH
	exit 1
fi

# Check existence of the sketch file
if [ ! -f "${CLI_PATH}/$1/$1.ino" ]; then
    echo "The sketch '$1' does not exist, please add, as argument, an existing sketch name from this list:"
	ls -1 $CLI_PATH
	exit 1
fi

# else run the script!
sketch=$1
echo "Set an Arduino MKR WAN 1300 in bootloader mode and upload the sketch: "$sketch

# Common path for all GPIO access
BASE_GPIO_PATH=/sys/class/gpio

# Assign name RESET to GPIO pin 23
RESET=23

# Assign names to states
ON="1"
OFF="0"

# Utility function to export a pin if not already exported
exportPin()
{
  if [ ! -e $BASE_GPIO_PATH/gpio$1 ]; then
	echo "$1" > $BASE_GPIO_PATH/export
  fi
}

# Utility function to set a pin as an output
setOutput()
{
  echo "out" > $BASE_GPIO_PATH/gpio$1/direction
}

# Utility function to change state of a Pin
setPinState()
{
  echo $2 > $BASE_GPIO_PATH/gpio$1/value
}

# Utility function to turn reset pin off
resetPinOff()
{
  setPinState $RESET $OFF
}

# Utility function to turn reset pin high
resetPinOn()
{
  setPinState $RESET $ON
}

# Export pins so that we can use them
exportPin $RESET

# Set pins as outputs
setOutput $RESET

# Turn RESET pin on and wait 1 s to begin the reset procedure (off-on-off-on)
resetPinOn
sleep 1

# Simulate double click to reset Arduino MKR WAN 1300

echo "First reset 0.1 s"
resetPinOff
sleep 0.1

resetPinOn
sleep 0.1

echo "Second reset 0.1 s"

resetPinOff
sleep 0.1

resetPinOn

# change active directory, required by arduino-cli
cd $CLI_PATH

echo "Updating arduino-cli indexes"
arduino-cli core update-index

echo "Uploading sketch"
arduino-cli upload -p $MYDEV --fqbn $MYFQBN $sketch

echo "Done!"
```

closed 2019-11-04 by lucadelu